esphome/issues #5495: bme680_bsec shows last read sensor value forever in case of I2C communication errors

bisbastuner, 2024-02-16:
### The problem

I have a Pimoroni PIM357 board hosting a BME680, and I am using it connected to a ESP32 DevKit C through a 3m CAT5e cable.
Sometimes it suddenly happens that the I2C communication fails, but in Home Assistant there's no trace of this faulty condition, except for the sensor values (especially temperature) that stand still with the last reported value forever.
What could be the right behavior of bme680_bsec to communicate this condition to Home Assistant?
I am a programmer and I already contributed to bme680_bsec; I intend to resolve this issue but I'm not an ESPHome expert, only searching for advice.
My ideas:
- report NaN for any listed sensor
- implement a I2C communication error counter, that starts from 0 at boot and increments every time there's a I2C error

What would be the right thing to do? @trvrnrth can you provide some guidance here?
(Side note: the real underlying problem is electrical: I simply need to know there's a problem, then I go disconnect/reconnect the cable and all start working again perfectly... for a couple of months. I may find a better connector in the future)

I see there's a related issue/feature request for this problem: https://github.com/esphome/feature-requests/issues/1568
and many other questions about this problem (inverter that shuts off, connection to temperature reading for climate water heater for fish tank, ...)

### Which version of ESPHome has the issue?

2023.8.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2024.1.6

### What platform are you using?

ESP32

### Board

ESP32 DevKit C NodeMCU (AZ-delivery)

### Component causing the issue

bme680_bsec

### Example YAML snippet

```yaml
##### HW CONFIGURATION #####
esphome:
  name: espino2-out

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

##### Enable logging #####
logger:
#  level: VERBOSE

##### Enable Home Assistant API #####
api:
  reboot_timeout: 24h

ota:
  password: ""

##### Wi-Fi SETUP #####
wifi:
  ssid: ""
  password: ""

  manual_ip:
    static_ip: ...
    subnet: ...
    gateway: ...
    dns1: ...
    dns2: ...

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "espino2-out"

##### LOCAL WEB SERVER #####
web_server:
  local: true

##### PERIPHERALS #####

### I2C bus for the BME680 devices
i2c:
  - id: "i2cbus_bme"
    sda: GPIO26
    scl: GPIO25
    scan: true

### BME680 device #1 using BSEC library
bme680_bsec:
  id: bme680_esterno
  i2c_id: "i2cbus_bme"
  address: 0x76
  sample_rate: ulp
  temperature_offset: 0.5

### Sensors list
sensor:
  # BME680 sensors
  - platform: bme680_bsec
    temperature:
      name: "espino2out_Temperatura"
    pressure:
      name: "espino2out_Pressione"
    humidity:
      name: "espino2out_RH"
    iaq:
      name: "espino2out_IAQ"
    co2_equivalent:
      name: "espino2out_CO2eq"
    breath_voc_equivalent:
      name: "espino2out_VOCeq"

  # Wifi sensors
  - platform: wifi_signal
    name: "espino2out_WifiLevel"

text_sensor:
  # BME680 text sensors
  - platform: bme680_bsec
    iaq_accuracy:
      name: "espino2out_IAQaccuracy"
```

### Anything in the logs that might be useful for us?

```txt
17:16:39	[W]	[bme680_bsec.sensor:227]	Failed to set sensor settings (BME680 Error Code -2)
17:16:49	[D]	[sensor:094]	'espino2out_WifiLevel': Sending state -60.00000 dBm with 0 decimals of accuracy
17:17:49	[D]	[sensor:094]	'espino2out_WifiLevel': Sending state -60.00000 dBm with 0 decimals of accuracy
17:18:49	[D]	[sensor:094]	'espino2out_WifiLevel': Sending state -60.00000 dBm with 0 decimals of accuracy
17:19:49	[D]	[sensor:094]	'espino2out_WifiLevel': Sending state -60.00000 dBm with 0 decimals of accuracy
17:20:49	[D]	[sensor:094]	'espino2out_WifiLevel': Sending state -60.00000 dBm with 0 decimals of accuracy
17:21:39	[W]	[bme680_bsec.sensor:227]	Failed to set sensor settings (BME680 Error Code -2)
```

ssieb, 2024-02-16:
You could use one of the status components to indicate the problem.  Otherwise, you can add a timeout filter to the sensor to set the value to nan.

closed 2024-02-16 by ssieb

bisbastuner, 2024-02-17:
> You could use one of the status components to indicate the problem. Otherwise, you can add a timeout filter to the sensor to set the value to nan.

You mean adding a [Binary Sensor](https://esphome.io/components/binary_sensor/index.html) and setting its [device_class](https://www.home-assistant.io/integrations/binary_sensor/#device-class) to `problem`?

And the right way to signal Home Assistant that a sensor value is unknown, is to set it to NaN?

If so, it seems to me the Right Thing would be to do both, correct?

Thank you for your answer and guidance.

ssieb, 2024-02-17:
https://esphome.io/components/light/status_led

trvrnrth, 2024-02-18:
I imagine you could make use of the `App.get_app_state()` or the component level `status_has...` methods with template binary sensor(s) depending on the level of granularity you require.

bisbastuner, 2024-02-25:
With your help, and many reading of the ESPHome docs, examples, and code, I managed to create a YAML that performs what I wanted.
Thank you very much for your help 🙏

```
##### HW CONFIGURATION #####
esphome:
  name: espino2-out

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

##### Enable logging #####
logger:
#  level: VERBOSE

##### Enable Home Assistant API #####
api:
  reboot_timeout: 24h

ota:
  password: ""

##### Wi-Fi SETUP #####
wifi:
  ssid: ""
  password: ""

  manual_ip:
    static_ip: ...
    subnet: ...
    gateway: ...
    dns1: ...
    dns2: ...

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "espino2-out"

##### LOCAL WEB SERVER #####
web_server:
  local: true

##### PERIPHERALS #####

### I2C bus for the BME680 devices
i2c:
  - id: "i2cbus_bme"
    sda: GPIO26
    scl: GPIO25
    scan: true

### BME680 device #1 using BSEC library
bme680_bsec:
  id: bme680_esterno
  i2c_id: "i2cbus_bme"
  address: 0x76
  sample_rate: ulp
  temperature_offset: 0.5

### Sensors list
sensor:
  # BME680 sensors
  - platform: bme680_bsec
    temperature:
      id: "espino2out_Temperatura"
      name: "espino2out_Temperatura"
    pressure:
      id: "espino2out_Pressione"
      name: "espino2out_Pressione"
    humidity:
      id: "espino2out_RH"
      name: "espino2out_RH"
    iaq:
      id: "espino2out_IAQ"
      name: "espino2out_IAQ"
    co2_equivalent:
      id: "espino2out_CO2eq"
      name: "espino2out_CO2eq"
    breath_voc_equivalent:
      id: "espino2out_VOCeq"
      name: "espino2out_VOCeq"

  # Wifi sensors
  - platform: wifi_signal
    name: "espino2out_WifiLevel"

text_sensor:
  # BME680 text sensors
  - platform: bme680_bsec
    iaq_accuracy:
      id: "espino2out_IAQaccuracy"
      name: "espino2out_IAQaccuracy"

binary_sensor:
  # BME680 status sensor
  - platform: template
    name: "espino2out_Status"
    # on means problem detected, off means no problem (OK)
    # Reference: https://www.home-assistant.io/integrations/binary_sensor/#device-class
    device_class: problem
    lambda: |-
      if (id(bme680_esterno).is_failed() || id(bme680_esterno).status_has_warning() || id(bme680_esterno).status_has_error()) {
        // Problem detected.
        // Return "has problem" status
        return true;
      } else {
        // All is OK: return "no problem" status
        return false;
      }
    on_press:
      then:
        - lambda: |-
            // Set all sensors to unknown value
            id(espino2out_Temperatura).publish_state(NAN);
            id(espino2out_Pressione).publish_state(NAN);
            id(espino2out_RH).publish_state(NAN);
            id(espino2out_IAQ).publish_state(NAN);
            id(espino2out_CO2eq).publish_state(NAN);
            id(espino2out_VOCeq).publish_state(NAN);
            id(espino2out_IAQaccuracy).publish_state("unknown");

```

I already saw the Status LED component, but I didn't like the idea of consuming an additional output only for showing a state in HA. More than that, I have no way to see the LED since the ESP is not visible.
The template binary sensor is perfect for this task.
One minor issue about this is that ESPHome prints a warning about too much time passed in the loop while executing the lambda.

bisbastuner, 2024-02-25:
What I'm thinking about right now is:
is it right that I need to code that YAML to have the sensor values set to `unknown` in HA, or should that behavior be integrated in the bme680_bsec component natively?

I could do that and send a PR if you think that's useful. To me, it seems the right thing to do, because of these words:
_`return NAN;` if the state should be considered invalid to indicate an error_
(from the [Template Sensor documentation](https://esphome.io/components/sensor/template.html))

bisbastuner, 2024-03-13:
@trvrnrth any thoughts / comments about this?

trvrnrth, 2024-03-13:
> @trvrnrth any thoughts / comments about this?

I think I would tend to agree that NAN should be published on comms error. However, this comes with the caveat that I don't think this is how this is handled in any(?) of the other sensor components so I'm not sure if it would be desirable/expected for the behavior here to be different.