esphome/issues #5496: on_boot: different behavior if connected to WiFi or not.

fxmike08, 2024-02-16:
### The problem

### First problem:
Scenario:

1.  When device boot and is able to connect to Wi-Fi the led is turned on (`switch.turn_on: led` from `on_boot` section). 
2. When device boot and is **not** able to connect to Wi-Fi led is always off.

### Second problem:
Is not related to  if connected or not to Wi-Fi.
Scenario:
1. If a business logic depends on the `on_boot` section, I need to set a flag on a global variable to verify if it was executed. However, setting the flag within the `on_boot` section does not produce the expected behavior.

Setting variable within `on_boot` section is not working at all.
#### To test this, first set:
globals:

```
  - id: global_init_knx_state
    type: bool
    initial_value: 'false'
    restore_value: no
```
When changing state of GPIO pin 23 will not turn on/off the led from GPIO pin 15. Which means that setting global variable inside boot is not working.

```
  - id: global_init_knx_state
    type: bool
    initial_value: 'true'
    restore_value: no
```
Doing this will affect the led from GPIO pin 15 when changing state of GPIO pin 23.

:heavy_exclamation_mark:  **Note**:  On older version of ESPHOME I didn't had those problems.

### Which version of ESPHome has the issue?

2023.12.9

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

latest

### What platform are you using?

ESP32

### Board

esp32dev

### Example YAML snippet

```yaml
esphome:
  name: esphome-web-672700
  friendly_name: ESPHomeTest
  on_boot:
    priority: -1000

    then:
      - switch.turn_on: led # first problem here
      - globals.set:
          id: global_init_knx_state
          value: 'true'
globals:
  - id: global_init_knx_state
    type: bool
    initial_value: 'false' #second problem is here
    restore_value: no

esp32:
  board: esp32dev
  framework:
    type: arduino

switch:
  - platform: gpio
    pin:
      number: 15
      mode:
        output: True
        pullup: true
    id: led
    name: "Led"
binary_sensor:
  - platform: gpio
    id: sw
    pin:
      number: 23
      mode:
        input: true
        pullup: True
    name: sw
    on_release: 
      then:
        - lambda: |
            if(global_init_knx_state){
              id(led).toggle();
            }
            return;      
           
web_server:
  port: 80
  
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "*****"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-672700"
    password: "pAArFGt0VQsq"

captive_portal:
```

fxmike08, 2024-02-16:
Upon further investigation, I discovered that adding a delay 2sec at the beginning of the `on_boot` section, allows the setting of a global variable to function properly. However, I observed that certain components, such as `switch.turn_on: led`, do not function as expected in this scenario (when device is unable to connect to Wi-Fi).

ssieb, 2024-02-16:
Your `on_boot` priority is set to being after wifi is connected.  That's your problem.

closed 2024-02-16 by ssieb