esphome/issues #5499: ESP32-CAM: Not possible to change contrast, brightness, saturation

Jpsy, 2024-02-19:
### The problem

Several method calls of the esp32_camera component do not result in any visible change, i.e. (but not complete):
* set_contrast(x); update_camera_parameters();
* set_brightness(x); update_camera_parameters();
* set_saturation(x); update_camera_parameters();

The reason is probably an oversimplified approach of the ESP32-CAM driver that is used by the component.
The central point here is the "SDE indirect register" of the OV2640, which contains a bit mask of enable flags for changes to contrast, brightness, saturation, color tint, negative and others. With each call to the driver's set_contrast / set_brightness / set_... the enable bit for the requested effect is set and all others are zeroed. Thus only the last changed effect remains active and all others are switched off again.

Now, the esp32_camera component's set_xyz() methods do not call the corresponding driver methods at all. 
Instead they store the requested effect strength parameter in an internal status var. 
All these status vars are then written to the OV2640 by a single call to update_camera_parameters(). This fires a series of method calls of the driver (including set_contrast, set_brightness, set_saturation and others). As described, only the last of these wins, which is probably the color tint change, as this seems to work. With this flow the brightness/saturation/contrast will never work.

I see two possible solutions here:

1. Simple but limited solution
Move the set_xyz() calls from update_camera_parameters() directly into the corresponding set_xyz() calls of the ESPHome component. This would at least allow to set all of the effects separately, but not in combination.

2. Take control of the enable flags in SDE indirect register
This would allow for all combinations of effects, but I see several comments on the web, that there are interferences between different effects that might need to be taken care of. I.e. brightness and contrast seem to interfere. But then again, this could also be handled by the user by sending adjusting method calls herself.

Solution 1. seems easy to implement and I think I can file a PR for that.
Solution 2. is beyond my reach. I tried but got stuck because I needed the write_SCCB() method of the driver. This method is currently not available in the ESPHome esp32_camera component and I have no idea how I could get hold of that. If someone could provide me a way to get hold of this method, I could also try to work on 2.

### Which version of ESPHome has the issue?

2023.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.6.1

### What platform are you using?

ESP32

### Board

ESP32-CAM (AI Thinker Clone)

### Component causing the issue

esp32_camera

### Example YAML snippet

```yaml
esp32_camera:
  id: espcam
  ...
  ...

number:
  - platform: template
    name: Kontrast
    id: cam_contrast
    optimistic: true
    initial_value: 0
    min_value: -2
    max_value: 2
    step: 1
    set_action:
      - lambda: |-
          id(espcam).set_contrast(x);
          id(espcam).update_camera_parameters();
  - platform: template
    name: Helligkeit
    id: cam_brightness
    optimistic: true
    initial_value: 0
    min_value: -2
    max_value: 2
    step: 1
    set_action:
      - lambda: |-
          id(espcam).set_brightness(x);
          id(espcam).update_camera_parameters();
  - platform: template
    name: Sättigung
    id: cam_saturation
    optimistic: true
    initial_value: 0
    min_value: -2
    max_value: 2
    step: 1
    set_action:
      - lambda: |-
          id(espcam).set_saturation(x);
          id(espcam).update_camera_parameters();
```

### Additional information

[OV2640 Datasheet v2.2](https://www.uctronics.com/download/OV2640_DS.pdf)  
[OV2640 Software Application Notes](https://www.dragonwake.com/download/camera/ov2640/OV2640%20Camera%20Module%20Software%20Application%20Notes.pdf)

[Information about enable flags in SDE indirect register](https://esp32.com/viewtopic.php?t=14376#p93178)
[Example of handling interference between brightness and contrast settings](https://github.com/espressif/esp32-camera/issues/401#issuecomment-1186605826)

ssieb, 2024-02-19:
Sorry, I had this referred by a question about changing the resolution and I didn't look close enough.  That would be a feature request.  But this does seem to be a valid issue since this should probably work.  However, the analysis doesn't seem to be entirely correct because all the initial settings are set using this same method and those work.  So I don't see why this wouldn't work as well.

ssieb, 2024-02-19:
With further thought, I remember that people have been adjusting these settings from the beginning, so I don't know why it wouldn't be working now.
And @fornellas , Otto is no longer involved with this project.

Jpsy, 2024-02-20:
Hi Samuel,
I don't really know what to respond now.
Changing these settings does not work for me and for others and I could track the reason down to an obvious code problem. I also provided YAML to reproduce the problem. If you "remember that people have been adjusting these settings" and that proves me wrong, then I have no other means left to convince you. Please tell me, what else would be needed.

Rudd-O, 2024-02-21:
Yes, the SDE indirect register bit mask.  If any of the settings controlled by that bit mask is changed, changes to the second or third setting from the same group have no effect.

It's therefore impossible to implement something in ESPHome like the ESP webcam demo (with settings on the left sidebar) that's common to see shipped in ESP devices and in YouTube videos.

ssieb, 2024-02-21:
If you set those values in the config, do they work?

Rudd-O, 2024-02-23:
Hard to tell because the changes are subtle.  But if I run the exact same code (using C++ lambda) that the settings run, no, they don't work.

fornellas, 2024-02-23:
In my tests:

- Changing most settings seem to work (eg: effect, auto-gain).
- Some settings (eg: brightness) seem to not do anything (though I'm not sure if this is the limited effect of the setting itself...).
- Quality and framesize setting were missing, https://github.com/fornellas/esphome/commit/0022486422cd02491c7204fc82f467c19dd0c4a4 should fix that.
  - Quality changes does work.
  - Resolution change does not work, and leads to a crash after any change.

Rudd-O, 2024-04-09:
I will soon try your code.

tonyis49, 2024-05-16:
Many thanks for this. I have spent the last 2 days trying to incorporate the camera into ESPHOME with the same parameters available as with the arduino installation and was about to give up until I found your code. So far,  with a couple of minor tweaks, its works just fine..

corgan2222, 2024-06-17:
> In my tests:
> 
>     * Changing most settings seem to work (eg: effect, auto-gain).
> 
>     * Some settings (eg: brightness) seem to not do anything (though I'm not sure if this is the limited effect of the setting itself...).
> 
>     * Quality and framesize setting were missing, [fornellas/esphome@0022486](https://github.com/fornellas/esphome/commit/0022486422cd02491c7204fc82f467c19dd0c4a4) should fix that.
>       
>       * Quality changes does work.
>       * Resolution change does not work, and leads to a crash after any change.
> 
> 
> PS: for reference, this is the config I'm using which enable various settings to be adjusted at either ESPHome web UI or at Home Assistant:
> 
> ```yaml
> esphome:
>   name: "esp32-cam"
>   friendly_name: "ESP32 Cam"
>   name_add_mac_suffix: true
>   on_boot:
>     - priority: 600.0
>       then:
>         - lambda: |-
>             // Default: Connecting
>             id(wifi_led_pwm).set_level(0.5);
>             id(wifi_led_pwm).set_period(1000);
>             id(wifi_led_pwm).turn_on();
>         - script.execute: fast_reboot_factory_reset
>   on_loop:
>     then:
>       - lambda: |-
>           // Captive portal
>           if(id(wifi_component).has_ap() && !id(wifi_component).has_sta()) {
>             id(wifi_led_pwm).set_level(0.5);
>             id(wifi_led_pwm).set_period(200);
>           } else {
>             id(wifi_led_pwm).set_period(1000);
>             // Connected
>             if(id(wifi_component).is_connected()) {
>               id(wifi_led_pwm).set_level(1);
>               if (id(esp32cam_update)) {
>                 id(esp32cam).update_camera_parameters();
>                 id(esp32cam_update) = false;
>               }
>             // Connecting
>             } else
>               id(wifi_led_pwm).set_level(0.5);
>           }
> 
> esp32:
>   board: esp32cam
> 
> external_components:
>   - source:
>       type: git
>       url: https://github.com/fornellas/esphome
>       ref: fix_esp32_camera_update
>     components: [ esp32_camera ]
>     refresh: 0s
> 
> globals:
>   - id: fast_reboot
>     type: int
>     restore_value: yes
>     initial_value: '0'
> 
>   - id: factory_reset_reboot_counter
>     type: int
>     initial_value: '5'
> 
[... 458 lines cut ...]
>     initial_value: 300
>     min_value: 0
>     max_value: 1200
>     step: 1
>     set_action:
>       - lambda: |-
>           id(esp32cam).set_aec_value(x);
>           id(esp32cam_update) = true;
> 
>   - platform: template
>     name: "Auto gain control value"
>     icon: mdi:auto-fix
>     entity_category: config
>     mode: slider
>     initial_value: 0
>     min_value: 0
>     max_value: 30
>     step: 1
>     set_action:
>       - lambda: |-
>           id(esp32cam).set_aec_value(x);
>           id(esp32cam_update) = true;
> ```

thank you so much for this! Works fine!