esphome/issues #5503: SONOFF S31 bricked after 2024.2.0

zwizard, 2024-02-21:
### The problem

Last night I upgraded to 2024.2.0 and picked two of my S31s to flash. They never came back online. I had to restore from backup and hook up both devices to USB to flash. Any ideas what might have happened or steps to troubleshoot?

### Which version of ESPHome has the issue?

2024.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.2

### What platform are you using?

ESP8266

### Board

SONOFF S31

bvarick, 2024-02-21:
Mine wasn't bricked, but I have issues with the `update_interval` not being recognized after the update. Here's that issue: https://github.com/esphome/issues/issues/5504

The S31 code was changed with this pull: https://github.com/esphome/esphome/pull/6180

x99percent, 2024-02-21:
Had the same issue with a now-bricked S31 as well as an ESP8266 board with a pretty empty YAML. I can try the #5504 fix on the S31, but what could have broken the plain ESP8266?

S31: 
`esphome:
  name: beacon1-outlet
  friendly_name: Beacon1 Outlet

esp8266:
  board: esp01_1m

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "[ENCRYPTION_CODE]"

ota:
  password: "[OTA_PASSWORD]"

wifi:
  ssid: "[SSID]"
  password: "[WIFI_PASSWORD]"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Beacon1-Outlet Fallback Hotspot"
    password: "[HOTSPOT_PASSWORD]"

captive_portal:

web_server:
  port: 80

uart:
  rx_pin: RX
  baud_rate: 4800

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Button"
    on_press:
      - switch.toggle: relay
  - platform: status
    name: "Status"

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s
  - platform: cse7766
    current:
      name: "Current"
      accuracy_decimals: 2
    voltage:
      name: "Voltage"
      accuracy_decimals: 1
    power:
      name: "Power"
      accuracy_decimals: 1
      id: my_power
  - platform: total_daily_energy
    name: "Daily Energy"
    power_id: my_power

switch:
  - platform: gpio
    name: "Relay"
    pin: GPIO12
    id: relay
    restore_mode: ALWAYS_ON
    on_turn_on:
      - light.turn_on: status_led_1
    on_turn_off:
      - light.turn_off: status_led_1

time:
  - platform: sntp
    id: my_time
    update_interval: 12h

light:
  - platform: status_led
    id: status_led_1
    name: "Status LED"
    pin:
      number: GPIO13
      inverted: True`

ESP8266:
`esphome:
  name: "temp-esp8266"
  friendly_name: Temp ESP8266

esp8266:
  board: esp01_1m

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "[ENCRYPTION_CODE]"

ota:
  password: "[OTA_PASSWORD]"

wifi:
  ssid: "[SSID]"
  password: "[WIFI_PASSWORD]"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Temp ESP8266 Fallback Hotspot"
    password: "[HOTSPOT_PASSWORD]"

captive_portal:

web_server:
  port: 80`

DAVe3283, 2024-02-21:
@zwizard did you also have `web_server` enabled in your S31?

Since that is basically the only thing enabled @x99percent 's bricked ESP8266 it implies that might be what is causing the bricking.

zwizard, 2024-02-21:
Yes. I was able to get to one of them over WiFi hotspot. Joined to my WiFi.
Then I tried to uploaded the bin file directly and got:
ERROR[4]: Not Enough Space

jjcarrol, 2024-02-21:
I also have 2 Sonoff S31s that went offline after updating to 2024.2.0.

robwasteground, 2024-02-23:
I've got about 25 Sonoff S31s that I upgraded to 2024.2.0 yesterday. 6 of them are now exhibiting this behavior but the other 19 are working fine. Behavior is pretty weird as @jjcarrol mentions - 3 of the 6 seem to be showing up running their wifi hotspots but they're also connected to the correct wifi network and replying to ping there. Some are allowing me to try uploading a new image that way (then failing with the "Update Failed: ERROR[4]: Not Enough Space" message, and others don't even show the web page but seem to occasionally respond to homeassistant randomly but not very consistently. I actually thought it was my Wifi network until I spotted this issue.

I'll pull the failed ones and flash them via USB to see if that cures it.

roadbikemike, 2024-02-23:
Having the same issue as [bvarick](https://github.com/bvarick).  While my S31 is not bricked, its not using the update interval at all.  Sending mass amounts of data!

DAVe3283, 2024-02-23:
@roadbikemike it sounds like you are experiencing #5504. There is a PR that reverts to the old behavior you can try.

derekpurdy, 2024-02-25:
mine didn't brick but they keep randomly resetting every 5 minutes to an hour.

puuu, 2024-02-26:
Have you read the documentation?

From the [2024.2.0 changelog](https://esphome.io/changelog/2024.2.0.html), the cse7766 component has a breaking change.

It is no longer a pulling sensor, so `update_interval` is no longer available.

As the [update frequency is very high](https://esphome.io/components/sensor/cse7766) you should use the `throttle_average` or `throttle` filter.

I use the following configuration without any problems:
```yaml
sensor:
  - platform: cse7766
    current:
      name: Current
      filters:
        - throttle_average: 60s
    voltage:
      name: Voltage
      filters:
        - throttle_average: 60s
    power:
      name: Power
      filters:
        - throttle_average: 60s
    energy:
      name: Energy
      filters:
        - throttle: 60s
```

Are you sure that your problems are not due to the high refresh rate, as you are not using the `throttle_average` or `throttle` filters?

Yes, I am not happy about the breaking changes either, but the new implementation has some beauty and does it the esphome way.

gadgetdavedenver, 2024-02-26:
Same issue as @x99percent here with: 

- An S31 that will not update (when trying manually OTA get the  "Update Failed: ERROR[4]: Not Enough Space" error). It seems to come back online and is even responding to HA but do seem to fall on and off. For example, one automation had to try 3 times overnight to turn it off. 
- A test esp32-wroom dev board that seemed to brick on the update and I can no longer connect to at all. YAML is below, it's VERY small. 
- In addition, when I try to setup a new esp32-wroom device this morning it gets to the initial install, completes the compile, and then almost immediately times out once the "installing this will take 2 minutes" message appears with the error: "Installation failed: Error: Timeout."  This is via USB plugged into the PC (not the HA machine). This is the initial install with NOTHING in the YAML yet.

YAML for the bricked esp32:
`esphome:
  name: testbench
  friendly_name: TestBench

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: [ENCRYPTION_KEY]

ota:
  password: [OTA_PW]

wifi:
  ssid: [SSID]
  password: [WIFI_PWD]

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Testbench Fallback Hotspot"
    password: [HOTSPOT_PW]

captive_portal:

switch:
  - platform: gpio
    name: "TEST WARNING BUZZER"
    id: warningbuzzer
    pin:
      number: 27
      inverted: true
      mode: 
        output: true
        open_drain: true      
`

gadgetdavedenver, 2024-02-27:
**Update to above** - I did get to take the S31 apart this afternoon and re-flashed it locally and it seems to be back to normal - with the cse7766 working the "new" way (not the fallback). I then did one OTA update to make a small YAML change and it succeeded as well. 

> Same issue as @x99percent here with:
> 
> * An S31 that will not update (when trying manually OTA get the  "Update Failed: ERROR[4]: Not Enough Space" error). It seems to come back online and is even responding to HA but do seem to fall on and off. For example, one automation had to try 3 times overnight to turn it off.  Update also fails after the cse7766 fix is in place.
> * A test esp32-wroom dev board that seemed to brick on the update and I can no longer connect to at all. YAML is below, it's VERY small.
> * In addition, when I try to setup a new esp32-wroom device this morning it gets to the initial install, completes the compile, and then almost immediately times out once the "installing this will take 2 minutes" message appears with the error: "Installation failed: Error: Timeout."  This is via USB plugged into the PC (not the HA machine). This is the initial install with NOTHING in the YAML yet.  **Edit/Update:** I know this new ESP32 device is good because I can flash it with the Arduino IDE just fine, just hot with ESPHome in HA.
> 
> YAML for the bricked esp32:
> 
> ```
> esphome:
>   name: testbench
>   friendly_name: TestBench
> 
> esp32:
>   board: esp32dev
>   framework:
>     type: arduino
> 
> # Enable logging
> logger:
> 
> # Enable Home Assistant API
> api:
>   encryption:
>     key: [ENCRYPTION_KEY]
> 
> ota:
>   password: [OTA_PW]
> 
> wifi:
>   ssid: [SSID]
>   password: [WIFI_PWD]
> 
>   # Enable fallback hotspot (captive portal) in case wifi connection fails
>   ap:
>     ssid: "Testbench Fallback Hotspot"
>     password: [HOTSPOT_PW]
> 
> captive_portal:
> 
> switch:
>   - platform: gpio
>     name: "TEST WARNING BUZZER"
>     id: warningbuzzer
>     pin:
>       number: 27
>       inverted: true
>       mode: 
>         output: true
>         open_drain: true      
> ```

solideus, 2024-03-02:
Fully configuration of UART parameters works for me, without external_components:

```
uart:
  rx_pin: GPIO16
  baud_rate: 4800
  id: uart_cse
  data_bits: 8
  parity: EVEN
  stop_bits: 1
``` 
SONOFF POW ORIGIN

DAVe3283, 2024-03-04:
I have noticed the firmware size spiked up dramatically in 2024.2.0 compared to previous versions, but I have been able to produce smaller binaries to recover devices by setting the log level to DEBUG or lower (INFO is a good choice too).

Then I flash a new firmware with just `ota` (not even `api`), let it reboot and respond to pings, and I can usually put the normal firmware back on at that stage. Saves having to open the device back up to reflash.

I am moderately curious why the core firmware components seem to have gotten so much larger in this release, but not enough to actually bisect the changes. It is a bit frustrating on ESP8266 based devices with small amounts of flash for sure.

dlasher, 2024-03-04:
I've spent the last 4h trying to get a single S31 back to pre-upgrade state, and still failing. Upgrading from the webUI is impossible once you cross some boundary of file size (which is crazy, considering I've verified we have a 4M flash).

It appears however, that using @DAVe3283 's template and the pr#6265 fixes, ONLY VIA USB/Serial download, at least I have now recovered 1 of 8. Time to pull the rest apart and serial flash I guess.. The webUI is buggered for firmware upgrades for now if you failed already.

DAVe3283, 2024-03-04:
If you are trying to load an older firmware, when you load the basic OTA-only firmware, be sure to select version 1, so previous versions of ESPHome can upload to it. For example:

```yaml
esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  comment: Sonoff S31 Smart Outlet

esp8266:
  board: esp01_1m
  restore_from_flash: true

logger:
  level: INFO

ota:
  version: 1 # Enables downgrading firmware later

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

I have also been unable to get ESPHome to use more than 1MB of flash on the Sonoff (or any other ESP8266), even though we know recent models of the Sonoff S31 have 4MB. But that is a separate issue.

dlasher, 2024-03-04:
> If you are trying to load an older firmware, when you load the basic OTA-only firmware, be sure to select version 1, so previous versions of ESPHome can upload to it.
<SNIP>
> And for what it is worth, the final version of cse7766 ([esphome/esphome#6292](https://github.com/esphome/esphome/pull/6292) will likely be included in ESPHome 2024.3.0) is now working fine for me, so you could try using pr#6292 to get the energy monitoring working again until that happens.

Thank you for both things. For now I'm just going to be happy to get all the plugs back and working again.. Lesson learned.. 50 successful upgrades doesn't mean the next one will be. I'll go back to doing them one at a time.... (//grumbles off to find out out to disable "upgrade all" in the webUI//)

dlasher, 2024-03-04:
Here's my composite template - 4/8 functional, will flash the rest and monitor over the next few days.

```
substitutions:
  name: s31-NUMBER
  friendly_name: S31-NUMBER
  update_interval: 30s
  relay_restore_mode: RESTORE_DEFAULT_ON

packages:
  s31:
    url: https://PigLab.ReaperLegion.net/home-automation/hass/esphome/packages/sonoff.git
    file: S31-4MB.yaml

# Fix for https://github.com/esphome/issues/issues/5501 and 5504
external_components:
  - source: github://pr#6265
    components: [ cse7766 ]

captive_portal:

web_server:
  port: 80

# Enable Home Assistant API
api:
  encryption:
    key: "key"

ota:
  password: "password"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "S31-NUMBER Fallback Hotspot"
    password: "fallback-password-here"
```

DAVe3283, 2024-03-04:
Do you need `web_server`? That seems to be a common factor in misbehaving S31s on this release. You might try removing that and see if you can get the others to start working right. And if so, that is good information trying to get to the root cause of this.

dlasher, 2024-03-04:
> Do you need `web_server`? That seems to be a common factor in misbehaving S31s on this release. You might try removing that and see if you can get the others to start working right. And if so, that is good information trying to get to the root cause of this.

w/o web server, port 80 never opens, and I can't check status, etc - First round of upgrades I left that out, had to start over again.

zwizard, 2024-03-06:
I see that 2024.2.2 is out. Does anyone know if this has been fixed in the newer version?

dlasher, 2024-03-07:
> I see that 2024.2.2 is out. Does anyone know if this has been fixed in the newer version?

I see nothing in the notes that indicates it has been. So if yes, not mentioned. (Was looking for cse7766)

robwasteground, 2024-03-07:
https://github.com/esphome/esphome/pull/6286 was included in 2024.2.2 per the release notes, and seems to have fixed things for me. I was able to remove the pr#6265 and flash via OTA to 2024.2.2 on all of my S31s. One of them refused to flash OTA and had to be pulled and flashed manually, but after that, seems that they're all working fine for me now.

ssieb, 2024-03-08:
You should be able to power cycle it until goes to safe mode where you can OTA update it.  Unless you haven't enabled saving to flash.

jjcarrol, 2024-03-13:
> [esphome/esphome#6286](https://github.com/esphome/esphome/pull/6286) was included in 2024.2.2 per the release notes, and seems to have fixed things for me. I was able to remove the pr#6265 and flash via OTA to 2024.2.2 on all of my S31s. One of them refused to flash OTA and had to be pulled and flashed manually, but after that, seems that they're all working fine for me now.

Care to share your code? I just tried to update to 2024.2.2 on one of my Sonoff S31s and it's doing the same thing as with the 2024.2.0 update. Not working properly at all.

zwizard, 2024-03-13:
I think it's pretty basic.

`esphome:
  name: hydrapump

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xRKRO8uDaoyFV3okvyqcmTDUysAtzfKid+vFTk903dk="

ota:
  password: "2e042b364950506fff1b49e6a64e574b"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hydrapump Fallback Hotspot"
    password: "JZGc0zZUCNFS"

captive_portal:
    
# Device Specific Config

uart:
  rx_pin: RX
  baud_rate: 4800

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "HydraPump Button"
    on_press:
      - switch.toggle: relay
  - platform: status
    name: "HydraPump Status"

sensor:
  - platform: wifi_signal
    name: "HydraPump WiFi Signal"
    update_interval: 60s
  - platform: cse7766
    current:
      name: "HydraPump Current"
      accuracy_decimals: 1
    voltage:
      name: "HydraPump Voltage"
      accuracy_decimals: 1
    power:
      name: "HydraPump Power"
      accuracy_decimals: 1
      id: my_power
  - platform: total_daily_energy
    name: "HydraPump Daily Energy"
    power_id: my_power

switch:
  - platform: gpio
    name: "HydraPump"
    pin: GPIO12
    id: relay
    restore_mode: ALWAYS_ON

time:
  - platform: sntp
    id: my_time

status_led:
  pin: GPIO13`

zwizard, 2024-03-13:
It's pretty basic I think. 

`esphome:
  name: hydrapump

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xRKRO8uDaoyFV3okvyqcmTDUysAtzfKid+vFTk903dk="

ota:
  password: "2e042b364950506fff1b49e6a64e574b"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hydrapump Fallback Hotspot"
    password: "JZGc0zZUCNFS"

captive_portal:
    
# Device Specific Config

uart:
  rx_pin: RX
  baud_rate: 4800

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "HydraPump Button"
    on_press:
      - switch.toggle: relay
  - platform: status
    name: "HydraPump Status"

sensor:
  - platform: wifi_signal
    name: "HydraPump WiFi Signal"
    update_interval: 60s
  - platform: cse7766
    current:
      name: "HydraPump Current"
      accuracy_decimals: 1
    voltage:
      name: "HydraPump Voltage"
      accuracy_decimals: 1
    power:
      name: "HydraPump Power"
      accuracy_decimals: 1
      id: my_power
  - platform: total_daily_energy
    name: "HydraPump Daily Energy"
    power_id: my_power

switch:
  - platform: gpio
    name: "HydraPump"
    pin: GPIO12
    id: relay
    restore_mode: ALWAYS_ON

time:
  - platform: sntp
    id: my_time

status_led:
  pin: GPIO13`

ssieb, 2024-03-13:
@zwizard are you aware of the issue here?  You didn't add any filters, so that's going to cause the problem.

ssieb, 2024-03-13:
But there was actually a full example pasted in here 2 weeks ago...
https://github.com/esphome/issues/issues/5503#issuecomment-1964142481

zwizard, 2024-03-14:
That did the trick. Thanks! Sorry for not reading the comments more.

closed 2024-03-14 by zwizard