esphome/issues #5504: [update_interval] is an invalid option for [sensor.cse7766]. Please check the indentation.

bvkrieken, 2024-02-21:
### The problem

After receiving the latest update to the ESPHome add-on for Home assistant I get the following error when I try to update an athom-smart-plug-v2 device. The error is:

[update_interval] is an invalid option for [sensor.cse7766]. Please check the indentation.

### Which version of ESPHome has the issue?

2024.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.2

### What platform are you using?

ESP8266

### Board

athom-smart-plug-v2

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.2.0
INFO Reading configuration /config/esphome/athom-smart-plug-v2-f1971b.yaml...
INFO Detected timezone 'Europe/Amsterdam'
Failed config

sensor.cse7766: [source /data/packages/43f53298/athom-smart-plug-v2.yaml:86]
  platform: cse7766
  
  [update_interval] is an invalid option for [sensor.cse7766]. Please check the indentation.
  update_interval: 10s
  current: 
    name: Current
    filters: 
      - lambda: if (x < 0.060) return 0.0; else return x;
  voltage: 
    name: Voltage
  power: 
    name: Power
    id: power_sensor
    filters: 
      - lambda: if (x < 3.0) return 0.0; else return x;
  energy: 
    name: Energy
    id: energy
    unit_of_measurement: kWh
    filters: 
      - multiply: 0.001
    on_value: 
      then: 
        - lambda: |-
            static float previous_energy_value = 0.0;
            float current_energy_value = id(energy).state;
            id(total_energy) += current_energy_value - previous_energy_value;
            previous_energy_value = current_energy_value;
```

bvarick, 2024-02-21:
I have the same issue. I didn't have update_interval specified before and it defaulted to 60s. After the update, its updating every 1s. I tried to change it by specifying  `update_interval: 60s` and it gave me this error.

bvarick, 2024-02-21:
This was the change for cse7766 that was included in 2024.2.0: https://github.com/esphome/esphome/pull/6180

DAVe3283, 2024-02-21:
You need to add a filter on the output sensors to avoid the insane update speed in 2024.2.0. I recommend also pulling in my fix for the energy sensor until that gets approved/merged/released:

```yaml
substitutions:
  update_interval: 30s

external_components:
  - source: github://pr#6261
    components: [ cse7766 ]

sensor:
  - platform: cse7766
    current:
      name: Current
      id: current
      filters:
        - throttle_average: ${update_interval}
    voltage:
      name: Voltage
      id: voltage
      filters:
        - throttle_average: ${update_interval}
    power:
      name: Power
      id: power
      filters:
        - throttle_average: ${update_interval}
    energy:
      name: Energy
      filters:
        - throttle: ${update_interval}
```

jwpine, 2024-02-21:
same issue on multiple devices. Was this not thoroughly tested?

DAVe3283, 2024-02-21:
IMO a better solution to this due to other side effects (e.g. https://github.com/esphome/issues/issues/5501#issuecomment-1957720730) is to revert to the old behavior. I submitted #6265 to do that, and you can use that until it gets merged/released:

```yaml
external_components:
  - source: github://pr#6265
    components: [ cse7766 ]

# Continue to use your previous cse7766 setup just like before the 2024.2.0 update!
```

bvkrieken, 2024-02-22:
And where would I change this in the HASS add-on interface?

Jimmythebib, 2024-02-22:
Hi @bvkrieken noob here offering my two cents. I pasted the YAML code above suggested by @DAVe3283 into my ESPHome device by going to the ESPHome page on my HA admin panel and clicking the edit link below the respective device. I had access to the YAML config there and added the lines below the packages definition indenting accordingly. I clicked Save and then Install which took a couple of mins.
The issue went away.

bvkrieken, 2024-02-22:
No joy, I get this error when I do that:

INFO ESPHome 2024.2.0
INFO Reading configuration /config/esphome/athom-smart-plug-v2-f1971b.yaml...
ERROR Error while reading config: Invalid YAML syntax:

mapping values are not allowed here
  in "/config/esphome/athom-smart-plug-v2-f1971b.yaml", line 7, column 15

It has a problem with the # character in this line: 
- source: github://pr#6265

DAVe3283, 2024-02-22:
That is odd, as it works for most of us. You could try putting it in quotes:

```yaml
external_components:
  - source: "github://pr#6261"
    components: [ cse7766 ]
```

Here is my entire file for a Sonoff S31 that works fine on 2024.2.0:

```yaml
substitutions:
  name: garage-extension-cord
  friendly_name: Garage Extension Cord
  update_interval: 30s
  relay_restore_mode: RESTORE_DEFAULT_OFF

packages:
  mdns: !include common/disable-mdns.yaml # Disable mDNS (it can't traverse from my IoT network anyway)
  s31:
    url: https://PigLab.ReaperLegion.net/home-automation/hass/esphome/packages/sonoff.git
    file: S31-4MB.yaml

# Fix for https://github.com/esphome/issues/issues/5501 and 5504
external_components:
  - source: github://pr#6265
    components: [ cse7766 ]

# Credentials
api:
  encryption:
    key: !secret garage_extension_cord_key

ota:
  password: !secret garage_extension_cord_ota

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret domain
  ap:
    password: !secret garage_extension_cord_ota
```

When I get weird compile errors, I usually hit the little 3-dot dropdown on the device and say Clean Build Files, then it usually works fine.

bvkrieken, 2024-02-22:
That seems to have fixed it, thanks a lot! 🙏

jwpine, 2024-02-24:
Is there a place to see when the next update comes out that rolls back the cause of this issue?

DAVe3283, 2024-02-24:
If the proposed rollback works for you, leave a comment or thumbs up on the PR (espome/esphome#6265) so the maintainers know it works right and people want it.

solideus, 2024-03-02:
Fully configuration of UART parameters works for me, without external_components:

```
uart:
  rx_pin: GPIO16
  baud_rate: 4800
  id: uart_cse
  data_bits: 8
  parity: EVEN
  stop_bits: 1
```

calas80, 2024-03-06:
The solution is this:

```
uart:
  rx_pin: RX
  baud_rate: 4800
  data_bits: 8
  parity: NONE
  stop_bits: 1
```

The important thing is **parity: NONE**. 
after setting the UART like this, the filters:
```
filters:
  - throttle_average: 10s
```
 work

DAVe3283, 2024-03-07:
Since 2024.2.2, this has worked as expected for me without changing any UART settings. Just remove the single `update_interval` and apply a `throttle` or `throttle_average` filter to each of the outputs.

antokara, 2024-03-07:
For me, the power (Watts), prior to 2024.x (i.e. with 2023.12.9) was reporting 0 when my devices were idle/standby but now, I get like ~1.82 W or ~3 W constantly when idle/standby.

I don't really know which one is the correct value. Before or now?
The throttle/update_interval, did not really change anything in regards to that...

gzcwnk, 2024-04-03:
Trying to flash my Athom plugs and I am still getting this with newer esphome.

==========
INFO ESPHome 2024.3.1
INFO Reading configuration /config/esphome/athom-plug-wanhao-d6.yaml...
INFO Detected timezone 'Pacific/Auckland'
Failed config

sensor.cse7766: [source /config/esphome/athom-plug-wanhao-d6.yaml:85]
  platform: cse7766
  
  [update_interval] is an invalid option for [sensor.cse7766]. Please check the indentation.
  update_interval: 10s
  current: 
    name: athom-plug-wanhaod6 Current
    filters: 
      - lambda: if (x < 0.060) return 0; else return x;
  voltage: 
    name: athom-plug-wanhaod6 Voltage
  power: 
    name: athom-plug-wanhaod6 Power
    id: power_sensor
    filters: 
      - lambda: if (x < 3.0) return 0; else return x;
  energy: 
    name: athom-plug-wanhaod6 Energy
    unit_of_measurement: kWh
    filters: 
      - multiply: 0.001
      -

============

ssieb, 2024-04-03:
@gzcwnk that's not going to change.  You need to change your config.

closed 2024-04-03 by ssieb

gzcwnk, 2024-04-03:
My code for the smarttv2 plug.

========
substitutions:
  device_name: "athom-plug-wanhoad6"
  friendly_name: "athom-plug-wanhoad6"

esphome:
  name: "athom-plug-wanhoad6"
preferences:
  flash_write_interval: 1min
  
esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API hh
api:
  encryption:
    key: "xxxxxxxxxxx"

ota:
#  password: "xxxxxxxxx"
   password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  manual_ip:
    static_ip: 192.168.31.170
    gateway: 192.168.31.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "athom-plug-wanhoad6"
    password: "xxxxxxx"

captive_portal:

time:
  - platform: sntp
    
    
    
web_server:

uart:
  rx_pin: RX
  baud_rate: 4800

#switch:
#  - platform: gpio
#    name: "Ender3 relay"
#    pin: GPIO12
#    id: relay
#    restore_mode: "restore_default_on"

binary_sensor:
  - platform: status
    name: "${friendly_name} Status"

  - platform: gpio
    pin:
      number: 5
      mode: INPUT_PULLUP
      inverted: true
    name: "${friendly_name} Power Button"
    disabled_by_default: true
    on_multi_click:
      - timing:
          - ON for at most 1s
          - OFF for at least 0.2s
        then:
          - switch.toggle: relay
      - timing:
          - ON for at least 4s
        then:
          - button.press: restart_button

sensor:
  - platform: uptime
    name: "${friendly_name} Uptime Sensor"
  - platform: cse7766
    update_interval: 10s
    current:
      name: "${friendly_name} Current"
      filters:
          - lambda: if (x < 0.060) return 0; else return x;   #For the chip will report less than 3w power when no load is connected

    voltage:
      name: "${friendly_name} Voltage"
    power:
      name: "${friendly_name} Power"
      id: power_sensor
      filters:
          - lambda: if (x < 3.0) return 0; else return x;    #For the chip will report less than 3w power when no load is connected

    energy:
      name: "${friendly_name} Energy"
      unit_of_measurement: kWh
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001

  - platform: total_daily_energy
    name: "${friendly_name} Total Daily Energy"
    restore: true
    #preferences:
    #  flash_write_interval: 1min 
    power_id: power_sensor
    unit_of_measurement: kWh
    accuracy_decimals: 2
    filters:
      - multiply: 0.001
     

button:
  - platform: restart
    id: restart_button
    name: "${friendly_name} Restart"

switch:
  - platform: gpio
    name: "${friendly_name}"
    pin: GPIO12
    id: relay
#    restore_mode: ${relay_restore_mode}
    restore_mode: "restore_default_on"

light:
  - platform: status_led
    name: "${friendly_name} Status LED"
    id: blue_led
    disabled_by_default: true
    pin:
      inverted: true
      number: GPIO13

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} IP Address"
      disabled_by_default: true

=======

gzcwnk, 2024-04-03:
> @gzcwnk that's not going to change. You need to change your config.

How do I do that?  I have no idea on coding

ssieb, 2024-04-03:
https://esphome.io/changelog/2024.2.0#cse7766-changes
The example in the [docs](https://esphome.io/components/sensor/cse7766.html) has been updated

DAVe3283, 2024-04-03:
@gzcwnk the documentation has been updated with an example: https://esphome.io/components/sensor/cse7766

Basically just replace your cse7766 section with the example one, and it should work again.

You could also go download the latest Athom config and just use that: https://github.com/athom-tech/athom-configs