esphome/issues #5505: Platform not found : sensor.bme280

josephhungkk, 2024-02-21:
### The problem

After ungrading to ESPHOME v2024.2.0, the yaml which compile successfully before become failed.

INFO ESPHome 2024.2.0
INFO Reading configuration /config/esphome/all-in-one-sensor-fddad0.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

sensor.bme280: [source /config/esphome/packages/sensor.yaml:130]
  
  Platform not found: 'sensor.bme280'.
  platform: bme280
  temperature: 
    name: Temperature
    oversampling: 16x
    id: bme280_temperature
  pressure: 
    name: Pressure
    oversampling: 16x
    id: bme280_pressure
  humidity: 
    name: Humidity
    oversampling: 16x
    id: bme280_humidity
  **address: 118** (the address is differnet from my config file)
  iir_filter: 16x
  update_interval: 15s

### Which version of ESPHome has the issue?

2024.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.2

### What platform are you using?

ESP32-IDF

### Board

ESP32

### Component causing the issue

BME280

### Example YAML snippet

```yaml
sensor:
  - platform: bme280
    temperature:
      name: "Temperature"
      oversampling: 16x
      id: bme280_temperature
    pressure:
      name: "Pressure"
      oversampling: 16x
      id: bme280_pressure
    humidity:
      name: "Humidity"
      oversampling: 16x
      id: bme280_humidity
    address: 0x76
    iir_filter: 16x
    update_interval: 15s
```

### Anything in the logs that might be useful for us?

```txt
Failed config

sensor.bme280: [source /config/esphome/packages/sensor.yaml:130]
  
  Platform not found: 'sensor.bme280'.
  platform: bme280
  temperature: 
    name: Temperature
    oversampling: 16x
    id: bme280_temperature
  pressure: 
    name: Pressure
    oversampling: 16x
    id: bme280_pressure
  humidity: 
    name: Humidity
    oversampling: 16x
    id: bme280_humidity
  address: 118
  iir_filter: 16x
  update_interval: 15s
```

kkalitowski, 2024-02-21:
@josephhungkk there was a breaking change in this area, see https://github.com/esphome/esphome/pull/5538. I believe you just need to change the platform name in your YAML.

randybb, 2024-02-21:
Yep, there is a breaking change, just change bme280 to bme280_i2c

closed 2024-02-21 by randybb