esphome/issues #5507: Fingerprint Grow cause that ESP32 "Can't connect to ESPHome API" after upgrade to 2024.2.0

filipotava, 2024-02-21:
### The problem

When I remove "fingerprint_grow:" ESP connects via API as usual.
When I add "fingerprint_grow" to the yaml it stop working and show error that API can't connect. 
It is possible to upload new compile foimware via OTA, but apart from this it doesn't work

### Which version of ESPHome has the issue?

2024.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.2

### What platform are you using?

ESP8266

### Board

ESP32

### Component causing the issue

fingerprint_grow

### Example YAML snippet

```yaml
uart:
  rx_pin: GPIO12
  tx_pin: GPIO14
  baud_rate: 57600

fingerprint_grow:

api:
```

### Anything in the logs that might be useful for us?

```txt
log:
INFO ESPHome 2024.2.0
INFO Reading configuration /config/esphome/fingerprint-scanner.yaml...
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.xxx.xxx using esphome API
WARNING Can't connect to ESPHome API for fingerprint-scanner-cap @ 192.168.xxx.xxx: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.xxx.xxx', port=6053))]: [Errno 111] Connect call failed ('192.168.xxx.xxx', 6053) (SocketAPIError)
INFO Trying to connect to fingerprint-scanner-cap @ 192.168.xxx.xxx in the background
```

alior101, 2024-02-22:
```
[C][wifi:426]:   Gateway: 192.168.1.1
[C][wifi:427]:   DNS1: 192.168.1.1
[C][wifi:428]:   DNS2: 0.0.0.0
[C][logger:447]: Logger:
[C][logger:448]:   Level: DEBUG
[C][logger:449]:   Log Baud Rate: 115200
[C][logger:451]:   Hardware UART: UART0
[C][uart.arduino_esp32:137]: UART Bus 1:
[C][uart.arduino_esp32:138]:   TX Pin: GPIO21
[C][uart.arduino_esp32:139]:   RX Pin: GPIO22
[C][uart.arduino_esp32:141]:   RX Buffer Size: 256
[C][uart.arduino_esp32:143]:   Baud Rate: 57600 baud
[C][uart.arduino_esp32:144]:   Data Bits: 8
[C][uart.arduino_esp32:145]:   Parity: NONE
[C][uart.arduino_esp32:146]:   Stop bits: 1
[C][fingerprint_grow:443]: GROW_FINGERPRINT_READER:
[C][fingerprint_grow:444]:   System Identifier Code: 0x0000
[C][fingerprint_grow:446]:   Touch Sensing Pin: GPIO12
[C][fingerprint_grow:447]:   Update Interval: 0.5s
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40170887  PS      : 0x00060330  A0      : 0x800d91ea  A1      : 0x3ffb2740
A2      : 0x00000000  A3      : 0x3f401976  A4      : 0x000001bf  A5      : 0x3f4019f2
A6      : 0x00000000  A7      : 0x3fe00000  A8      : 0x800e1451  A9      : 0x3ffb26d0
A10     : 0x00000004  A11     : 0x3f401976  A12     : 0x000001bf  A13     : 0x3f4019f2
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000024  LBEG    : 0x40089a79  LEND    : 0x40089a89  LCOUNT  : 0xfffffff0

Backtrace:0x40170884:0x3ffb27400x400d91e7:0x3ffb2760 0x40170b91:0x3ffb27b0 0x400dfc09:0x3ffb27d0 0x400e1f22:0x3ffb2800 0x400f0429:0x3ffb2820

```

alexborro, 2024-02-22:
Could you please post your complete yaml file ? And you also mentioned you are using an ESP8266 but the platform is ESP32.. that is confusing. What hardware are you using ??

alior101, 2024-02-22:
I'm not the OP. My setup is on a esp32 

```
# enrollment is done with 
# service: esphome.finger_enroll
# data:
#   finger_id: 0
#   num_scans: 2

esphome:
  name: finger
  friendly_name: finger

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "Vlvrat3Kuc0TBugMEN8m6hIT9xHa1wcyLUj4Pd5cKZg="
  services:
  - service: enroll
    variables:
      finger_id: int
      num_scans: int
    then:
      - fingerprint_grow.enroll:
          finger_id: !lambda 'return finger_id;'
          num_scans: !lambda 'return num_scans;'
  - service: cancel_enroll
    then:
      - fingerprint_grow.cancel_enroll:
  - service: delete
    variables:
      finger_id: int
    then:
      - fingerprint_grow.delete:
          finger_id: !lambda 'return finger_id;'
  - service: delete_all
    then:
      - fingerprint_grow.delete_all:

ota:
  password: "364061c0c8cb129ef02f3b1aaa896275"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Finger Fallback Hotspot"
    password: "rVx97NlFDXWD"

captive_portal:

uart:
  rx_pin: GPIO22
  tx_pin: GPIO21
  baud_rate: 57600

fingerprint_grow:
  sensing_pin: GPIO12
  on_finger_scan_matched:
    - homeassistant.event:
        event: esphome.test_node_finger_scan_matched
        data:
          finger_id: !lambda 'return finger_id;'
          confidence: !lambda 'return confidence;'
    - fingerprint_grow.aura_led_control:
        state: BREATHING
        speed: 200
        color: BLUE
        count: 1
    - homeassistant.service:  
        service: button.press
        data_template:
          entity_id: 'button.esphome_web_1a6505_open_lock'
          #target: entity_id: button.esphome_web_1a6505_open_lock

  on_finger_scan_unmatched:
    - homeassistant.event:
        event: esphome.test_node_finger_scan_unmatched
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: RED
        count: 2
  on_enrollment_scan:
    - homeassistant.event:
        event: esphome.test_node_enrollment_scan
        data:
          finger_id: !lambda 'return finger_id;'
          scan_num: !lambda 'return scan_num;'
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: BLUE
        count: 2
  on_enrollment_done:
    - homeassistant.event:
        event: esphome.test_node_enrollment_done
        data:
          finger_id: !lambda 'return finger_id;'
    - fingerprint_grow.aura_led_control:
        state: BREATHING
        speed: 100
        color: BLUE
        count: 2          
  on_enrollment_failed:
    - homeassistant.event:
        event: esphome.test_node_enrollment_failed
        data:
          finger_id: !lambda 'return finger_id;'
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: RED
        count: 4
```

filipotava, 2024-02-23:
> Could you please post your complete yaml file ? And you also mentioned you are using an ESP8266 but the platform is ESP32.. that is confusing. What hardware are you using ??

sorry, I entered wrong description. I'm running it on ESP32.

I've removed all further settings from fingerprint_grow component and it still causes the error¨
here is the yaml:

esphome:
  name: fingerprint-scanner-cap

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: very_verbose
  baud_rate: 0

ota:
  password: !secret ota_password
  on_error:
    then:
      - logger.log:
          format: "OTA update error %d"
          args: ["x"]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on
  power_save_mode: none
  manual_ip:
    static_ip: !secret wifi_ip_fingerprint_scanner_cap
    gateway: !secret wifi_gateway
    subnet: !secret wifi_subnet

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fingerprint Fallback Hotspot"
    password:  !secret ap_wifi_password

captive_portal:

api:
  # Enable Home Assistant API
  encryption:
    key:  !secret api_passkey

uart:
  rx_pin: GPIO12
  tx_pin: GPIO14
  baud_rate: 57600

fingerprint_grow:

output:
  - platform: ledc
    id: red_led
    pin: GPIO1
    inverted: True

light:
  - platform: monochromatic
    id: status_led
    name: "Fingerprint status LED"
    output: red_led
    effects:
      - pulse:
      - pulse:
          name: "Fast Pulse"
          transition_length: 0.5s
          update_interval: 0.5s
      - pulse:
          name: "Slow Pulse"
          # transition_length: 1s      # defaults to 1s
          update_interval: 2s

alexborro, 2024-02-23:
@filipotava , @alior101 
Ok, I have found the problem. A silly and stupid mistake on my side. I will fix it in the next release. Meanwhile, add this to your yaml file, it will workaround the problem.
Sorry the inconvenience.
By the way, next time you need to provide logs, please, if possible, attach the ESP32 to the computer using a USB-to-serial cable instead of WiFi. Through the serial, we can get the initialization logs which is important in many situations, like this one.

Cheers.
Alex.

```
sensor:
  - platform: fingerprint_grow
    fingerprint_count:
      name: "FrontDoor Fingerprint Count"
    last_finger_id:
      name: "FrontDoor Fingerprint Last Finger ID"
    last_confidence:
      name: "FrontDoor Fingerprint Last Confidence"
    status:
      name: "FrontDoor Fingerprint Status"
    capacity:
      name: "FrontDoor Fingerprint Capacity"
    security_level:
      name: "FrontDoor Fingerprint Security Level"
```

alexborro, 2024-02-23:
This issue if fixed by PR [#6116](https://github.com/esphome/esphome/pull/6116).

alior101, 2024-02-23:
@alexborro : esp doesnt segfault now but no finger is detected... I don't even see any log line when I touch the sensor..
Maybe adding the above lines "as-is" is the fault ? should they be changed to soemthing else related to other config params ?

closed 2024-02-23 by filipotava

alior101, 2024-02-23:
Still no go...
When I touch the sensor there is no activity on the log - not even a fail trial ..
My conmplete yaml is attached in previous post. There you can see the GPIO connections..

I'm attaching the first few lines of the serial out - I'm seeing something about misplaced finger - could that be a clue ?
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:359]: Log initialized
[C][ota:483]: There have been 1 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][uart.arduino_esp32:077]: Setting up UART...
[C][fingerprint_grow:060]: Setting up Grow Fingerprint Reader...
[D][fingerprint_grow:232]: Checking password
[D][fingerprint_grow:237]: Password verified
[D][fingerprint_grow:260]: Getting parameters
[D][fingerprint_grow:263]: Got parameters
[D][fingerprint_grow:285]: Getting fingerprint count
[D][fingerprint_grow:288]: Got fingerprint count
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: C8:F0:9E:79:76:50
[D][wifi:459]: Starting scan...
[D][fingerprint_grow:125]: Scan and match
[D][fingerprint_grow:156]: Getting image 1
[D][fingerprint_grow:168]: Finger Misplaced
[D][wifi:474]: Found networks:

```

alexborro, 2024-02-23:
> Still no go... When I touch the sensor there is no activity on the log - not even a fail trial .. My conmplete yaml is attached in previous post. There you can see the GPIO connections..
>
> I'm [...]
I would recommend you to change the pins you are using:
1. The GPIO12 is a strapping pin, used during boot.. it is not the best option for a touch sensing pin. For example, I use GPIO4 for this purpose.
2. GPIO21 and GPIO22 also do not have hardware UART, so it will be forced to use software UART, which sucks.. So I really recommend you to use hw UART, like GPIO16 and GPIO17
This webpage has some nice information about ESP32, specially about GPIO pins that are Ok to use.
https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

alior101, 2024-02-23:
Ok, I changed the detection pin to GPIO04 and the uart to the HW uart on my board (minikit) which are GPIO09/10 
and I think I discovered another bug... checking the password on the sensor is crashing it ... 
```

[I][logger:359]: Log initialized
[C][ota:483]: There have been 9 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][uart.arduino_esp32:077]: Setting up UART...
[C][fingerprint_grow:060]: Setting up Grow Fingerprint Reader...
[D][fingerprint_grow:232]: Checking password
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:359]: Log initialized
[C][ota:483]: There have been 10 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[E][ota:490]: Boot loop detected. Proceeding 
```

alexborro, 2024-02-23:
> Ok, I changed the detection pin to GPIO04 and the uart to the HW uart on my board (minikit) which are GPIO09/10 and I think I discovered another bug... checking the password on the sensor is crashing it ...

The GPIO9/10 are shared with the SPI used to access the ESP32 flash, you must not use it. It is very clear in the documentation (link) I've sent before. 

Use the GPIO16/17 instead.

alior101, 2024-02-23:
after fixing the rx/tx  pins (I looked at the wrong desrcription), the esp doesnt crash anymore but I htink there is still no uart comm with  the grow module and sometimes uart comm with a crash ...
Something has gone bad with the uart ..  

```

and 

```
[C][logger:449]:   Log Baud Rate: 115200
[C][logger:451]:   Hardware UART: UART0
[C][uart.arduino_esp32:137]: UART Bus 1:
[C][uart.arduino_esp32:138]:   TX Pin: GPIO10
[C][uart.arduino_esp32:139]:   RX Pin: GPIO9
[C][uart.arduino_esp32:141]:   RX Buffer Size: 256
[C][uart.arduino_esp32:143]:   Baud Rate: 57600 baud
[C][uart.arduino_esp32:144]:   Data Bits: 8
[C][uart.arduino_esp32:145]:   Parity: NONE
[C][uart.arduino_esp32:146]:   Stop bits: 1
[C][fingerprint_grow:443]: GROW_FINGERPRINT_READER:
[C][fingerprint_grow:444]:   System Identifier Code: 0x0000
[C][fingerprint_grow:446]:   Touch Sensing Pin: GPIO4
[C][fingerprint_grow:447]:   Update Interval: 0.5s
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40170883  PS      : 0x00060330  A0      : 0x800d91ea  A1      : 0x3ffb2740
A2      : 0x00000000  A3      : 0x3f40196e  A4      : 0x000001bf  A5      : 0x3f4019ea
A6      : 0x00000000  A7      : 0x3fe00000  A8      : 0x800e1451  A9      : 0x3ffb26d0
A10     : 0x00000004  A11     : 0x3f40196e  A12     : 0x000001bf  A13     : 0x3f4019ea
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000024  LBEG    : 0x40089a79  LEND    : 0x40089a89  LCOUNT  : 0xfffffff0

Backtrace:0x40170880:0x3ffb27400x400d91e7:0x3ffb2760 0x40170b8d:0x3ffb27b0 0x400dfc09:0x3ffb27d0 0x400e1f22:0x3ffb2800 0x400f0425:0x3ffb2820

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee

```

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:359]: Log initialized
[C][ota:483]: There have been 9 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][uart.arduino_esp32:077]: Setting up UART...
[C][fingerprint_grow:055]: Setting up Grow Fingerprint Reader...
[D][fingerprint_grow:205]: Checking password
[E][fingerprint_grow:437]: No response received from reader
[E][component:113]: Component fingerprint_grow was marked as failed.
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: C8:F0:9E:79:76:50
```

alior101, 2024-02-23:
GPIO 16/17  are UART0 and used by the serial logger - I tried it before using UART1 and got a warning and not working also so I switched to UART1 thinking that comm with the module was disrupted by the serial logger ... Do you use it on your board with serial logging disabled ? 
While being s/w UART, my previous UART assigmenets were working with previous esphome  version (12.23 if im not mistaken ) - it stopped working when upgraded to 2.24 - so I my gut feeling tells me that this issue is not directly connected to s/w vs h/w UART ...

alior101, 2024-02-23:
OK - good news!
Confirming with the below is working. Used GPIO 16/17 which is UART2  and detection pin GPIO04. 
```
# enrollment is done with 
# service: esphome.finger_enroll
# data:
#   finger_id: 0
#   num_scans: 2

esphome:
  name: finger
  friendly_name: finger

esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "Vlvrat3Kuc0TBugMEN8m6hIT9xHa1wcyLUj4Pd5cKZg="
  services:
  - service: enroll
    variables:
      finger_id: int
      num_scans: int
    then:
      - fingerprint_grow.enroll:
          finger_id: !lambda 'return finger_id;'
          num_scans: !lambda 'return num_scans;'
  - service: cancel_enroll
    then:
      - fingerprint_grow.cancel_enroll:
  - service: delete
    variables:
      finger_id: int
    then:
      - fingerprint_grow.delete:
          finger_id: !lambda 'return finger_id;'
  - service: delete_all
    then:
      - fingerprint_grow.delete_all:

ota:
  password: "364061c0c8cb129ef02f3b1aaa896275"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Finger Fallback Hotspot"
    password: "rVx97NlFDXWD"

captive_portal:

uart:
  rx_pin: GPIO16
  tx_pin: GPIO17
  baud_rate: 57600

external_components:
  - source: github://alexborro/esphome@Sleep_mode_rebase
    refresh: 0s
    components: [ fingerprint_grow ]

fingerprint_grow:
  sensing_pin: GPIO04
  on_finger_scan_matched:
    - homeassistant.event:
        event: esphome.test_node_finger_scan_matched
        data:
          finger_id: !lambda 'return finger_id;'
          confidence: !lambda 'return confidence;'
    - fingerprint_grow.aura_led_control:
        state: BREATHING
        speed: 200
        color: BLUE
        count: 1
    - homeassistant.service:  
        service: button.press
        data_template:
          entity_id: 'button.esphome_web_1a6505_open_lock'
          #target: entity_id: button.esphome_web_1a6505_open_lock

  on_finger_scan_unmatched:
    - homeassistant.event:
        event: esphome.test_node_finger_scan_unmatched
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: RED
        count: 2
  on_enrollment_scan:
    - homeassistant.event:
        event: esphome.test_node_enrollment_scan
        data:
          finger_id: !lambda 'return finger_id;'
          scan_num: !lambda 'return scan_num;'
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: BLUE
        count: 2
  on_enrollment_done:
    - homeassistant.event:
        event: esphome.test_node_enrollment_done
        data:
          finger_id: !lambda 'return finger_id;'
    - fingerprint_grow.aura_led_control:
        state: BREATHING
        speed: 100
        color: BLUE
        count: 2          
  on_enrollment_failed:
    - homeassistant.event:
        event: esphome.test_node_enrollment_failed
        data:
          finger_id: !lambda 'return finger_id;'
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: RED
        count: 4

```

alior101, 2024-02-23:
Please update when we can remote the external componenet .
Thanks again!

alexborro, 2024-02-23:
> Please update when we can remote the external componenet . Thanks again!

I'm glad it worked. I hope the PR https://github.com/esphome/esphome/pull/6116 is merged as soon as possible and included in next release.

nicx, 2024-02-27:
@alexborro I have the same behavior... after upgrading esp home the fingerprint sensor is just "dead". no log output, nothing happens when putting a finger on it. the setup works before upgrading for months/years without any problem.

I tried to add the external component, but that didn't help. any further hints what I should do?

Here is my yaml:

```
# Insert your SSID and Your PWD after inital setup
wifi:
  networks:
    - ssid: 'x'          # Enter your WiFi SSID here. Example: `ssid: 'your_network_name'`
      password: 'x'      # Enter your wifi password here. Example: `password: 'abcde123456'`
  #use_address: 192.168.1.10
  manual_ip:
    static_ip: 192.168.1.10
    gateway: 192.168.1.250
    subnet: 255.255.255.0
  ap:
    ssid: ${devicename}

# Enable the captive portal for inital WiFi setup
captive_portal:

substitutions:
  devicename: esp-haustuer
  friendly_name: "ESP Haustür"

esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini

  on_boot:
    priority: -10
    then:
    - wait_until:
        api.connected:
    - logger.log: API is connected!

# Enable logging
logger:
  # level: VERY_VERBOSE
  # level: VERBOSE

# Enable OTA upgrade
ota:
  password: ""

# Sensors
sensor:
  - platform: fingerprint_grow
    fingerprint_count:
      name: "Fingerprint Count"
    last_finger_id:
      name: "Fingerprint Last Finger ID"
    last_confidence:
      name: "Fingerprint Last Confidence"
    status:
      name: "Fingerprint Status"
    capacity:
      name: "Fingerprint Capacity"
    security_level:
      name: "Fingerprint Security Level"

text_sensor:
  - platform: template
    id: fingerprint_state
    name: "Fingerprint State"

binary_sensor:
[... 183 lines cut ...]
    - lambda: 'id(pn532_board).read_mode();'

  # Fingerprint services
  - service: fingerprint_enroll
    variables:
      finger_id: int
      num_scans: int
    then:
      - fingerprint_grow.enroll:
          finger_id: !lambda 'return finger_id;'
          num_scans: !lambda 'return num_scans;'
  - service: fingerprint_cancel_enroll
    then:
      - fingerprint_grow.cancel_enroll:
  - service: fingerprin_delete
    variables:
      finger_id: int
    then:
      - fingerprint_grow.delete:
          finger_id: !lambda 'return finger_id;'
  - service: fingerprint_delete_all
    then:
      - fingerprint_grow.delete_all:

```

nicx, 2024-02-27:
I just downgraded to 2023.12.9 and everything is working again.

alexborro, 2024-02-27:
@nicx , I have a hint about your problem and I guess it is not related to the `fingerprint_grow` component. I think it is related to UART. The pins you are using for UART (GPIO12/14)  doesn't have hardware UART, so it is using software UART - and I think something it not working well in this release with software UART. So you have 2 options:

1- Use the latest ESPHome version and only the `fingerprint_grow` from the release 2023.12.9. That way you can figure it out if the problem is in the `fingerprint_grow` component or not. To test this, update your external component like this:

```
external_components:
  - source: github://esphome/esphome@2023.12.9
    refresh: 0s
    components: [ fingerprint_grow ]
```
It is important to compile this with the latest ESPHome release. Only the fingerprint component will come from 2023.12.9

2- Update your ESP board to a new one. The ESP8266 is very problematic and the board you are using has too few pins for what you want. So time to upgrade to an ESP32. And I would also use the "sensing pin" feature, it improves a lot the sensor robustness and response.

Nadav-K, 2024-03-15:
Hi @alexborro ,

First of all let me say thank you so much for the fantastic work and support in the community - this is very much appreciated!

One question: Is it possible to determine the LED behavior of the fingerprint sensor? e.g. so it's off until the finger is rested on the plate, then only turns on for the scan.

alexborro, 2024-03-17:
> I have a FPM10A sensor (compatible with R307 from what I gather..) and it seems to basically work running over ESP32-C3 super mini. One issue is that randomly, after a finger scan, the sensor seems to be irresponsive for ~12-15 seconds. I've changed the log level to VV - it's observed that after the scan (at 17:40:29) - a status "[V][fingerprint_grow:021]: No touch sensing" is not reported for a long period of time (until 17:40:42)... Do you have any idea what could be the issue?
> 

Hi @Nadav-K . I prepared an external component for testing, adding some debug info so we can try to understand where the "update" function is stuck. 
But I need you to **open a new issue and tag me** on it. I don't like to re-use closed issues for new issues.
Please, in this new issue post your yaml file and the board you are using. As soon as you do it, I will post the external component address, so you can compile and test with it.