esphome/issues #5509: ESP devices have many unsuccessful attempts to connect to Wifi in mesh network

IliasEagle, 2024-02-22:
### The problem

My home network is a mesh set up consisting of 5 TP-Link Deco Access Points. The 2.4GHz and 5GHz channels share the same SSID. 

The problem I am seeing is that some of the 16 devices running ESPHome frequently disconnect from my network and then take from minutes (unusually) to tens of minutes to connect back to the network. 
Looking at the logs on a couple of those devices it can be seen that after each scan the connection is unsuccessful for a different reason each time. 

After each failed connection attempt the device will may try to connect to a different access point many times resulting in connecting to a remote AP with **weaker signal**, as one would expect.

This is an example of only the disconnection reasons (filtered out all the rest). It can be seen that the BSSID is not the same one (the full log: 
```
[W][wifi_esp32:458]:	Event: Disconnected ssid='BuburuzaWifi' bssid=00:31:92:FA:EA:96[redacted] reason='Not Authenticated'
[W][wifi_esp32:458]:	Event: Disconnected ssid='BuburuzaWifi' bssid=00:31:92:FA:EA:96[redacted] reason='Authentication Failed'
[W][wifi_esp32:455]:	Event: Disconnected ssid='BuburuzaWifi' reason='Probe Request Unsuccessful'
[W][wifi_esp32:458]:	Event: Disconnected ssid='BuburuzaWifi' bssid=1C:3B:F3:16:F5:E6[redacted] reason='Authentication Failed'
[W][wifi_esp32:458]:	Event: Disconnected ssid='BuburuzaWifi' bssid=1C:3B:F3:16:F5:E6[redacted] reason='Auth Expired'
[W][wifi_esp32:458]:	Event: Disconnected ssid='BuburuzaWifi' bssid=00:31:92:FA:EB:5E[redacted] reason='4-Way Handshake Timeout'
[W][wifi_esp32:458]:	Event: Disconnected ssid='BuburuzaWifi' bssid=00:31:92:FA:EA:96[redacted] reason='Auth Expired'
[W][wifi_esp32:458]:	Event: Disconnected ssid='BuburuzaWifi' bssid=00:31:92:FA:EA:96[redacted] reason='Association Expired'
[W][wifi_esp32:458]:	Event: Disconnected ssid='BuburuzaWifi' bssid=1C:3B:F3:16:F9:B8[redacted] reason='Auth Expired'
[W][wifi_esp32:458]:	Event: Disconnected ssid='BuburuzaWifi' bssid=1C:3B:F3:16:F9:B8[redacted] reason='Auth Expired'
```

I am suspecting an issue with ESPHome primarily due to the reason that no non-ESPHome device suffers from the same issue. More closely to the ESPHome devices are a few Sonoff switches (with ESP micros), that I have,  with stock firmware that remain connected with no signs of disconnection. 

On some devices I have set static IP in the YAML (as can be seen in the YAML snippet) but all devices' IPs have been assigned a static IP on the mesh system 

### Which version of ESPHome has the issue?

2024.2 recently but the same behaviour existed definitely from 2023.12

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.2

### What platform are you using?

ESP32

### Board

Wemos  ESP32

### Component causing the issue

wifi

### Example YAML snippet

```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.68.143
    gateway: 192.168.68.1
    subnet: 255.255.255.0
```

### Anything in the logs that might be useful for us?

```txt
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
[... 584 lines cut ...]
----
After posting the issue on the home assistant community forum, I was advised to disable fast roaming on the DECO system that as I understand has to do with 802.11r. There has been no change in the behaviour.

----

Not sure how related this is but at the end of the log it can be seen that when I am attempting to perform OTA the module resets mid update so I cannot update the firmware on the first attempt.

```
[256598][D][Updater.cpp:133] begin(): OTA Partition: app1
E (290748) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (290748) task_wdt:  - loopTask (CPU 1)
E (290748) task_wdt: Tasks currently running:
E (290748) task_wdt: CPU 0: IDLE
E (290748) task_wdt: CPU 1: IDLE
E (290748) task_wdt: Aborting.

abort() was called at PC 0x4011c590 on core 0

Backtrace:0x400842c1:0x3ffbebac |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46
```

randybb, 2024-02-22:
Configure your WiFi properly. Basically you need to configure your WiFi based on your clients - disable fast roaming if it is not allowed (only esp32 supports that), disable BSS (the same) and if you have 5GHz for the same SSID then your APs can actively trying to force clients to 5 GHz by denying them to connect to 2.4 GHz - this is totally normal behavior and when clients are only 2.4GHz, then it is making mess. => create a separate SSID for your IoT 2.4 GHz only devices.

IliasEagle, 2024-02-23:
@randybb many thanks for your response. 
As I said, the reason why I suspected ESPHome is that while I have not changed my network setup at least as of 2023.12 I observed this weird behaviour.
On top, as I included in my post, other ESP powered devices don't exhibit the same behaviour so I can't explain why only a few devices would be affected. 

Nevertheless, I tried to follow your advise and separate the two bands. I am not sure how it is done on other systems, unfortunately, on the DECO system you only have a main and a guest network each with their own 2.4 and 5GHz bands. 
The problem I ran into is that devices on the main network and the guest network can't speak to each other out of the box and I didn't find how can this be changed. Nevertheless, for the past hour after having attempted your suggestion and then reverted back to using the main network with 2.4 and 5GHz bands, the ESPhome devices connected relatively quickly. I will continue to observe the behaviour and may re-raise the issue.

Again, many thanks for taking the time to reply to my post. Really appreciate the community!

closed 2024-02-23 by IliasEagle