esphome/issues #5510: Unable to build on ESP-IDF on Home Assistant since 2023.12.8

wtaferner, 2024-02-22:
### The problem

First I was not able to build a working Everything Presence Lite from Lewis although it was and is working with the build done directly from his website (ESP Home) and the integration was successful, despite setting the encryption key for the `api` as it needs to rebuild via HA which was not working from the start.

Now it also happens for an older Everything Presence One which was recently working to build in `2023.12.8`, but now shows the same issue in `2024.02`

I already tried to wipe the cache, reinstall and many more, but it seems not to help to resolve the issue for building in HA.

**HA Versions**

**Core**: 2024.2.2
**Supervisor**: 2024.02.0
**Operating** System: 11.5
**Frontend**: 20240207.1
**ESP Home**: 2024.02

### Which version of ESPHome has the issue?

2023.12.8, 2024.02

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.2

### What platform are you using?

ESP32-IDF

### Board

ESP32-WROOM-32E

### Example YAML snippet

```yaml
substitutions:
  name: everything-presence-lite-xxxxxx
  friendly_name: 'Everything Presence Lite'
packages:
  EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc 
Build flags: -mlongcalls;-Wno-frame-address;
Id flags:  

The output was:
1
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
collect2: error: ld returned 1 exit status

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ 
Build flags: -mlongcalls;-Wno-frame-address;
Id flags:  

The output was:
1
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
collect2: error: ld returned 1 exit status

Determining if the C compiler works failed with the following output:
Change Dir: /data/build/everything-presence-lite-xxxxxx/.pioenvs/everything-presence-lite-xxxxxx/CMakeFiles/CMakeTmp

Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_9dc86 && [1/2] Building C object CMakeFiles/cmTC_9dc86.dir/testCCompiler.c.obj
[2/2] Linking C executable cmTC_9dc86
FAILED: cmTC_9dc86 
: && /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_9dc86.dir/testCCompiler.c.obj  -o cmTC_9dc86   && :
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

MayorBug, 2024-02-23:
Same issue here for me it started with: 2024.02.0

T1ppes, 2024-03-07:
Same here since 2024.02. other ESP Home devices are updating without issue, but the Everything Presence 1 won't compile correctly.

randybb, 2024-03-07:
I don't see there anything specific and I can compile it without issues under the HA Addon and DEV and 2024.2.2.
What happens when you do Clear build files first? And then on what HW is running your HA and what is cpu and memory utilization before compilation?

wtaferner, 2024-03-07:
> I don't see there anything specific and I can compile it without issues under the HA Addon and DEV and 2024.2.2. What happens when you do Clear build files first? And then on what HW is running your HA and what is cpu and memory utilization before compilation?

As stated in the description, none of those basic mitigation measures worked. The HW should be fine, during trying to build my RPi4 has maximum of 60% CPU and 35% of RAM usage, so I think this can be excluded as the reason for failing. 

Here the explicit problematic lib which is downloaded and installed right before it fails (removed the cache directly within the docker container).

[image]

ssieb, 2024-03-07:
Since you know how to get into the container, try running that gcc directly and see what happens.  It would be better if you could come to discord to do this more interactively.

wtaferner, 2024-03-08:
Alright, the reason why ESP-IDF was not possible to build anymore on my home assistant instance was that I was still using a very old image (32 bit) on my RPi4 Model B (4 GB) which was interestingly working without a flaw until this issue came up. For some reason the libs became inconsistent or stopped supporting 32 bit systems and therefore denied to build which was working before without any flaw.

Nevertheless, it was time to upgrade, so I took a new image, installed HA from scratch and restored my backup. Now, ESPHome and specifically ESP-IDF builds are working properly again.

Thx to @ssieb who took the time to track down this specific situation with me. Much appreciated 🙏🏼

closed 2024-03-08 by wtaferner