esphome/issues #5511: Try to create Esphome Bluetooth Proxy on dev board ESP32s (ESP32-WROOM-32) without success on scanning Bluetooth device and BLE

lsteelandt, 2024-02-22:
### The problem

Even I use the Ready Made Project page or by using ESPHome add-on on my Home assistant server, I successfully generate a ESPHome component "Bluetooth Proxy" but it detect no Bluetooth device. 

The bord I use is an ESP32-Wroom-32 Boarded on a NodeMCU 32S [see photo](https://ae01.alicdn.com/kf/Sa54d00cd8fc547438d1da9f27b096b4fg/ESP32-TYPE-C-USB-CH340C-CP2102-WiFi-Bluetooth-Ultra-Faible-Puissance-touristes-Core-ESP32-DevKitC-32.jpg_640x640.jpg_.webp)

I tried with either  framework:  type: esp-idf and type: arduino, but with no result at all. 

I would babe pleased I anybody can tell me what is wrong. 😉 

### Which version of ESPHome has the issue?

2024.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.1

### What platform are you using?

ESP32

### Board

nodemcu 32S v1.1

### Example YAML snippet

```yaml
substitutions:
  name: esp32-bluetooth-proxy-2c540c
  friendly_name: Bluetooth Proxy 2c540c
packages:
  esphome.bluetooth-proxy: github://esphome/firmware/bluetooth-proxy/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: wdWdtyz20L7dWe6JN1UYUAkX6LKRXlDvOgVcxQhjXHs=

esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf

status_led:
  pin: GPIO2

logger:
  level: DEBUG

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
```
[... 266 lines cut ...]
09:30:50.475 -> �[0m
09:30:50.475 -> �[0;35m[C][esp32_ble_tracker:645]: BLE Tracker:�[0m
09:30:50.475 -> �[0;35m[C][esp32_ble_tracker:646]:   Scan Duration: 300 s�[0m
09:30:50.475 -> �[0;35m[C][esp32_ble_tracker:647]:   Scan Interval: 1100.0 ms�[0m
09:30:50.475 -> �[0;35m[C][esp32_ble_tracker:648]:   Scan Window: 1100.0 ms�[0m
09:30:50.475 -> �[0;35m[C][esp32_ble_tracker:649]:   Scan Type: ACTIVE�[0m
09:30:50.475 -> �[0;35m[C][esp32_ble_tracker:650]:   Continuous Scanning: True�[0m
09:30:50.521 -> �[0;35m[C][mdns:115]: mDNS:�[0m
09:30:50.521 -> �[0;35m[C][mdns:116]:   Hostname: esp32-bluetooth-proxy-2c540c�[0m
09:30:50.521 -> �[0;35m[C][ota:096]: Over-The-Air Updates:�[0m
09:30:50.565 -> �[0;35m[C][ota:097]:   Address: esp32-bluetooth-proxy-2c540c.local:3232�[0m
09:30:50.565 -> �[0;35m[C][ota:103]:   OTA version: 2.�[0m
09:30:50.565 -> �[0;35m[C][api:139]: API Server:�[0m
09:30:50.565 -> �[0;35m[C][api:140]:   Address: esp32-bluetooth-proxy-2c540c.local:6053�[0m
09:30:50.565 -> �[0;35m[C][api:142]:   Using noise encryption: YES�[0m
09:30:50.565 -> �[0;35m[C][improv_serial:032]: Improv Serial:�[0m
09:31:22.156 -> �[0;36m[D][api:102]: Accepted 192.168.50.50�[0m
09:31:22.222 -> �[0;33m[W][component:214]: Component api took a long time for an operation (0.06 s).�[0m
09:31:22.222 -> �[0;33m[W][component:215]: Components should block for at most 20-30ms.�[0m
09:31:22.293 -> �[0;36m[D][api.connection:1121]: Home Assistant 2024.2.1 (192.168.50.50): Connected successfully�[0m
```

### Additional information

The device I try to be capable to scan is a LD2411 presence sensor (has Bleutooth capabilities).

reivaj1, 2024-09-24:
i have the same problem, any comments?

jonny190, 2024-10-20:
Same issue, Any updates?

filmkorn, 2024-10-20:
Having tried 4 different esp32 boards - I think this might be a hardware issue of these particular ESP32-Wroom-32 boards. I have two of [these ](https://www.aliexpress.com/item/1005001621773806.html) (type usp-C) and the WiFi connectivity of this board is terrible as well. Btw. if you want to see if any packets are relayed via the BT proxy you'll have to make the logger verbose.

```
logger:
  level: VERBOSE
```