esphome/issues #5512: Connecting multiple EMC2101.

Microbeen, 2024-02-22:
### The problem

I'm trying to connect several EMC2101 modules to the ESP. 
The config allows you to use only one.
When trying to register several modules, the error "expected a dictionary" appears.
Several modules need to be connected. How to do it?

### Which version of ESPHome has the issue?

11.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.0

### What platform are you using?

ESP32

### Board

ESP32-WROOVER-IB

### Component causing the issue

emc2101

ssieb, 2024-02-22:
Try the PR as an external component.

ssieb, 2024-02-22:
Don't close it yet.  It will auto close when the PR is merged.

ssieb, 2024-02-22:
I assume that you're saying it works.

closed 2024-02-23 by kbx81