esphome/issues #5514: D1 mini with BME280 configuration failure after update ESPhome container.

RPiNut, 2024-02-22:
### The problem

After updating ESPHome to the lastest docker image, I would like to update my sensors to the latest version, but there was only one error. The BME280 sensor wouldn't update anymore. I hope there is a configuration edit after the BME280 SPI module is added and I didn't found the diffrence. The error mentioned: 

````
Failed config

sensor.bme280: [source /config/d1-mini.yaml:37]
  
  Platform not found: 'sensor.bme280'.
````

### Which version of ESPHome has the issue?

2024.02.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.2.2

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

sensor.bme280

### Example YAML snippet

```yaml
i2c:
  sda: D2
  scl: D1
  scan: False
#  id: bus_a

sensor:
  - platform: bme280
    temperature:
      name: "BME280 Temperature"
      id: bme280_temperature
      accuracy_decimals: 2
      filters:
        - sliding_window_moving_average:
            window_size: 12
            send_every: 1
    pressure:
      name: "BME280 Pressure"
      id: bme280_pressure
    humidity:
      name: "BME280 Humidity"
      id: bme280_humidity
      accuracy_decimals: 2
      filters:
        - sliding_window_moving_average:
            window_size: 12
            send_every: 1
    address: 0x76
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.2.0
INFO Reading configuration /config/d1-mini.yaml...
Failed config

sensor.bme280: [source /config/d1-mini.yaml:37]
  
  Platform not found: 'sensor.bme280'.
  platform: bme280
  temperature: 
    name: BME280 Temperature
    oversampling: 1x
    id: bme280_temperature
    accuracy_decimals: 2
    filters: 
      - sliding_window_moving_average: 
          window_size: 12
          send_every: 1
  pressure: 
    name: BME280 Pressure
    oversampling: 1x
    id: bme280_pressure
  humidity: 
    name: BME280 Humidity
    oversampling: 1x
    id: bme280_humidity
    accuracy_decimals: 2
    filters: 
      - sliding_window_moving_average: 
          window_size: 12
          send_every: 1
  address: 118
  update_interval: 60s
```

### Additional information

The line 37 of the information above, starts in the configuration at line 36 with `sensor:` After ` - Platform: bme280` all lines have red underline.

RPiNut, 2024-02-22:
After updating the sensor to `- platform: bme280_i2c` I got the following errors:

```
INFO ESPHome 2024.2.0
INFO Reading configuration /config/d1-mini.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing d1-mini (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- Wire @ 1.0
Compiling .pioenvs/d1-mini/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/d1-mini/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/d1-mini/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/d1-mini/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/d1-mini/src/esphome/components/api/subscribe_state.cpp.o
xtensa-lx106-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [.pioenvs/d1-mini/src/esphome/components/api/api_connection.cpp.o] Error 1
========================== [FAILED] Took 3.54 seconds ==========================
```

ssieb, 2024-02-22:
That compile error is caused by running out of RAM.

closed 2024-02-22 by ssieb