esphome/issues #5515: MH-Z19C  Co2 Sensor: Zeropoint Calibration not working

ar-ml, 2024-02-22:
### The problem

- MH-Z19C CO2 Sensor is connected to ESP32 and integrated in Home Assistant
- Sensor has been outside for 30 Minutes, but it shows 800ppm
- I tried to use a manually triggered Zero-Point calibration via Home Assitent
- **Expected Result:** CO2 should immediately go down to 400ppm
- **Observed Result:** In ESP Log can see the the Zero-Point Calibration being triggered, but the ppm values do not change!

Here is my code:

```
esphome:
  name: zw-esp32-co2
  friendly_name: ZW_ESP32_CO2

esp32:
  board: esp32dev
  framework:
    type: arduino

# Example configuration co2 sensor MH-Z19
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

sensor:
  - platform: mhz19
    co2:
      name: "MH-Z19 CO2 Value"
    temperature:
      name: "MH-Z19 Temperature"
    update_interval: 25s
    automatic_baseline_calibration: true
    id: zw_esp32_co2_id

button: #creates a button to calibrate the sensor - this will reset the zero point so use it only when it's been 20 minutes either outside or in a room you know the co2 level is at a minimum
  - platform: template
    name: "Calibrate Zero (20 minutes at 400ppm)"
    id: calibrate_zero
    entity_category: diagnostic
    on_press:
      - mhz19.calibrate_zero: zw_esp32_co2_id

switch: # this will expose a switch to home assistant to turn on and off the ABC
  - platform: template
    name: "Automatic Baseline Calibration"
    entity_category: diagnostic
    optimistic: true
    on_turn_on:
      mhz19.abc_enable: zw_esp32_co2_id
    on_turn_off:
      mhz19.abc_disable: zw_esp32_co2_id

binary_sensor: # exposes online status
  - platform: status
    name: "Sensor Status"

text_sensor:
  - platform: wifi_info
    ip_address: # exposes the IP Address when connected
      internal: true
      id: wifi_ip_addr
      name: "IP Address"
    ssid: # exposes the SSID when connected
      internal: true
      id: wifi_ssid

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXX"

ota:
  password: "XXXXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Zw-Esp32-Co2 Fallback Hotspot"
    password: "XXXXXXX"

captive_portal:
    
    
```

I also tried it using the code from the documentation, but it did not work either:
https://esphome.io/components/sensor/mhz19.html

This is what I see in the logs, when I push the button for triggering zero point calibration:
```
[00:32:47][D][button:010]: 'Calibrate Zero (20 minutes at 400ppm)' Pressed.
[00:32:47][D][mhz19:065]: MHZ19 Calibrating zero point
[00:32:47][W][component:214]: Component api took a long time for an operation (0.15 s).
[00:32:47][W][component:215]: Components should block for at most 20-30ms.
[00:32:52][D][mhz19:057]: MHZ19 Received CO₂=1129ppm Temperature=20°C Status=0x00
[00:32:52][D][sensor:094]: 'MH-Z19 CO2 Value': Sending state 820.00000 ppm with 0 decimals of accuracy
[00:32:53][D][sensor:094]: 'MH-Z19 Temperature': Sending state 5.00000 °C with 0 decimals of accuracy
```

**Could it be that mhz19.calibrate_zero Action is not working with** MH-Z19**C** ???
Can you fix that?

### Which version of ESPHome has the issue?

2024.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.3

### What platform are you using?

ESP32

### Board

Berrybase NMCU-ESP32

### Example YAML snippet

```yaml
esphome:
  name: zw-esp32-co2
  friendly_name: ZW_ESP32_CO2

esp32:
  board: esp32dev
  framework:
    type: arduino

# Example configuration co2 sensor MH-Z19
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

sensor:
  - platform: mhz19
    co2:
      name: "MH-Z19 CO2 Value"
    temperature:
      name: "MH-Z19 Temperature"
    update_interval: 25s
    automatic_baseline_calibration: true
    id: zw_esp32_co2_id

button: #creates a button to calibrate the sensor - this will reset the zero point so use it only when it's been 20 minutes either outside or in a room you know the co2 level is at a minimum
  - platform: template
    name: "Calibrate Zero (20 minutes at 400ppm)"
    id: calibrate_zero
    entity_category: diagnostic
    on_press:
      - mhz19.calibrate_zero: zw_esp32_co2_id

switch: # this will expose a switch to home assistant to turn on and off the ABC
  - platform: template
    name: "Automatic Baseline Calibration"
    entity_category: diagnostic
    optimistic: true
    on_turn_on:
      mhz19.abc_enable: zw_esp32_co2_id
    on_turn_off:
      mhz19.abc_disable: zw_esp32_co2_id

binary_sensor: # exposes online status
  - platform: status
    name: "Sensor Status"

text_sensor:
  - platform: wifi_info
    ip_address: # exposes the IP Address when connected
      internal: true
      id: wifi_ip_addr
      name: "IP Address"
    ssid: # exposes the SSID when connected
      internal: true
      id: wifi_ssid

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXX"

ota:
  password: "XXXXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Zw-Esp32-Co2 Fallback Hotspot"
    password: "XXXXXXX"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
[00:32:47][D][button:010]: 'Calibrate Zero (20 minutes at 400ppm)' Pressed.
[00:32:47][D][mhz19:065]: MHZ19 Calibrating zero point
[00:32:47][W][component:214]: Component api took a long time for an operation (0.15 s).
[00:32:47][W][component:215]: Components should block for at most 20-30ms.
[00:32:52][D][mhz19:057]: MHZ19 Received CO₂=1129ppm Temperature=20°C Status=0x00
[00:32:52][D][sensor:094]: 'MH-Z19 CO2 Value': Sending state 820.00000 ppm with 0 decimals of accuracy
[00:32:53][D][sensor:094]: 'MH-Z19 Temperature': Sending state 5.00000 °C with 0 decimals of accuracy
```

ar-ml, 2024-03-02:
I contacted the manufactorer Winsen Sensor and got the following reply:

**Q1: Why is Zeropoint Calibration not working?**
For the MH-Z19C Co2 Sensor there are two ways to calibrate the zero point:
- One is by inputting a low-level signal for more than 7 seconds.
- The other is by sending a command: FF 01 87 00 00 00 00 00 78. 
In a well-ventilated environment, the sensor output is calibrated to around 400ppm.
The command for successful calibration return instruction: FF 87 01 00 00 00 00 00 78.

**Q2: How to switch auto-calibration (ABC) between 1 day and 7 day cycle? **

-  The command for automatic calibration period of 7 days is:FF 01 AF 01 07 00 00 00 48，
- After successful transmission, the sensor returns the command:FF AF 01 07 00 00 00 00 49.

**Is there anyone who can update the ESPhome mhz19 component code, to support these commands?**
https://esphome.io/components/sensor/mhz19.html
https://github.com/esphome/esphome/tree/dev/esphome/components/mhz19

ar-ml, 2024-03-04:
Update:
A dev checked the code and found out that zero point calibration should already be using the command mentioned above...
No idea why it is not working then...

Maybe someone else wants to continue here one day...

I will probably switch to a different model / manufacturer now for further development:
https://esphome.io/components/sensor/senseair.html
(Seems to have much better ABC auto-calibration, and should work with sensor models Senseair Sunlight, Senseair Sunrise and Senseair S8.)

DSchreyer, 2025-02-23:
Did you find a soution for this ? I am facing a similar issue that the measurement doesnt drop below 800ppm even after zero point calibration.