esphome/issues #5516: Syntax Highlight Broken in v2024.2.0

samuelolteanu, 2024-02-23:
### The problem

Like the title says, yaml syntax highlight in HA addon is broken. No highlight at all unless saved and open file again where it recognises the previous error, but not new ones. Cleaning build files did not do anything.
Downgrading the addon (with a backup) fixes the issue except but I have to go back and forth with the versions because it gives "ERROR Unsupported OTA version 2" when uploafing with the old version

### Which version of ESPHome has the issue?

2024.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.2

### What platform are you using?

ESP32

### Board

ESP32C3

### Component causing the issue

Addon syntax highlight, OTA Downgrading

### Anything in the logs that might be useful for us?

```txt
ERROR Unsupported OTA version 2
```

x99percent, 2024-02-23:
Flash with version 2024.2.0 using...
```
ota:
  version: 1
```
This will allow you to go back to 2023.12.9. You will need to remove the `version: 1` when flashing from the older ESPHome.

EOUpSL93Y, 2024-03-27:
Looks like fixed here: https://github.com/esphome/esphome/pull/6431

samuelolteanu, 2024-04-05:
> Looks like fixed here: [esphome/esphome#6431](https://github.com/esphome/esphome/pull/6431)

closed 2024-04-05 by samuelolteanu