esphome/issues #5525: Gfonts usage fail in `2024.2.0`

rhoriguchi, 2024-02-24:
### The problem

When trying to update my device. The full config can be found [here](https://github.com/rhoriguchi/nixos-setup/blob/9fb7715bbde9aa11883bb50bd9bc514da205cb0d/configuration/devices/headless/server/home-assistant/esphome/airgradient-one/_config.yaml).

```console
INFO ESPHome 2024.2.0
INFO Reading configuration /var/lib/esphome/airgradient-one.yaml...
Failed config

font: [source /var/lib/esphome/.esphome/packages/61611988/configuration/devices/headless/server/home-assistant/esphome/airgradient-one/display_sh1106.yaml:2]
  
  Component not found: font.
  - file: gfonts://Open Sans
    id: open_sans_14
    size: 14
    glyphs: !"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³
  - file: gfonts://Open Sans
    id: open_sans_9
    size: 9
    glyphs: !"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³
  - file: gfonts://Open Sans
    id: open_sans_20
    size: 20
    glyphs: !"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³
```

### Which version of ESPHome has the issue?

2023.12.9 and 2024.2.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32-c3-devkitm-1

### Component causing the issue

display

### Example YAML snippet

```yaml
font:
  - file: "gfonts://Open Sans"
    id: open_sans_14
    size: 14
    glyphs: '!"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³'
  - file: "gfonts://Open Sans"
    id: open_sans_9
    size: 9
    glyphs: '!"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³'
  - file: "gfonts://Open Sans"
    id: open_sans_20
    size: 20
    glyphs: '!"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³'
```

### Anything in the logs that might be useful for us?

```txt
esphome -v --dashboard run /var/lib/esphome/airgradient-one.yaml --device OTA
INFO ESPHome 2024.2.0
INFO Reading configuration /var/lib/esphome/airgradient-one.yaml...
INFO Cloning https://github.com/rhoriguchi/nixos-setup.git@None
DEBUG Location: /var/lib/esphome/.esphome/packages/61611988
DEBUG Running git command: git clone --depth=1 -- https://github.com/rhoriguchi/nixos-setup.git /var/lib/esphome/.esphome/packages/61611988
\033[1;31mFailed config\033[0m

\033[1;31mfont:\033[0m \033[36m[source /var/lib/esphome/.esphome/packages/61611988/configuration/devices/headless/server/home-assistant/esphome/airgradient-one/display_sh1106.yaml:2]\033[0m

  \033[1;31mComponent not found: font.\033[0m
  - file: gfonts://Open+Sans
    id: open_sans_14
    size: 14
    glyphs: !"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³
  - file: gfonts://Open+Sans
    id: open_sans_9
    size: 9
    glyphs: !"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³
  - file: gfonts://Open+Sans
    id: open_sans_20
    size: 20
    glyphs: !"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³
```

ssieb, 2024-02-24:
You mention gfonts, but that's not where the problem is.  It's the font component itself.  Did you try with local fonts?
Make sure you have pillow version 10.2.0.  I know you've had problems with dependency versions before on nixos.

rhoriguchi, 2024-02-24:
> You mention gfonts, but that's not where the problem is. It's the font component itself. Did you try with local fonts? Make sure you have pillow version 10.2.0. I know you've had problems with dependency versions before on nixos.

I'm running pillow 10.2.0, here is the dependency tree: 
[image]

ssieb, 2024-02-24:
I remember the last person that had a problem with nixos found that the required dependency was "installed", but not available in the right place or something.
It might not be pillow because that should cause a validation error.  A fail like you're getting is more likely to be one of the critical dependencies from this list:
```
import functools
from pathlib import Path
import hashlib
import os
import re
from packaging import version

import requests
```
Try running python and copy and paste that chunk.

rhoriguchi, 2024-02-25:
> I remember the last person that had a problem with nixos found that the required dependency was "installed", but not available in the right place or something. It might not be pillow because that [...]

Thank you for the help, the missing dependency is `packaging`.

```console
Traceback (most recent call last):
  File "/tmp/tmp.6MISZZjurT/./test.py", line 12, in <module>
    from packaging import version
```

closed 2024-02-25 by rhoriguchi