esphome/issues #5526: IR Remote Component api took a long time for an operation

Dragontamer333, 2024-02-24:
### The problem

I have a single wemos d1 mini running an IR transmitter. Whenever I try and send the data using a button I get this error

[10:50:47][D][button:010]: 'Test Button' Pressed.
[10:50:47][D][remote_transmitter:075]: Sending remote code...
[10:50:47][W][component:214]: Component api took a long time for an operation (0.07 s).
[10:50:47][W][component:215]: Components should block for at most 20-30ms.

What does this mean. Some research shows that its just a warning, but the ir transmitter doesn't send any data. I have tried two different transmitters and 2 different esp devices. 

### Which version of ESPHome has the issue?

2024.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.3

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

remote_transmitter

### Example YAML snippet

```yaml
captive_portal:

remote_receiver:
  pin: 
    number: GPIO02
    inverted: True
  dump:
    - lg

remote_transmitter:
  pin: GPIO16
  carrier_duty_percent: 50%

button:
  - platform: template
    name: Test Button
    on_press:
      - remote_transmitter.transmit_lg:
          data: 0x20DF10EF # power on/off
          nbits: 32
```

### Anything in the logs that might be useful for us?

```txt
[11:09:26][C][logger:447]: Logger:
[11:09:26][C][logger:448]:   Level: DEBUG
[11:09:26][C][logger:449]:   Log Baud Rate: 115200
[11:09:26][C][logger:451]:   Hardware UART: UART0
[11:09:26][C][remote_transmitter:018]: Remote Transmitter...
[11:09:26][C][remote_transmitter:019]:   Carrier Duty: 50%
[11:09:26][C][remote_transmitter:020]:   Pin: GPIO16
[11:09:26][C][remote_receiver.esp8266:060]: Remote Receiver:
[11:09:26][C][remote_receiver.esp8266:061]:   Pin: GPIO2
[11:09:26][C][remote_receiver.esp8266:066]:   Buffer Size: 1000
[11:09:26][C][remote_receiver.esp8266:067]:   Tolerance: 25%
[11:09:26][C][remote_receiver.esp8266:068]:   Filter out pulses shorter than: 50 us
[11:09:26][C][remote_receiver.esp8266:069]:   Signal is done after 10000 us of no changes
[11:09:26][C][captive_portal:088]: Captive Portal:
[11:09:26][C][mdns:115]: mDNS:
[11:09:26][C][mdns:116]:   Hostname: remote
[11:09:26][C][ota:096]: Over-The-Air Updates:
[11:09:26][C][ota:097]:   Address: remote.local:8266
[11:09:26][C][ota:100]:   Using Password.
[11:09:26][C][ota:103]:   OTA version: 2.
[11:09:26][C][api:139]: API Server:
[11:09:26][C][api:140]:   Address: remote.local:6053
[11:09:26][C][api:142]:   Using noise encryption: YES
[11:09:27][D][api:102]: Accepted 192.168.0.101
[11:09:27][W][component:214]: Component api took a long time for an operation (0.29 s).
[11:09:27][W][component:215]: Components should block for at most 20-30ms.
[11:09:29][D][api.connection:1121]: Home Assistant 2024.2.3 (192.168.0.101): Connected successfully
[11:10:16][D][button:010]: 'Test Button' Pressed.
[11:10:16][D][remote_transmitter:075]: Sending remote code...
[11:10:16][W][component:214]: Component api took a long time for an operation (0.07 s).
[11:10:16][W][component:215]: Components should block for at most 20-30ms.
```

DAVe3283, 2024-02-24:
You can get all the details here: #4717

randybb, 2024-02-24:
Since there is already an issue (and pinned, so everybody would spot it...) and all components have been already mentioned I am closing this.

closed 2024-02-24 by randybb