esphome/issues #5530: Unable to build across 2024.02.0

npawelek, 2024-02-25:
### The problem

Upgraded docker install from previous version to 2024.2.0 and any build across ESP32/ESP32-IDF are consistently hanging at the same compile step of `subscribe_state.cpp.o`. At this point, disk activity skyrockets and processes start reading hundreds of MB/s and continue until stopped. The container has to be stopped to kill off the active processes for platformio, pressing stop in the esphome console closes the UI window, but the processes remain active. Not entirely sure what's happening. I've attempted to clear my cache directory and build files and the same behavior is noted.

### Which version of ESPHome has the issue?

2024.2.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

Any

### Component causing the issue

compiling subscribe_state.cpp.o

### Example YAML snippet

```yaml
substitutions:
  device_name: ble-proxy-garage-poe
  friendly_name: BLE Proxy Garage PoE

esphome:
  name: ${device_name}

esp32:
  board: esp32-poe-iso
  framework:
    type: esp-idf

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12
  manual_ip:
    static_ip: 192.168.10.11
    gateway: 192.168.10.1
    subnet: 255.255.255.0

# improv_serial:
logger:
  # level: INFO

# Enable Home Assistant API
api:
  encryption:
    key: !secret noise_base64_encryption_key

ota:
  password: !secret ota_password

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

time:
- platform: homeassistant
  id: homeassistant_time

button:
- platform: restart
  name: "${friendly_name} Restart"
- platform: safe_mode
  name: "${friendly_name} Safe Mode"

text_sensor:
- platform: version
  name: ${friendly_name} ESPHome Version
  icon: "mdi:unfold-more-vertical"
  hide_timestamp: True
- platform: template
  name: ${friendly_name} Uptime
  id: uptime_human
  icon: mdi:clock-start
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.2.0
INFO Reading configuration /config/ble-proxy-garage-poe.yaml...
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'America/Chicago'
INFO Generating C++ source...
INFO Compiling app...
Processing ble-proxy-garage-poe (board: esp32-poe-iso; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40406.240122 (4.4.6) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
Compiling .pioenvs/ble-proxy-garage-poe/src/esphome/components/api/api_connection.o
Compiling .pioenvs/ble-proxy-garage-poe/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/ble-proxy-garage-poe/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/ble-proxy-garage-poe/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/ble-proxy-garage-poe/src/esphome/components/api/api_server.o
Compiling .pioenvs/ble-proxy-garage-poe/src/esphome/components/api/list_entities.o
Compiling .pioenvs/ble-proxy-garage-poe/src/esphome/components/api/proto.o
Compiling .pioenvs/ble-proxy-garage-poe/src/esphome/components/api/subscribe_state.o
```

### Additional information

Config provided is relatively barebones bluetooth proxy configuration for an olimex poe board. ESPHome is configured with the following platformio env variables:

```
    env:
      TZ: ${TIMEZONE}
      ESPHOME_DASHBOARD_USE_PING: "true"
      PLATFORMIO_CORE_DIR: /cache/platformio
      PLATFORMIO_GLOBALLIB_DIR: /cache/platformio/lib
      PLATFORMIO_SETTING_ENABLE_TELEMETRY: No
```

This is due to permissions issues noted in https://github.com/esphome/issues/issues/3558.

paulo-serrao, 2024-02-26:
@npawelek I was facing the same issue, but on my case the compilation would stop on
'''Compiling .pioenvs/XXXXX/src/esphome/components/api/list_entities.o''' and I could get popups of homeassistant components restarting.

I turns out I was running out of memory (100% swap and memory usage, you can check that with https://github.com/hassio-addons/addon-glances), increasing my swap file to 2GB (https://community.home-assistant.io/t/how-to-increase-the-swap-file-size-on-home-assistant-os/272226) solved the issue and I can now compile ESPHome

npawelek, 2024-02-26:
Thank you for including this detail. I had a 512MB memory limit on the container. Removing this resolved the issue entirely.

closed 2024-02-26 by npawelek