esphome/issues #5532: sntp component compilation failure with recent esp-idf

mag1024, 2024-02-25:
### The problem

```
esphome:
  name: "test"

esp32:
  variant: esp32s3
  board: lolin_s3_mini
  framework:
    type: esp-idf
    version: latest
    platform_version: 6.5.0

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

time:
  - platform: sntp
    servers: 10.0.5.1
```

```
In file included from src/esphome/components/sntp/sntp_component.cpp:7:
/data/cache/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:55:2: warning: #warning "Defined!" [-Wcpp]
   55 | #warning "Defined!"
      |  ^~~~~~~
/data/cache/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:235:6: error: 'void sntp_setoperatingmode(u8_t)' was declared 'extern' and later 'static' [-fpermissive]
  235 | void sntp_setoperatingmode(u8_t operating_mode)
      |      ^~~~~~~~~~~~~~~~~~~~~
In file included from src/esphome/components/sntp/sntp_component.cpp:5:
/data/cache/platformio/packages/framework-espidf/components/lwip/lwip/src/include/lwip/apps/sntp.h:51:6: note: previous declaration of 'void sntp_setoperatingmode(u8_t)'
   51 | void sntp_setoperatingmode(u8_t operating_mode);
      |      ^~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:242:1: error: expected unqualified-id before '{' token
  242 | {
      | ^
/data/cache/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:249:6: error: 'void sntp_setservername(u8_t, const char*)' was declared 'extern' and later 'static' [-fpermissive]
  249 | void sntp_setservername(u8_t idx, const char *server)
      |      ^~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/lwip/lwip/src/include/lwip/apps/sntp.h:66:6: note: previous declaration of 'void sntp_setservername(u8_t, const char*)'
   66 | void sntp_setservername(u8_t idx, const char *server);
      |      ^~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:255:6: error: 'void sntp_init()' was declared 'extern' and later 'static' [-fpermissive]
  255 | void sntp_init(void)
      |      ^~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/lwip/lwip/src/include/lwip/apps/sntp.h:54:6: note: previous declaration of 'void sntp_init()'
   54 | void sntp_init(void);
      |      ^~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:261:13: error: 'const char* sntp_getservername(u8_t)' was declared 'extern' and later 'static' [-fpermissive]
  261 | const char *sntp_getservername(u8_t idx)
      |             ^~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/lwip/lwip/src/include/lwip/apps/sntp.h:67:13: note: previous declaration of 'const char* sntp_getservername(u8_t)'
   67 | const char *sntp_getservername(u8_t idx);
      |             ^~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:267:18: error: 'const ip_addr_t* sntp_getserver(u8_t)' was declared 'extern' and later 'static' [-fpermissive]
  267 | const ip_addr_t* sntp_getserver(u8_t idx)
      |                  ^~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/lwip/lwip/src/include/lwip/apps/sntp.h:59:18: note: previous declaration of 'const ip_addr_t* sntp_getserver(u8_t)'
   59 | const ip_addr_t* sntp_getserver(u8_t idx);
      |                  ^~~~~~~~~~~~~~
src/esphome/components/sntp/sntp_component.cpp: In member function 'virtual void esphome::sntp::SNTPComponent::setup()':
src/esphome/components/sntp/sntp_component.cpp:34:24: warning: 'void sntp_setoperatingmode(u8_t)' is deprecated: use esp_sntp_setoperatingmode() instead [-Wdeprecated-declarations]
   34 |   sntp_setoperatingmode(SNTP_OPMODE_POLL);
      |   ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:235:6: note: declared here
  235 | void sntp_setoperatingmode(u8_t operating_mode)
      |      ^~~~~~~~~~~~~~~~~~~~~
src/esphome/components/sntp/sntp_component.cpp:40:21: warning: 'void sntp_setservername(u8_t, const char*)' is deprecated: use esp_sntp_setservername() instead [-Wdeprecated-declarations]
   40 |   sntp_setservername(0, strdup(this->server_1_.c_str()));
      |   ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:249:6: note: declared here
  249 | void sntp_setservername(u8_t idx, const char *server)
      |      ^~~~~~~~~~~~~~~~~~
src/esphome/components/sntp/sntp_component.cpp:42:23: warning: 'void sntp_setservername(u8_t, const char*)' is deprecated: use esp_sntp_setservername() instead [-Wdeprecated-declarations]
   42 |     sntp_setservername(1, strdup(this->server_2_.c_str()));
      |     ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:249:6: note: declared here
  249 | void sntp_setservername(u8_t idx, const char *server)
      |      ^~~~~~~~~~~~~~~~~~
src/esphome/components/sntp/sntp_component.cpp:45:23: warning: 'void sntp_setservername(u8_t, const char*)' is deprecated: use esp_sntp_setservername() instead [-Wdeprecated-declarations]
   45 |     sntp_setservername(2, strdup(this->server_3_.c_str()));
      |     ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:249:6: note: declared here
  249 | void sntp_setservername(u8_t idx, const char *server)
      |      ^~~~~~~~~~~~~~~~~~
src/esphome/components/sntp/sntp_component.cpp:51:12: warning: 'void sntp_init()' is deprecated: use esp_sntp_init() instead [-Wdeprecated-declarations]
   51 |   sntp_init();
      |   ~~~~~~~~~^~
/data/cache/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:255:6: note: declared here
  255 | void sntp_init(void)
      |      ^~~~~~~~~
*** [.pioenvs/test/src/esphome/components/sntp/sntp_component.o] Error 1

```

### Which version of ESPHome has the issue?

2024.2.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Component causing the issue

sntp

ssieb, 2024-02-25:
If you modify the platform versions then any issues are yours to deal with as the compile warning says.  That is not supported and issues are expected.

closed 2024-02-25 by ssieb

randybb, 2024-02-25:
ESPHome is still on ESP-IDF 4.4, you have 5.1 and that is not yet supported. The same issue was raised (and closed) couple days ago.