esphome/issues #5533: Editor Code Check not 'Live' anymore

3ative, 2024-02-25:
### The problem

When editing code, the editor no longer updates live with errors. 
 - You have to save, close and re-open for them to appear.
 

### Which version of ESPHome has the issue?

2024.2.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

3ative, 2024-02-25:
Leaving this open of others to find. However, I've been informed there **IS** a fix coming. 💖

3ative, 2024-02-26:
Maybe related to:
 -  https://github.com/esphome/issues/issues/5342
 - https://github.com/esphome/esphome/pull/6280

BigBobbas, 2024-02-26:
I have have the same issue, was hoping that https://github.com/esphome/esphome/pull/6280 would resolve this issue as well as the save speed issue. 
The validation problem was mentioned in https://github.com/esphome/issues/issues/5342 which has been closed as resolved. However the validation issue is still the same after update to 2024.2.1
I Have to save and close editor for validation errors to show.

3ative, 2024-02-26:
Indeed... I put "_maybe related to_" as there seemed to be some similarities

closed 2024-03-26 by jesserockz