esphome/issues #5536: Adding `button` breaks build when using MQTT

fsaris, 2024-02-26:
### The problem

I added the restart button

```
button:
  - platform: restart
    name: "Restart"
```
And got the following error during the linking step:

```
Linking .pioenvs/awox-ble-mesh-hub/firmware.elf
/esphome/.temp/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/awox-ble-mesh-hub/src/esphome/components/mqtt/mqtt_button.o:(.literal._ZNSt17_Function_handlerIFvRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_EZN7esphome4mqtt19MQTTButtonComponent5setupEvEUlS7_S7_E_E9_M_invokeERKSt9_Any_dataS7_S7_+0x14): undefined reference to `esphome::button::Button::press()'
/esphome/.temp/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/awox-ble-mesh-hub/src/esphome/components/mqtt/mqtt_button.o: in function `std::_Function_handler<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&), esphome::mqtt::MQTTButtonComponent::setup()::{lambda(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)#1}>::_M_invoke(std::_Any_data const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/workspaces/esphome/config/.esphome/build/awox-ble-mesh-hub/src/esphome/components/mqtt/mqtt_button.cpp:21: undefined reference to `esphome::button::Button::press()'
/esphome/.temp/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/awox-ble-mesh-hub/src/main.o:(.literal._Z5setupv+0xf4): undefined reference to `vtable for esphome::restart::RestartButton'
/esphome/.temp/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/awox-ble-mesh-hub/src/main.o:(.literal._Z5setupv+0xf8): undefined reference to `vtable for esphome::restart::RestartButton'
collect2: error: ld returned 1 exit status
````

### Which version of ESPHome has the issue?

2024.2.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Example YAML snippet

```yaml
button:
  - platform: restart
    name: "Restart"
```
```

### Anything in the logs that might be useful for us?

```txt
Linking .pioenvs/awox-ble-mesh-hub/firmware.elf
/esphome/.temp/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/awox-ble-mesh-hub/src/esphome/components/mqtt/mqtt_button.o:(.literal._ZNSt17_Function_handlerIFvRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_EZN7esphome4mqtt19MQTTButtonComponent5setupEvEUlS7_S7_E_E9_M_invokeERKSt9_Any_dataS7_S7_+0x14): undefined reference to `esphome::button::Button::press()'
/esphome/.temp/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/awox-ble-mesh-hub/src/esphome/components/mqtt/mqtt_button.o: in function `std::_Function_handler<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&), esphome::mqtt::MQTTButtonComponent::setup()::{lambda(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)#1}>::_M_invoke(std::_Any_data const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/workspaces/esphome/config/.esphome/build/awox-ble-mesh-hub/src/esphome/components/mqtt/mqtt_button.cpp:21: undefined reference to `esphome::button::Button::press()'
/esphome/.temp/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/awox-ble-mesh-hub/src/main.o:(.literal._Z5setupv+0xf4): undefined reference to `vtable for esphome::restart::RestartButton'
/esphome/.temp/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/awox-ble-mesh-hub/src/main.o:(.literal._Z5setupv+0xf8): undefined reference to `vtable for esphome::restart::RestartButton'
collect2: error: ld returned 1 exit status
````
```

ssieb, 2024-02-26:
Do a clean build.

fsaris, 2024-02-26:
Ah, that fixed it indeed. Tnx

_mist that instruction_

closed 2024-02-26 by fsaris