esphome/issues #5537: ESP Port of keyble BLE lock not working anymore after 2024.2.0

user45876, 2024-02-26:
### The problem

After updating to ESPHome 2024.2, the ESP implementation of keyble BLE lock is not working any more. The issue is described here as well: https://github.com/digaus/esphome-components-eqiva/issues/11 . The issue is persisting with 2024.2.1.

### Which version of ESPHome has the issue?

2024.2.x

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

ble_client probably?

### Example YAML snippet

```yaml
https://github.com/digaus/esphome-components-eqiva/blob/main/example.yaml
```

### Anything in the logs that might be useful for us?

```txt
Please kindly see log at https://github.com/digaus/esphome-components-eqiva/issues/11
```

### Additional information

Downgrading to (and flashing the ESP with) 2023.12.9 instantly resolved the issue.

Could it be related to https://github.com/esphome/esphome/pull/5277 ?

randybb, 2024-02-26:
If you have a problem with an external component that is not compatible with the latest version, then it has to be resolved on their side.

closed 2024-02-26 by randybb

user45876, 2024-02-26:
Dear @randybb , it seems likely, that the new version of esphome, especially the work on BLE has broken it.

Can you please reconsider and reread the issue? I only pointed to the other issue for reference.

As said, it is working fine in 2024.12.9 but instabreaks in 2024.2.x.

randybb, 2024-02-26:
ESPHome doens't need to be compatible with all external components. External components have to be compatible with ESPHome. So if something is changed, they have to fix it. If you can replicate the issue in pure ESPHome, then it would be a valid issue.

user45876, 2024-02-26:
Understood, thanks.

user45876, 2024-02-26:
Just for documentation and in case somebody else falls over this: Using the 2023.12.9 components for the esp32 ble components is a workaround until the equiva keyble component might be updated:

```
external_components:
  - source: github://digaus/esphome-components-eqiva

  # use esp32_ble components from ESPHome 2023.12.9
  - source: github://esphome/esphome@2023.12.9
    components: [ esp32_ble, esp32_ble_client, esp32_ble_tracker ]
```