esphome/issues #5538: on_rc_switch can only be used with the predefined protocols

gabest11, 2024-02-26:
### The problem

Under the binary_sensor / rc_switch_raw we can define our own rcswitch protocol, but cannot use the code value, it's just a binary_sensor afterall.

Under remote_receiver / on_rc_switch we can get the code and the protocol number, but only one of the predefined 9 protocols gets there.

If all the user defined protocols were included, then it would be possible to catch more codes with changing bitfields, like weather stations or rolling codes, and report them through template sensors.

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.1.6

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

remote_receiver

### Example YAML snippet

```yaml
remote_receiver:
  pin: GPIO32
  dump:
    - rc_switch
  tolerance: 50%
  buffer_size: 2kb
  filter: 250us
  idle: 4ms
  on_rc_switch: # only predefined protocols, not the one below
    - logger.log:
        format: "on_rc_switch: %i %llu"
        args: [x.protocol, x.code]
      - lambda: |- 
          if(matches) id(some_sensor).publish_state(extracted bits);

binary_sensor:
  - platform: remote_receiver
    name: "auriol rc weather station with rain gauge"
    internal: true
    on_press:
      - lambda: |-
          ESP_LOGD("rc_switch", "auriol rc weather station with rain gauge");
    rc_switch_raw:
      # ID(8)BATTERY(1)0(1)CHANNEL(2)TEMP_C(12)1(4)RAIN(24)
      code: 'xxxxxxxxx0xxxxxxxxxxxxxx1111xxxxxxxxxxxxxxxxxxxxxxxx'
      protocol: # yes, this catches it, PPM, same ON time, different OFF time
        pulse_length: 49
        sync: [10,81]
        zero: [10,21]
        one: [10,41]
        inverted: false
    filters:
      - delayed_off: 1000ms
```

gabest11, 2024-02-26:
Then that's my issue. It isn't called.

ssieb, 2024-02-26:
Show your actual config and the logs.

gabest11, 2024-02-26:
```
substitutions:
  plug_name: cc1101_test_esp32

external_components:
  - source:
      type: local
      path: components
  
packages:
  api: !include packages/api_zsibi4.yaml
  #mqtt: !include packages/mqtt_zsibi4.yaml
  wifi: !include packages/wifi_zsibi4.yaml
  info: !include packages/device_info.yaml

esphome:
  name: ${plug_name}

esp32:
  board: nodemcu-32s
  variant: esp32
  framework:
    type: arduino
    #type: esp-idf
    #version: recommended

logger:
#  level: DEBUG
  level: VERBOSE
#  level: VERY_VERBOSE
#  level: NONE

ota:
  password: !secret every_password

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

switch:
  - platform: cc1101
    name: "${plug_name} cc1101"
    id: transceiver
    cs_pin: 5
    gdo0: 32
    gdo2: 33
    bandwidth: 200
    frequency: 433920
  
remote_transmitter:
  - pin: GPIO32 # This is GDO0
    carrier_duty_percent: 100%

remote_receiver:
  pin: GPIO33 # This is GDO2
  dump:
    - rc_switch
    #- raw
  # Settings to optimize recognition of RF devices
  tolerance: 50%
  #buffer_size: 2kb
  filter: 250us
  idle: 4ms
  on_rc_switch:
#    - lambda: |-
#        ESP_LOGD("rc_switch", "protocol = %d, code = '%llu'", x.protocol, x.code);
    - logger.log:
        format: "on_rc_switch: %i %llu"
        args: [x.protocol, x.code]
[... 106 lines cut ...]
[23:30:24][C][wifi_info:009]: WifiInfo IPAddress 'cc1101_test_esp32 IP'
[23:30:24][V][wifi_info:009]:   Unique ID: 'fcb4674e6674-wifiinfo-ip'
[23:30:24][C][wifi_info:011]: WifiInfo SSID 'cc1101_test_esp32 SSID'
[23:30:24][V][wifi_info:011]:   Unique ID: 'fcb4674e6674-wifiinfo-ssid'
[23:30:24][C][wifi_signal.sensor:009]: WiFi Signal 'cc1101_test_esp32 WiFi Signal'
[23:30:24][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[23:30:24][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[23:30:24][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[23:30:24][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[23:30:24][V][wifi_signal.sensor:009]:   Unique ID: 'fcb4674e6674-wifisignal'
[23:30:28][V][sensor:043]: 'cc1101_test_esp32 WiFi Signal': Received new state -62.000000
[23:30:28][D][sensor:094]: 'cc1101_test_esp32 WiFi Signal': Sending state -62.00000 dBm with 0 decimals of accuracy
[23:30:29][D][binary_sensor:036]: 'cc1101_test_esp32 lidl auriol rc weather station with rain gauge': Sending state ON
[23:30:29][D][rc_switch:081]: lidl auriol rc weather station with rain gauge
[23:30:30][D][binary_sensor:036]: 'cc1101_test_esp32 lidl auriol rc weather station with rain gauge': Sending state OFF
[23:31:26][D][binary_sensor:036]: 'cc1101_test_esp32 lidl auriol rc weather station with rain gauge': Sending state ON
[23:31:26][D][rc_switch:081]: lidl auriol rc weather station with rain gauge
[23:31:27][D][binary_sensor:036]: 'cc1101_test_esp32 lidl auriol rc weather station with rain gauge': Sending state OFF
[23:31:28][V][sensor:043]: 'cc1101_test_esp32 WiFi Signal': Received new state -62.000000
[23:31:28][D][sensor:094]: 'cc1101_test_esp32 WiFi Signal': Sending state -62.00000 dBm with 0 decimals of accuracy
[23:31:29][I][ota:117]: Boot seems successful, resetting boot loop counter.
[23:31:29][D][esp32.preferences:114]: Saving 1 preferences to flash...
[23:31:29][V][esp32.preferences:126]: sync: key: 233825507, len: 4
[23:31:29][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed`
```

ssieb, 2024-02-26:
Looks like it's working to me.  What's the issue?

gabest11, 2024-02-26:
Updated the config and the log with a window sensor. There is triggers the on_rc_switch, because it uses one of the predefined protocols and I think those are analyzed separately for the trigger mechanism.

gabest11, 2024-02-26:
Can you see on_rc_switch for the weather station?

ssieb, 2024-02-26:
I don't see a received rc_switch message for the weather station either.
That's a strange protocol to use for a weather station anyway...

gabest11, 2024-02-26:
It's towards the end, I logged it from on_press "ESP_LOGD("rc_switch", "lidl auriol rc weather station with rain gauge");"

ssieb, 2024-02-27:
Oh, you're just making up a protocol.  That's not rc switch.  You're just hacking the timings to make the rc switch decoder accept it.  You need to make a proper decoder or use the pronto or raw receiver for it.

gabest11, 2024-02-27:
Well. I'm not making one up. The 9 timings in RC_SWITCH_PROTOCOLS are made up, any device I get have slightly different timings. Even the last one on the list RCSwitchBase(250, 2500, 250, 1250, 250, 250, false) looks PPM to me. 250+1250 and 250+250 are not equally long for one and zero, one the high pulse is the same. Similar to that weather station.

ssieb, 2024-02-27:
No, it's not going to accept your random timings.  If `remote.rc_switch` isn't going to detect the transmission, then the trigger isn't going to work either.

gabest11, 2024-02-27:
I does detect it perfectly fine. This is not my first device where I analyze the timings and set them in a custom protocol. But the first one where it would be nice to report the masked values in a sensor.

ssieb, 2024-02-27:
The rc switch protocol timings are not made up.  There's a list of standard timings.  And yes, there are usually slightly different timings depending on the device that's sending and receiving and that's why there are receiver options to account for that.

ssieb, 2024-02-27:
This is not going to work the way you want it to.

gabest11, 2024-02-27:
So you think there cannot be other timings, only those you know of? Why does the custom protocol option exist?

ssieb, 2024-02-27:
You can use it for the binary sensor and for sending, but it won't work for general receiving.

gabest11, 2024-02-27:
I see, so as you suggested I need to create a no_rc_switch_that_works_for_me receiver. It will be a learning experience.

ssieb, 2024-02-27:
If you're going to do that, you should just create a decoder for the actual protocol.

TheChatty, 2024-06-07:
@gabest11: There's a workaround for you.

Basically you're patching `rc_switch_protocol.cpp` with your own protocols.

One way to achieve this is to make this file available outside it's container, e.g. in `/config/esphome`. You will need to redo the last step after each esphome update.

```
# ssh username@homeassistant
... enter password...
# docker exec -it $(docker ps -f name=esphome -q) cp /esphome/esphome/components/remote_base/rc_switch_protocol.cpp /config/esphome/
# docker exec -it $(docker ps -f name=esphome -q) ln -sf /config/esphome/rc_switch_protocol.cpp /esphome/esphome/components/remote_base/
```

So after including your own protocols you can simply edit your yaml and install it normally like you did before.

gabest11, 2024-06-07:
That's what I do to make the dashboard work with IIS (https://github.com/esphome/esphome/pull/6845), but the changes are lost with every update, so it's a bit inconvenient.

KHit6, 2025-03-25:
I think I understand what is meant. With some bit timings one has to use a extremely high 50% tolerance to get successfully data receives. The 50% tolerance will be useless  if you like to send. For some devices none of the bit timings offered match the actually bit timings, so it would be nice to set a custom bit timing. 

Please consider my description regarding a project "[TFA Dostmann (La Crosse) 30.32XX temperature humidity outdoor sensor - rc_switch](https://community.home-assistant.io/t/tfa-dostmann-la-crosse-30-32xx-temperature-humidity-outdoor-sensor-rc-switch/812847)".

The "[rc-switch project](https://github.com/sui77/rc-switch/blob/436a74b03f3dc17a29ee327af29d5a05d77f94b9/RCSwitch.cpp)" provides a method to set custom bit timings:
```
/**
  * Sets the protocol to send.
  */
void RCSwitch::setProtocol(Protocol protocol) {
  this->protocol = protocol;
}
```
A similar possibility with ESPHOME would be nice.

KHit6, 2025-03-26:
[rc_switch_protocol.cpp](https://github.com/esphome/esphome/blob/dev/esphome/components/remote_base/rc_switch_protocol.cpp) defines 8 fixed bit timings and does not allow to set a custom bit timing. 

It would be advantageous if the user could define his custom bit timing himself. At the user level - the ESPHome configuration yaml file - I suggest something like this:

```
remote_receiver:
  ...
  rc_switch:   # define custom bit timing parameters in usec for rc_switch
    sync-low: 820
    sync_high: 880
    zero_low: 220
    zero_high: 480
    one_low: 480
    one_high: 220
    inverted: false
  ...
```
With this parameters a user would be able to define a custom bit timing used with rc_switch. Similar I suggest to set parameters for the remote_transmitter.