esphome/issues #5541: speed_fan.h includes

kr0ner, 2024-02-26:
### The problem

SpeedFan should not require output component anymore. 
> output (Optional, [ID](https://esphome.io/guides/configuration-types#config-id)): The id of the [float output](https://esphome.io/components/output/#output) to use for this fan.

Omitting the output leads to compilation errors due to missing headers, cause there is still an implicit dependency to output component.

### Which version of ESPHome has the issue?

ESPHome 2024.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.3

### What platform are you using?

ESP8266

### Board

NodeMCU v2

### Component causing the issue

speed

### Example YAML snippet

```yaml
esphome:
  name: test
  friendly_name: test

esp8266:
  board: nodemcuv2

# Enable logging
logger:

fan:
  - platform: speed
    name: my_fan
    id: my_fan
    speed_count: 3
    on_speed_set:
      - logger.log: "Fan Speed was changed!"
```

### Anything in the logs that might be useful for us?

```txt
In file included from src/esphome/components/speed/fan/speed_fan.cpp:1:
src/esphome/components/speed/fan/speed_fan.h:6:10: fatal error: esphome/components/output/binary_output.h: No such file or directory
    6 | #include "esphome/components/output/binary_output.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test/src/esphome/components/speed/fan/speed_fan.cpp.o] Error 1
========================= [FAILED] Took 28.12 seconds =========================
```
```

kr0ner, 2024-02-26:
can be worked around by adding a dummy output

```
 output:
  - platform: gpio
    pin: D1
    id: gpio_d1
```

ssieb, 2024-02-26:
I guess I only tested removing the output from the fan, but not the output from the config.
A better workaround is to use a template output, then you don't need a pin, but that's exactly what I was trying to avoid with the change in the first place.  I'll fix this shortly.

kr0ner, 2024-02-26:
No worries ... there is a workaround :) thanks for the effort 👍

closed 2024-03-05 by jesserockz