esphome/issues #5543: ESP Offline, cant update

nmeliasp, 2024-02-26:
### The problem

My ESP device is showing offline in Home Assistant and i am unable to install updates. When looking at the logs, i see the following warning message 
INFO ESPHome 2024.2.0
INFO Reading configuration /config/esphome/xxxx.yaml...
INFO Starting log output from xxxx.local using esphome API
WARNING Can't connect to ESPHome API for xxxx.local: Error resolving IP address: [Errno -5] No address associated with hostname (APIConnectionError)

i confirmed my device is connected to Wifi and i am able to ping the IP address. 

Please Help!

### Which version of ESPHome has the issue?

2024.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.3

### What platform are you using?

ESP32

### Component causing the issue

api

ssieb, 2024-02-26:
github issues are not for support requests.  Come ask on the esphome discord server.

closed 2024-02-26 by ssieb

nmeliasp, 2024-02-26:
i'm getting an unable to accept invite for the discord server. Is there another way i can join it?

ssieb, 2024-02-26:
Try this one: https://discord.gg/95YZUQmX