esphome/issues #5544: INPUT_PULLUP no longer works with MCP23017

petewill, 2024-02-27:
### The problem

I am having an issue with my MCP23017 I/O expanders seemingly no longer activating the pullup resistor.  I recently added an additional window sensor to my home and needed to add it to my ESPHome device.  I have other door/window sensors already configured on the same device using the same I/O expander so I just needed to copy/past another binary sensor in the yaml code.  After I added the additional window sensor, I uploaded the code and all my sensors stopped reporting open/closed.  If I connect 3.3v to the pin the sensor will show as open so I know the expander is still functioning.  It seems that somehow the pullup functionality stopped working.  

Additional info:

- I have my contact sensors connected to ground and the I/O pin on the MCP23017 so I'm trying to use the internal pullup resistor.  
- I don't update the code on my devices every time ESPHome gets updated so I'm not sure when this issue started.
- I am using an Ethernet LAN8720 (WT32-ETH01) device that uses 3 MCP23017 I/O expanders. They are all showing as connected in the logs.
- I don't know of a way to roll back to a prior version without restoring from backup so I haven't been able to test running the same code on an older version of ESPHome.

### Which version of ESPHome has the issue?

2024.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.2

### What platform are you using?

ESP32

### Board

WT32-ETH01

### Component causing the issue

MCP23017

### Example YAML snippet

```yaml
esphome:
  name: ethernetsensors

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: VERBOSE
#  level: NONE

# Enable Home Assistant API
api:
  encryption:
    key: xxxxxxxxxxxxx

ota:
  password: !secret esp_ota_pw

#Configure Ethernet
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16
  use_address: 192.168.100.100

# I2C Configuration
i2c:
  sda: 15
  scl: 14
  scan: true
  id: bus_a

# MCP23017 Config
mcp23017:
  - id: 'mcp23017_hub1'
    address: 0x20

  - id: 'mcp23017_hub2'
    address: 0x21

  - id: 'mcp23017_hub3'
    address: 0x24

# Individual inputs BINARY
binary_sensor:

#Contact Sensor Hub 3
  - platform: gpio
    name: "Guest Window"
    device_class: window
    pin:
      mcp23xxx: mcp23017_hub3
      # Use pin A0 (A0 = 0, B7 = 15)
      number: 6 #A6
      # One of INPUT or INPUT_PULLUP
      mode:
        input: true
        pullup: true
      inverted: false
    filters:
      - delayed_on_off: 50ms #Debounce
```

randybb, 2024-02-27:
Just tested that - no issues. Anyway, nobody touched that component for last 2 years.

petewill, 2024-03-01:
@randybb Thank you for testing/confirming.  I can confirm this is my error.  I found that I had to have the Reset pin pulled high on these boards.  Once I did that my pullup resistors started working again.  I'm not sure why this just broke now after all these months but at least it's working again.  I also don't know how I missed this when I originally looked at the wiring diagram.  I suspect I found one that did not have the reset pin connected.  Anyway, if anyone has this issue in the future make sure you connect 3.3v to the reset pin.

closed 2024-03-01 by petewill