esphome/issues #5545: ESP32-S3 N32R8V reboot loop

Lateralleap, 2024-02-27:
### The problem

There is an ongoing issue the the Espressif ESP32-S3 DevKitC 1 N328RV board entering a reboot loop after configuration download or reset

I have tried all the YAML configurations I can find documented for the N16R8 and N8R8 board variants

This behaviour is exhibited under both Arduino and Esp-idf platforms

This issue is documented in this [issue](https://github.com/esphome/issues/issues/4149) but i has been marked as stale

### Which version of ESPHome has the issue?

2024.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.4

### What platform are you using?

ESP32

### Board

Espressif ESP32-S3 DevKitC-1 N32R8V

### Component causing the issue

N/A

### Anything in the logs that might be useful for us?

```txt
Please see this issue: https://github.com/esphome/issues/issues/4149
```

MHeuvel, 2024-03-04:
Same here, I can only get

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377474
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8

No matter what minimalistic config. I can flash tasmota esp32s3-firmware version and it works so the hardware is good.
It did work with ESPHome a few times but failed to work with voice assistant examples reported to work. After a third attempt it's no longer possible to get ANY ESPHome on it.

ssieb, 2024-03-04:
You can set arbitrary platformio settings in the `esphome:` section.
https://esphome.io/components/esphome#esphome-platformio-options

Lateralleap, 2024-03-08:
Thanks both. Hopefully this issue will get picked up and support for this device added to ESPHome. Espressif ESP32-S3 DevKitC 1 N8R8 works fine

ssieb, 2024-03-08:
Did you try setting the options as indicated?  I'm pretty sure the device is supported.  But we can't have specific options for every board.  You need to set options for your specific board as needed.

Lateralleap, 2024-03-08:
I understand that but surely these Espressif boards are mainstream?

I have read conflicting reports of the settings that might work outside of
ESPHone. I will try a few and let you know

Thanks

Paul Clarke

ssieb, 2024-03-08:
I think the only one you need is the memory type.  Maybe the cache one, but I don't know what that one is for.  The PSRAM and CDC are handled automatically by esphome according to the config.

Lateralleap, 2024-03-08:
OK, I will try taking the others out

Lateralleap, 2024-03-08:
You are correct

These settings also appear to work and with full flash size:
```
esphome:
  name: n32r8v-test
  friendly_name: N32R8V Test
  platformio_options:
    board_build.arduino.memory_type: opi_opi

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
  flash_size: 32MB
```

closed 2024-03-09 by ssieb

SeByDocKy, 2024-03-31:
Is the ```platformio_options:
    board_build.arduino.memory_type: opi_opi``` required with ESP-IDF ?

cowboyrushforth, 2024-04-06:
Regarding ESP-IDF, I had to experiment quite a bit to get it work for the 32MB version, hopefully this helps @SeByDocKy 

esphome:
  name: esp32s3devkit1
  friendly_name: esp32s3devkit1
  platformio_options:
    board_build.flash_mode: dio
    board_build.arduino.memory_type: opi_opi
    board_upload.maximum_ram_size: 524288
    build_flags: 
      -DBOARD_HAS_PSRAM 
      -DARDUINO_USB_CDC_ON_BOOT=1
      -mfix-esp32-psram-cache-issue

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 32MB
  partitions: "/config/esphome/32mb.csv"
  framework:
    type: esp-idf
    version: latest
    platform_version: "6.6.0"
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_32MB: "y"

SeByDocKy, 2024-04-06:
Thx ... where do you find the 32mb.csv file ?

cowboyrushforth, 2024-04-06:
@SeByDocKy 

this is the one i used: https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/large_littlefs_32MB.csv

there is others in that folder, too

SeByDocKy, 2024-04-06:
> @SeByDocKy
> 
> this is the one i used: https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/large_littlefs_32MB.csv
> 
> there is others in that folder, too

Ok thx, I will try with .... :)