esphome/issues #5546: ESPHome Device Goes Offline after OTA Update After update to 2024.2.0 On Beken Chip

KnoaITC, 2024-02-27:
### The problem

OTA Compiles and the installs. Device reboots and does not com online. Logs starts after the update and below stops but continuous ping fails. The device comes back online after about 5 Mins.

### Which version of ESPHome has the issue?

 2024.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.4

### What platform are you using?

BK72XX

### Board

board: generic-bk7231n-qfn32-tuya

### Example YAML snippet

```yaml
substitutions:
  name: "knoapwmsocket1relay"
  friendly_name: "Knoa PWM Socket 1 Relay"
  comment: Knoa PWM Socket 1 Relay
  relay_restore_mode: 'RESTORE_DEFAULT_ON'
  sensor_prefix: "Relay:"
  iot_dnsdomain: "knoaiot.local"
  ap_fallbackssid: "${name}-ap"
  ap_fallbackpw: "xxxxxx"
  connect_address: "${name}.${iot_dnsdomain}"

esphome:
  name: ${name}
  comment: ${comment}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false

bk72xx:
  board: generic-bk7231n-qfn32-tuya

# Enable logging
logger:

debug:
  update_interval: 30s  

api:

ota:

web_server:
  port: 80

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: ${connect_address}

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${ap_fallbackssid}"
    password: "${ap_fallbackpw}"
    
captive_portal:

button:
  - platform: restart
    name: Restart

output:
  - platform: gpio
    id: relay_output
    pin: P7

light:
  - platform: status_led
    id: led
    name: "${sensor_prefix} Blue LED"
    #disabled_by_default: true
    pin:
      number: P26
#      inverted: true

switch:
  - platform: output
    id: mini_relay
    output: relay_output
    name: "${sensor_prefix} Switch"
    restore_mode: ${relay_restore_mode}
    on_turn_on:
      - light.turn_on: led
    on_turn_off:
      - light.turn_off: led

uart:
  id: uart_bus
  tx_pin: P11
  rx_pin: P10
  baud_rate: 4800
  stop_bits: 1

sensor:
  - platform: bl0942
    uart_id: uart_bus
    current:
      name: "${sensor_prefix} Current"
    voltage:
      name: "${sensor_prefix} Voltage"
    power:
      name: "${sensor_prefix} Power"
      filters:
        multiply: 0.93
    energy:
      name: "${sensor_prefix} Energy"
    frequency:
      name: "${sensor_prefix} Frequency"
      accuracy_decimals: 2
    update_interval: 30s

  - platform: uptime
    name: "${sensor_prefix} Uptime"
    #disabled_by_default: true

  - platform: wifi_signal
    name: "${sensor_prefix} WiFi Signal"
    update_interval: 60s

binary_sensor:
  # Wired switch
  - platform: gpio
    id: the_switch
    name: "${sensor_prefix} Power Switch"
    #disabled_by_default: true
    pin:
      number: P6
      mode: INPUT_PULLUP
    on_state:
      - light.toggle: led
      - switch.toggle: mini_relay
    
  # Button on mini switch
  - platform: gpio
    pin:
      number: P23
      mode: INPUT_PULLUP
      inverted: true
    name: "${sensor_prefix} Power Button"
    #disabled_by_default: true
    on_multi_click:
      - timing:
          - ON for at most 1s
          - OFF for at least 0.2s
        then:
          - switch.toggle: mini_relay
          - light.toggle: led

  - platform: status
    name: "${sensor_prefix} Status"

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "${sensor_prefix} IP Address"

  - platform: debug
    reset_reason:
      name: "${sensor_prefix} Reset Reason"

  - platform: libretiny
    version:
      name: LibreTiny Version

time:
  - platform: sntp
    id: my_time
```

### Anything in the logs that might be useful for us?

```txt
========================= [SUCCESS] Took 9.30 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 100.75.50.171
INFO Uploading /data/build/esp-livingroomtvrelay/.pioenvs/esp-livingroomtvrelay/firmware.uf2 (3095552 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 29.32 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 100.75.50.171 using esphome API
INFO Successfully connected to esp-livingroomtvrelay @ 100.75.50.171 in 15.630s
INFO Successful handshake with esp-livingroomtvrelay @ 100.75.50.171 in 0.105s
[11:34:54][I][app:102]: ESPHome version 2024.2.1 compiled on Feb 27 2024, 10:51:50
[11:34:54][C][status_led:065]: Status Led Light:
[11:34:54][C][status_led:066]:   Pin: 26
[11:34:54][C][wifi:577]: WiFi:
[11:34:54][C][wifi:409]:   Local MAC: C4:82:E1:C4:F7:1F
[11:34:54][C][wifi:414]:   SSID: [redacted]
[11:34:54][C][wifi:415]:   IP Address: 100.75.50.171
[11:34:54][C][wifi:416]:   BSSID: [redacted]
[11:34:54][C][wifi:418]:   Hostname: 'esp-livingroomtvrelay'
[11:34:54][C][wifi:420]:   Signal strength: -67 dB ▂▄▆█
[11:34:54][C][wifi:424]:   Channel: 6
[11:34:54][C][wifi:425]:   Subnet: 255.255.255.0
[11:34:54][C][wifi:426]:   Gateway: 100.75.50.1
[11:34:54][C][wifi:427]:   DNS1: 100.75.50.2
[11:34:54][C][wifi:428]:   DNS2: 100.75.50.2
[11:34:54][C][logger:447]: Logger:
[11:34:54][C][logger:448]:   Level: DEBUG
[11:34:54][C][logger:449]:   Log Baud Rate: 115200
[11:34:54][C][logger:451]:   Hardware UART: UART2
[11:34:54][C][uart.lt:101]: UART Bus:
[11:34:54][C][uart.lt:102]:   Type: hardware
[11:34:54][C][uart.lt:104]:   Port number: 1
[11:34:54][C][uart.lt:106]:   TX Pin: 11
[11:34:54][C][uart.lt:107]:   RX Pin: 10
[11:34:54][C][uart.lt:109]:   RX Buffer Size: 256
```

donburch888, 2024-03-12:
Sorry, but I don't understand what you wrote as the description of your problem
... however I did notice in the log you provided ...

The install **did** compile and upload. 
But from the different timestamp and compile time, it looks as though your device did not activate the newly compiled and downloaded firmware

> [11:34:54][I][app:102]: ESPHome version 2024.2.1 compiled on Feb 27 2024, 10:51:50

I find this often with my Beken based Arlec PC191HA smart power sockets, and thought it might be specific to the Arlec units. 

I find that I have to power off the units for a while, and hopefully when powered back on there is a delay of a few minutes while it activates the new firmware before restarting.  This is verified by looking at the ESPHome version and compile date/time displayed when you request [LOGS]

If it still reports the old firmware, a longer period powered off is required.  I have left some stubborn units mine powered off overnight :-(

szupi-ipuzs, 2024-03-12:
You might want to report this issue in [libretiny repo](https://github.com/libretiny-eu/libretiny/issues), since this is the code that supports Beken chips in esphome.