esphome/issues #5547: ESPHome not supporting all device classes suppoered by Home Assistant; Volume Flow

bashers222, 2024-02-27:
### The problem

Home Assistant now supports Volume flow rate (litres per minutre), rather than volume (litres)
I have a water flow meter that has been erroring for the last 12 months due to an unsupported device class for volume flow
HA now supports this device class (https://github.com/home-assistant/architecture/discussions/915) , but ESPhome marks my yaml as invalid when entered there

- SensorDeviceClass.VOLUME_FLOW_RATE | m³/h, ft³/min, L/min, gal/min | Volume flow rate, this device class should be used for sensors representing a flow of some volume, for example the amount of water consumed momentarily.

Can w eget this added to ESPhome so it does not error

### Which version of ESPHome has the issue?

20240207.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

20240207.1

### What platform are you using?

ESP8266

### Component causing the issue

ESPhome web page validation

### Example YAML snippet

```yaml
- platform: pulse_meter
    pin:
      number: GPIO14
      inverted: true
      mode:
        input: true
        pullup: true
    name: "Water Flow Rate"
    internal_filter: 100ms
    device_class: volume
    unit_of_measurement: "L/min"
    icon: "mdi:water"
```

### Anything in the logs that might be useful for us?

```txt
HA Logs:
Entity sensor.water_flow_rate (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'L/min' which is not a valid unit for the device class ('volume') it is using; expected one of ['gal', 'ft³', 'L', 'mL', 'fl. oz.', 'm³', 'CCF']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22

ESPHome logs when validating:
INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/esp8266-04-watermeter.yaml...
Failed config

sensor.pulse_meter: [source <unicode string>:48]
  platform: pulse_meter
  pin: 
    number: GPIO14
    inverted: True
    mode: 
      input: True
      pullup: True
  name: Water Flow Rate
  internal_filter: 100ms
  
  Unknown value 'volume_flow_rate', did you mean 'volume_storage'?.
  device_class: volume_flow_rate
  unit_of_measurement: L/min
  icon: mdi:water
  total: 
    name: Water Meter Total
    unit_of_measurement: m³
    id: water_meter_total
    accuracy_decimals: 3
    device_class: water
    state_class: total_increasing
    filters: 
      - multiply: 0.001
```

MrIcemanLE, 2024-04-10:
@balloob this would be very useful. Who's the right person to mention this idea?

LucasCZE, 2025-11-24:
The `volume_flow_rate` device class was already added a long time ago so this issue can be closed.

[Synchronise Device Classes from Home Assistant #6158](https://github.com/esphome/esphome/pull/6158)