esphome/issues #5548: Cannot resolve pin name ******** for board *****

MStapelfeldt, 2024-02-27:
### The problem

I'm building a custom board, based on ESP32-S3-WROOM-1-N16R8 and there is no board type for a plain WROOM module, so I'm using variant esp32s3.
If you checkthe Datasheet, you see all avaible pins, even the once which should be used with care, but ESPhome doesn't know them:
```
Cannot resolve pin name 'GPI036' for board esp-wrover-kit.
Cannot resolve pin name 'GPI040' for board esp-wrover-kit.
```
Please look at https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf 
Page 10 Pin 29 & 33
Page 11&12:
| Name | No. | Type | Function |
| ------ | --- | --- | ------------- |
| IO36 | 29 | I/O/T | SPIIO7, GPIO36, FSPICLK, SUBSPICLK |
| IO40 | 33 | I/O/T | MTDO, GPIO40, CLK_OUT2 |

As you see these pins are there and avaible. Pin 26 GPIO45 shouldn't be used at all and Pin 16 GPIO46 is Input only (Maybe a warning `input only` instead of `strapping pin`).

### Which version of ESPHome has the issue?

ESPHome 2024.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.3

### What platform are you using?

ESP32

### Board

ESP32-S3-WROOM-1 ESP32-S3-WROOM-1U

### Component causing the issue

esp32

### Example YAML snippet

```yaml
esp32:
  board: esp-wrover-kit #esp32-s3-devkitc-1 
  variant: esp32s3
  flash_size: 16MB
  framework:
    type: arduino # esp-idf
    version: recommended
    
psram:
  mode: octal
  speed: 80MHz
```

randybb, 2024-02-27:
Ehm, GPI036 != GPIO36
Just use 36 if you don't know difference between 0 and O.

closed 2024-02-27 by randybb

MStapelfeldt, 2024-02-27:
> if you don't know difference between 0 and O.

What a beautiful answer! Yes it was a typo and I didn't see it. I'm sry for that. Still it would be nice to get a more respectful answer.

randybb, 2024-02-27:
It wasn't meant to be offensive - just use number instead as it is bit more cleaner.