esphome/issues #5549: Dallas sensor not found on Sonoff THR316

mangusbrother, 2024-02-27:
### The problem

Reopening this instead of https://github.com/esphome/issues/issues/5542
since the previous issue was closed without waiting for a reply based on one mistake in the description -_-

I have just flashed a sonoff thr316 with esp home.
End goal: be able to read the temp from the sensor in a card on home assistant.

It is wired to mains with the sensor connected (phone port, rj22 i think)

I am using the configruation from here: https://devices.esphome.io/devices/Sonoff-THR316

The sensor is not being found however.

### Which version of ESPHome has the issue?

2024.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.4 (OS 12.0)

### What platform are you using?

ESP32

### Board

sonoff thr316

### Component causing the issue

dallas

### Example YAML snippet

```yaml
esphome:
  name: esphome-web-e9aca0
  friendly_name: Aquarium Sonoff TH Origin

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable Home Assistant API
api:
  encryption:
    key: "encryptionKeyHere"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-E9Aca0"
    password: "passwordHere"

captive_portal:
  

logger:
  level: INFO
  baud_rate: 0

#optional
web_server:
  port: 80

time:
  - platform: homeassistant
    id: homeassistant_time

sensor:
  - platform: wifi_signal
    name: $friendly_name Wifi RSSI
    update_interval: 60s

  - platform: uptime
    id: uptime_sensor
    internal: True
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime_sensor).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? to_string(days) + "d " : "") +
                (hours ? to_string(hours) + "h " : "") +
                (minutes ? to_string(minutes) + "m " : "") +
                (to_string(seconds) + "s")
              ).c_str();

  - platform: template
    name: $friendly_name ESP32 Internal Temp
    device_class: temperature
    unit_of_measurement: °C
    id: esp32_temp
    lambda: return temperatureRead();

  # External DS18B20 Digital Temperature Sensor
  - platform: dallas
    index: 0
    name: $friendly_name Temperature
    id: temp

binary_sensor:
  - platform: gpio
    pin: GPIO00
    id: reset
    internal: true
    filters:
      - invert:
      - delayed_off: 10ms
    on_click:
      - max_length: 350ms # short press to toggle the relay
        then:
          - switch.toggle: relay
          # Turn off climate control
          - climate.control:
              id: climate_control
              mode: "OFF"
      - min_length: 360ms # long press to toggle climate control
        max_length: 3s
        then:
          - if:
              condition:
                lambda: |-
                  return id(climate_control).mode == CLIMATE_MODE_OFF;
              then:
                # We need to turn off realy due to a bug in the climate
                # component. If the relay is on and climate control is
                # turned on in idle mode (target temp < set temp) the relay
                # will not be turned off automatically. It works fine if
                # the climate control is turned on in heat mode (target temp >
                # set temp), i.e. the relay will be turned on.
                - switch.turn_off: relay
                - climate.control:
                    id: climate_control
                    mode: "HEAT"
              else:
                climate.control:
                  id: climate_control
                  mode: "OFF"
switch:
  - platform: gpio
    name: $friendly_name
    pin: GPIO21
    id: relay
    restore_mode: RESTORE_DEFAULT_OFF
    on_turn_on:
      - delay: 500ms
      - light.turn_on: switch_led
    on_turn_off:
      - delay: 500ms
      - light.turn_off: switch_led
  - platform: restart
    name: $friendly_name Restart
  # This is needed to power the external sensor.
  # It receives 3v3 from this pin, which is pulled up on boot.
  - platform: gpio
    pin: GPIO27
    id: sensor_power
    restore_mode: ALWAYS_ON

text_sensor:
  - platform: template
    name: $friendly_name Uptime
    id: uptime_human
    icon: mdi:clock-start
  - platform: wifi_info
    ip_address:
      name: $friendly_name IP
    ssid:
      name: $friendly_name SSID
    bssid:
      name: $friendly_name BSSID

light:
  - platform: status_led
    id: switch_led
    internal: True
    pin:
      number: GPIO16
      inverted: True
  - platform: status_led
    id: wifi_status_led
    internal: True
    pin:
      number: GPIO15
      inverted: True
  - platform: status_led
    id: auto_led
    internal: True
    pin:
      number: GPIO13
      inverted: True

interval:
  - interval: 30s
    then:
      if:
        condition:
          wifi.connected:
        then:
          - light.turn_on: wifi_status_led
        else:
          - light.turn_off: wifi_status_led

# DS18B20 Digital Temperature Sensor Hub
dallas:
  - pin: GPIO25
    update_interval: 10s

climate:
  - platform: thermostat
    id: climate_control
    name: $friendly_name Climate Control
    sensor: temp
    min_heating_off_time: 5s
    min_heating_run_time: 5s
    min_idle_time: 5s
    heat_deadband: 2
    heat_action:
      - switch.turn_on: relay
    idle_action:
      - switch.turn_off: relay
    off_mode:
      - switch.turn_off: relay
    default_preset: Home
    preset:
      - name: Home
        default_target_temperature_low: 20 °C
    on_state:
    - if:
        condition:
          lambda: |-
            return id(climate_control).mode == CLIMATE_MODE_OFF;
        then:
          - light.turn_off: auto_led
    - if:
        condition:
          lambda: |-
            return id(climate_control).mode == CLIMATE_MODE_HEAT;
        then:
          - light.turn_on: auto_led
```

### Anything in the logs that might be useful for us?

```txt
INFO Successful handshake with esphome-web-e9aca0 @ 192.168.3.100 in 0.103s
[21:06:35][I][app:102]: ESPHome version 2024.2.1 compiled on Feb 26 2024, 21:05:09
[21:06:35][W][dallas.sensor:080]:   Found no sensors!
[21:06:35][E][dallas.sensor:093]: Couldn't find sensor by index - not connected. Proceeding without it.
[21:06:44][E][dallas.sensor:112]: Requesting conversion failed
.... repeated some 20 more times
[21:11:24][E][dallas.sensor:112]: Requesting conversion failed
[21:11:27][I][ota:117]: Boot seems successful, resetting boot loop counter.
[21:11:34][E][dallas.sensor:112]: Requesting conversion failed
[21:11:44][E][dallas.sensor:112]: Requesting conversion failed
[21:11:54][E][dallas.sensor:112]: Requesting conversion failed
```

### Additional information

also note that the documentation here is misleading:
https://esphome.io/components/sensor/dallas.html

```
Configuration variables:
pin (Required, number): The pin the sensor bus is connected to. Please note that 1-wire is a bi-directional bus so it requires both input and output from the pin.
```

If pin accepts "GPIO25" (as seen in the example further down) it is not in fact a number

randybb, 2024-02-27:
Pin accepts number, but you can use an alias that gets converted to the right pin number as is described here https://esphome.io/guides/configuration-types.html#pin

Issues are not for help with configuration, so this is the reason why the other one was closed (and can be reopened). If you need help, join our discord channel and somebody will help you.

closed 2024-02-27 by randybb

mangusbrother, 2024-02-27:
if the default configuration that you have on your website is not working i wouldn't say this is a configuration issue really.. clearly something is wrong. 
I don't know how many pages i found from across the last year regarding some timing having changed in the dallas implemetnation in esp home causing it to stop working. 
Honestly, i'm not as invested in esp home to care at this point, if you want to play off a clear bug as a configuration issue go ahead.

randybb, 2024-02-27:
https://devices.esphome.io started as a separate site, later it was moved under esphome domain. These are not reference configs or officially supported devices, it is just a list of devices which somebody got working and provides an example config, which might be not valid anymore. As is mentioned on that page, if you spot a problem, you can fix it.

Yes, issues which nobody from devs can reproduce...

randybb, 2024-02-27:
Found an old crusty esp32 board, configured dallas for GPIO25 and...
```
[18:34:24][C][dallas.sensor:075]: DallasComponent:
[18:34:24][C][dallas.sensor:076]:   Pin: GPIO25
[18:34:24][C][dallas.sensor:077]:   Update Interval: 10.0s
[18:34:24][D][dallas.sensor:082]:   Found sensors:
[18:34:24][D][dallas.sensor:084]:     0x7f041750eb74ff28
...
[18:34:24][D][dallas.sensor:143]: 'Dev Temperature': Got Temperature=22.1°C
[18:34:24][D][sensor:094]: 'Dev Temperature': Sending state 22.12500 °C with 1 decimals of accuracy
[18:34:34][D][dallas.sensor:143]: 'Dev Temperature': Got Temperature=22.1°C
[18:34:34][D][sensor:094]: 'Dev Temperature': Sending state 22.12500 °C with 1 decimals of accuracy
```
So no issue with the dallas component with GPIO25.  If there is other problem in that config, like with powering that sensor or there is a missing 4k7 resistor (based on the datasheet)  between DQ and Vdd on your module, it is not esphome's fault.