esphome/issues #5551: wifi_info: address_0 required if address_1 needed

Nardol, 2024-02-28:
### The problem

To display both IPv4 and IPv6 addresses, there are new sub-sensors of `ip_address` in `wifi_info` component.
The main `ip_address` sensor already displays IPv4, so I did not removed it to avoid strange behavior with an entity not existing anymore in Home Assistant.
To avoid IPv4 displayed twice, I did not add `address_0` and only 1 and 2 to display both local and global IPv6.
IPv4 was displayed right but `both address_1` and 2 were displayed as unknown.
When I added an `address_0` sub-sensor which I marked as internal, my IPv6 addresses were also displayed.
It looks we have to put an `address_0` sensor if we need `address_1`. I have not tried to only defina 0 and 2 to see if address 2 is displayed or marked as unknown.

### Which version of ESPHome has the issue?

2024.3.0-dev

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2024.2.4

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

wifi_info

### Example YAML snippet

```yaml
esphome:
  name: diffuseur-bureau
  project:
    name: "Asakuki.500ml_diffuser"
    version: "1.5.0"
  on_boot:
    priority: 1
    then:
    - uart.write: [0x55, 0xAA, 0x00, 0x06, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x00, 0x0D]

esp8266:
  board: esp01_1m
  restore_from_flash: false

web_server:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_key

network:
  enable_ipv6: true
  min_ipv6_addr_count: 2

text_sensor:
- platform: wifi_info
  ip_address:
    name: Adresse IP Diffuseur bureau
    address_1:
      name: IPv6 locale
    address_2:
      name: IPv6 globale
```

### Additional information

Also observed by @jeremyp3 with an ESP32 az-delivery devkitc so not specific to ESP8266.
I have no ethernet device to try with.

Nardol, 2024-02-28:
I've just tested and it is the same if only defining address_0`` and `address_2` E.G. to only have the IPv4 and the global IPv6 address.

HeMan, 2024-02-29:
Could you please try my repo and branch `fix_sensors`?

Nardol, 2024-02-29:
I confirm the fix for both cases :+1:

Nardol, 2024-02-29:
This issue brings something to my mind.
`address_0` is always the IPv4, anyway every times I tested.
The main `ip_address` sensor is required.
So it is completely not possible to only display IPv6 without adding an internal sensor for the main `ip_address` sensor?

closed 2024-03-01 by jesserockz