esphome/issues #5554: ld returned 1 exit status *** [.pioenvs/hendoor/firmware.elf]

mbirrell66, 2024-02-29:
### The problem

when trying to install on a Pi Pico I get ld returned 1 exit status
*** [.pioenvs/hendoor/firmware.elf]. I have all the latest patches installed. Full error is:
[logs_hendoor_compile.txt](https://github.com/esphome/issues/files/14441798/logs_hendoor_compile.txt)

### Which version of ESPHome has the issue?

2024.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Latest stable release - last update done this morning

### What platform are you using?

RP2040

### Board

Pi Pico W

### Component causing the issue

see error log

### Anything in the logs that might be useful for us?

```txt
see attached
```

### Additional information

this has happened before. I update ESPHome and then it breaks. This time I have done a couple of updates so I don't know which one broke it.

ssieb, 2024-02-29:
It's an issue with platformio and/or the pico toolchain.  You need to re-install the esphome addon.

mbirrell66, 2024-02-29:
> It's an issue with platformio and/or the pico toolchain. You need to re-install the esphome addon.

thanks for your quick response... will cause me some pain but will do that now and let you know the result.

ssieb, 2024-02-29:
Shouldn't be any pain.  Should be quite quick and easy.

mbirrell66, 2024-02-29:
thanks again for your help. That fixed the issue. I have quite a few devices so I backed them all up before I did it.

closed 2024-02-29 by mbirrell66