esphome/issues #5556: micro-wake-word-models file not found

ebw44, 2024-02-29:
### The problem

Trying to compile a simple micro_wake_word example I get the error below
```
  File "C:\Users\username\mambaforge\envs\esphome\Lib\site-packages\esphome\components\micro_wake_word\__init__.py", line 292, in _load_model_data
    with open(model_path, "rb") as f:
         ^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'okay_nabu.tflite'
```

After stepping through the code, it looks like the issue is somewhere in the call to `urllib.parse.urljoin` line 290 of function `_load_model_data` that return just the name of the tflite file instead of the full path.

The solutions was to download the 'okay_nabu.tflite' in the folder where I run esphome. The other one was to replace `model_path = urljoin(str(manifest_path), manifest[CONF_MODEL])` by `model_path = manifest_path.parent / manifest[CONF_MODEL]`.

### Which version of ESPHome has the issue?

2024.2.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32-IDF

### Board

NA

### Component causing the issue

micro_wake_word

### Example YAML snippet

```yaml
esphome:
  name: kitchen-proxy

esp32:
  board: esp32s3box
  variant: esp32s3
  framework:
    type: esp-idf

micro_wake_word:
  model: okay_nabu
```

stefancvetkovic, 2024-03-22:
is this fix released in version 2024.3.0 CLI ?

ebw44, 2024-03-25:
@stefancvetkovic, I have created a [pull request](https://github.com/esphome/esphome/pull/6426)