esphome/issues #5560: Animation doesn't work with esp32s3

scarolan, 2024-03-02:
### The problem

Trying out the example from the docs on an M5Stack AtomS3.  This results in errors when compiling:

```
INFO Compiling app...
Processing atoms3 (board: m5stack-atoms3; framework: espidf; platform: platformio/espressif32@6.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.50101.230828 (5.1.1) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
Compiling .pioenvs/atoms3/src/main.o
Linking .pioenvs/atoms3/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/atoms3/src/main.o:(.literal._ZNSt17_Function_handlerIFvRN7esphome7display7DisplayEEZ5setupvEUlS3_E_E9_M_invokeERKSt9_Any_dataS3_+0x4): undefined reference to `_ZN7esphome9animation9Animation10next_frameEv'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/atoms3/src/main.o:(.literal._Z5setupv+0x2d4): undefined reference to `_ZN7esphome5image5ImageC1EPKhiiNS0_9ImageTypeE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/atoms3/src/main.o:(.literal._Z5setupv+0x2d8): undefined reference to `_ZN7esphome9animation9AnimationC1EPKhiimNS_5image9ImageTypeE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/atoms3/src/main.o: in function `_ZNSt17_Function_handlerIFvRN7esphome7display7DisplayEEZ5setupvEUlS3_E_E9_M_invokeERKSt9_Any_dataS3_':
/config/esphome/atoms3.yaml:174: undefined reference to `_ZN7esphome9animation9Animation10next_frameEv'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/atoms3/src/main.o: in function `_Z5setupv':
/data/build/atoms3/src/main.cpp:481: undefined reference to `_ZN7esphome5image5ImageC1EPKhiiNS0_9ImageTypeE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /data/build/atoms3/src/main.cpp:484: undefined reference to `_ZN7esphome9animation9AnimationC1EPKhiimNS_5image9ImageTypeE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /data/build/atoms3/src/main.cpp:488: undefined reference to `_ZN7esphome9animation9AnimationC1EPKhiimNS_5image9ImageTypeE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /data/build/atoms3/src/main.cpp:491: undefined reference to `_ZN7esphome9animation9AnimationC1EPKhiimNS_5image9ImageTypeE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /data/build/atoms3/src/main.cpp:500: undefined reference to `_ZN7esphome9animation9AnimationC1EPKhiimNS_5image9ImageTypeE'
collect2: error: ld returned 1 exit status
*** [.pioenvs/atoms3/firmware.elf] Error 1
========================= [FAILED] Took 19.04 seconds =========================
```

### Which version of ESPHome has the issue?

ESPHome 2024.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Current

### What platform are you using?

ESP32

### Board

M5Stack AtomS3

### Component causing the issue

animation

### Example YAML snippet

```yaml
animation:
  - file: "_gifs_128x128/blinky.gif"
    id: blinky
    resize: 128x128

display:
  - platform: st7789v # https://esphome.io/components/display/st7789v.html
    id: disp
    model: Custom
    backlight_pin: GPIO16
    cs_pin: GPIO15
    dc_pin: GPIO33
    reset_pin: GPIO34
    height: 128
    width: 128
    offset_height: 2
    offset_width: 1
    eightbitcolor: true
    update_interval: 3s
    lambda: |-
      //Ingress shown animation Frame.
      id(blinky).next_frame();
      // Draw the animation my_animation at position [x=0,y=0]
      it.image(0, 0, id(blinky), COLOR_ON, COLOR_OFF);
```

### Anything in the logs that might be useful for us?

```txt
INFO Compiling app...
Processing atoms3 (board: m5stack-atoms3; framework: espidf; platform: platformio/espressif32@6.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.50101.230828 (5.1.1) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
Compiling .pioenvs/atoms3/src/main.o
Linking .pioenvs/atoms3/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/atoms3/src/main.o:(.literal._ZNSt17_Function_handlerIFvRN7esphome7display7DisplayEEZ5setupvEUlS3_E_E9_M_invokeERKSt9_Any_dataS3_+0x4): undefined reference to `_ZN7esphome9animation9Animation10next_frameEv'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/atoms3/src/main.o:(.literal._Z5setupv+0x2d4): undefined reference to `_ZN7esphome5image5ImageC1EPKhiiNS0_9ImageTypeE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/atoms3/src/main.o:(.literal._Z5setupv+0x2d8): undefined reference to `_ZN7esphome9animation9AnimationC1EPKhiimNS_5image9ImageTypeE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/atoms3/src/main.o: in function `_ZNSt17_Function_handlerIFvRN7esphome7display7DisplayEEZ5setupvEUlS3_E_E9_M_invokeERKSt9_Any_dataS3_':
/config/esphome/atoms3.yaml:174: undefined reference to `_ZN7esphome9animation9Animation10next_frameEv'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/atoms3/src/main.o: in function `_Z5setupv':
/data/build/atoms3/src/main.cpp:481: undefined reference to `_ZN7esphome5image5ImageC1EPKhiiNS0_9ImageTypeE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /data/build/atoms3/src/main.cpp:484: undefined reference to `_ZN7esphome9animation9AnimationC1EPKhiimNS_5image9ImageTypeE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /data/build/atoms3/src/main.cpp:488: undefined reference to `_ZN7esphome9animation9AnimationC1EPKhiimNS_5image9ImageTypeE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /data/build/atoms3/src/main.cpp:491: undefined reference to `_ZN7esphome9animation9AnimationC1EPKhiimNS_5image9ImageTypeE'
/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /data/build/atoms3/src/main.cpp:500: undefined reference to `_ZN7esphome9animation9AnimationC1EPKhiimNS_5image9ImageTypeE'
collect2: error: ld returned 1 exit status
*** [.pioenvs/atoms3/firmware.elf] Error 1
========================= [FAILED] Took 19.04 seconds =========================
```
```

randybb, 2024-03-02:
This has nothing to do with animation => "Clean build files"

scarolan, 2024-03-02:
You're right!  🤦  Clean build files got me unstuck, thanks!

closed 2024-03-02 by scarolan