esphome/issues #5561: Deleted switches still provided to Home Assistant

tomlut, 2024-03-02:
### The problem

I changed some switches to buttons. All went well buttons work great, except the switches are still being provided by ESPHome to Home Assistant and so they can not be deleted.

[image]

[image]

### Which version of ESPHome has the issue?

2024.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2024.2.5

### What platform are you using?

ESP32

### Board

QinLED ESP32

### Component causing the issue

Switch

### Example YAML snippet

```yaml
## Before
switch:
  - platform: restart
    name: "Doorbell Upstairs Restart"

# After
button:
  - platform: restart
    name: "Doorbell Upstairs Restart"
```

nagyrobi, 2024-03-02:
@bdraco

bdraco, 2024-03-02:
https://github.com/home-assistant/core/pull/111076

closed 2024-07-01 by tomlut