esphome/issues #5564: Unable to compile for framework type esp-idf

clarencelocke, 2024-03-02:
### The problem

ESPHome Version 2024.2.1
macOS 13.5.1 on M1 Apple Silicon
Python 3.12.2

I am evaluating a Olimex ESP32-POE board and a BME680 temp sensor.  Building with the following configuration:
```
esp32:
  board: esp32-poe
  framework:
    type: arduino
```

successfully compiles and runs on the board.  I am currently tracking down an issue related to hard-wired ethernet failing to connect at (re)boot and attempted to explore a solution / work-around as hinted at by:

> [15:42:26][W][ethernet:155]: Connecting via ethernet failed! Re-connecting...
>    [15:42:26][D][esp-idf:000]: E (173891) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size
>
>    https://github.com/platformio/platform-espressif32/issues/1309#issuecomment-1943577881

which required changing the framework type to `esp-idf`.

When doing so and recompiling the project (clean build, completely removed the `.esphome` directory), I encountered the following error:

```
ModuleNotFoundError: No module named 'chardet'
```

### Which version of ESPHome has the issue?

2024.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

Olimex ESP32-POE

### Example YAML snippet

```yaml
substitutions:
  name: environment
  friendly_name: Environment

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: true
  project:
    name: "Environment.BME680"
    version: "1.0.0"

esp32:
  board: esp32-poe
  framework:
#    type: arduino
    type: esp-idf

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12

api:
logger:
ota:
```

### Anything in the logs that might be useful for us?

```txt
$ esphome compile bug.yaml 
INFO ESPHome 2024.2.1
INFO Reading configuration bug.yaml...
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing environment (board: esp32-poe; framework: espidf; platform: platformio/espressif32@5.4.0)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40406.240122 (4.4.6) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.3 (Apple Git-145)") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/clarence/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/clarence/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/clarence/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Users/clarence/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /Users/clarence/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/Users/clarence/Documents/Development/HomeAutomation/olimex-poe/.esphome/build/environment/.pioenvs/environment/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at /Users/clarence/.platformio/packages/framework-espidf/tools/cmake/build.cmake:463 (message):
  Traceback (most recent call last):

    File "/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/requests/compat.py", line 11, in <module>
      import chardet

  ModuleNotFoundError: No module named 'chardet'

  

  During handling of the above exception, another exception occurred:

  

  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/idf_component_manager/prepare_components/__main__.py", line 4, in <module>
      from .prepare import main
    File "/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/idf_component_manager/prepare_components/prepare.py", line 18, in <module>
      from ..core import ComponentManager
    File "/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/idf_component_manager/core.py", line 18, in <module>
      import requests
    File "/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/requests/__init__.py", line 45, in <module>
      from .exceptions import RequestsDependencyWarning
    File "/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/requests/exceptions.py", line 9, in <module>
      from .compat import JSONDecodeError as CompatJSONDecodeError
    File "/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/requests/compat.py", line 13, in <module>
      import charset_normalizer as chardet
    File "/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/charset_normalizer/__init__.py", line 24, in <module>
      from .api import from_bytes, from_fp, from_path, is_binary
    File "/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/charset_normalizer/api.py", line 5, in <module>
      from .cd import (
    File "/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/charset_normalizer/cd.py", line 14, in <module>
      from .md import is_suspiciously_successive_range

  ImportError:
  dlopen(/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/charset_normalizer/md.cpython-312-darwin.so,
  0x0002): tried:
  '/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/charset_normalizer/md.cpython-312-darwin.so'
  (mach-o file, but is an incompatible architecture (have 'arm64', need
  'x86_64')),
  '/System/Volumes/Preboot/Cryptexes/OS/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/charset_normalizer/md.cpython-312-darwin.so'
  (no such file),
  '/Users/clarence/.platformio/penv/.espidf-4.4.6/lib/python3.12/site-packages/charset_normalizer/md.cpython-312-darwin.so'
  (mach-o file, but is an incompatible architecture (have 'arm64', need
  'x86_64'))

Call Stack (most recent call first):
  /Users/clarence/.platformio/packages/framework-espidf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:3 (project)
```

randybb, 2024-03-03:
Looks like this is caused by a dependency for platformio which is broken on M1 https://github.com/platformio/platformio-core/issues/4702 Probably you can try to install it manually: https://esp32.com/viewtopic.php?t=32884
Definitely not something that we can fix in esphome - we need to wait until it is fixed in the upstream.

clarencelocke, 2024-03-03:
Finding the right directory for my environment and performing the `pip install` from within did the trick.  Thanks!

closed 2024-03-03 by clarencelocke