esphome/issues #5565: If the Chinese name is the same length, the device will not be displayed on the web page.

Ainndy, 2024-03-03:
### The problem

https://github.com/esphome/issues/assets/37214762/393b63c3-77dd-4694-b92c-8d3c5e145ab9

### Which version of ESPHome has the issue?

2024.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.5

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

all

### Example YAML snippet

```yaml
web_server:
  port: 80

binary_sensor:
  - platform: gpio
    pin:
      number: D1
      mode:
        input: true
        pullup: true
    name: "水位传感器"
    id: water_gpio

  - platform: gpio
    pin:
      number: D7
      inverted: true
      mode:
        input: true
        pullup: true
    name: "指示灯开关"  
    id: water_gpio_switch
```

catsimple, 2024-04-23:
SAME ISSUE.

ZDZX-T, 2024-11-29:
原来是因为字数相同，我还以为是因为“温度”编码被误识别了（上下文有相同字数的“湿度”）。也算是有曲线救国的方法了。

ZoftTy, 2024-12-25:
SAME ISSUE.

Ass-amite, 2025-04-16:
我也遇到同样的问题，找了几天发现是读取yaml时object_id的生成方式的原因
```
async def setup_entity(var, config):
    """Set up generic properties of an Entity"""
    add(var.set_name(config[CONF_NAME]))
    if not config[CONF_NAME]:
        add(var.set_object_id(sanitize(snake_case(CORE.friendly_name))))
    else:
        add(var.set_object_id(sanitize(snake_case(config[CONF_NAME]))))
    add(var.set_disabled_by_default(config[CONF_DISABLED_BY_DEFAULT]))
    if CONF_INTERNAL in config:
        add(var.set_internal(config[CONF_INTERNAL]))
    if CONF_ICON in config:
        add(var.set_icon(config[CONF_ICON]))
    if CONF_ENTITY_CATEGORY in config:
        add(var.set_entity_category(config[CONF_ENTITY_CATEGORY]))
```
这段代码会把优先使用名字来生成object_id，但是英文和数字以外会被当成非法字符转换成“_"
最终在HA里就会发生unique_id重复错误，导致只显示最后一个传感器。
这段代码来自cpp_helpers.py，把这里改掉或者使用英文然后在HA里改名字。

closed 2025-04-20 by Ainndy