esphome/issues #5566: ESP-IDF dev or custom version can't compile

DAVe3283, 2024-03-04:
### The problem

It is entirely possible that I am doing something wrong here, but I can't even get a simple test project to compile with ESP-IDF on anything but the "latest" or "recommended" versions. Using "dev" or a custom version fails with an AssertionError (details below).

I have a device that worked with ESPHome 2023.12.6 + ESP-IDF 5.1.2. Some changes in ESPHome 2024.2.x hit a bug in ESP-IDF that is fixed in 5.1.3, but it appears there is no way for me to actually use 5.1.3 without recompiling ESPHome.

But even this simple YAML below doesn't compile, and I would expect it should?

### Which version of ESPHome has the issue?

2024.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.2.5

### What platform are you using?

ESP32-IDF

### Example YAML snippet

```yaml
esphome:
  name: test
  friendly_name: test

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: dev

logger:

api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
AssertionError: Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/run/cli.py", line 145, in cli
    process_env(
  File "/usr/local/lib/python3.11/dist-packages/platformio/run/cli.py", line 208, in process_env
    ).process()
      ^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/run/processor.py", line 81, in process
    install_project_env_dependencies(
  File "/usr/local/lib/python3.11/dist-packages/platformio/package/commands/install.py", line 132, in install_project_env_dependencies
    _install_project_env_platform(project_env, options),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/package/commands/install.py", line 149, in _install_project_env_platform
    PlatformPackageManager().install(
  File "/usr/local/lib/python3.11/dist-packages/platformio/package/manager/platform.py", line 62, in install
    p.install_required_packages(force=force)
  File "/usr/local/lib/python3.11/dist-packages/platformio/platform/_packages.py", line 76, in install_required_packages
    self.install_package(name, force=force)
  File "/usr/local/lib/python3.11/dist-packages/platformio/platform/_packages.py", line 70, in install_package
    return self.pm.install(spec or self.get_package_spec(name), force=force)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/package/manager/_install.py", line 47, in install
    pkg = self._install(spec, skip_dependencies=skip_dependencies, force=force)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/package/manager/_install.py", line 91, in _install
    self.install_dependencies(pkg, print_header=False)
  File "/usr/local/lib/python3.11/dist-packages/platformio/package/manager/_install.py", line 139, in install_dependencies
    self.install_dependency(dependency)
  File "/usr/local/lib/python3.11/dist-packages/platformio/package/manager/_install.py", line 151, in install_dependency
    dependency_compatibility = PackageCompatibility.from_dependency(dependency)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/platformio/package/meta.py", line 72, in from_dependency
    assert isinstance(dependency, dict)
AssertionError

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

ssieb, 2024-03-05:
If you change the platform versions, you're on your own.  As the big warning when you compile says, it's not supported.
What is the change that hit an IDF bug?

closed 2024-03-05 by ssieb

randybb, 2024-03-05:
5.1.3 is not yet supported by [platformio](https://github.com/platformio/platform-espressif32/releases)

anyway, I am running most of my devices on 5.1.2 and no issues (ESPHome dev and 2024.2.1), but as ssieb mentioned - no support for that as ESPHome is still running on 4.4 and many components are not compatible with 5.x.

```yaml
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
    version: 5.1.2
    platform_version: 6.5.0
```

DAVe3283, 2024-03-05:
Maybe I confused the issue by mentioning a bug in ESP-IDF 5.1.2. The problem here is ESPHome seems unable to use a custom version, _even if the custom version is same as the hard-coded "latest"_. For example, dev and latest are currently identical, just one comes in as a package, and one comes from source: https://github.com/esphome/esphome/blob/357ac3b85f6cff3c93ec8a3c29daca1e945a4fda/esphome/components/esp32/__init__.py#L274

```py
lookups = {
    "dev": (cv.Version(5, 1, 2), "https://github.com/espressif/esp-idf.git"),
    "latest": (cv.Version(5, 1, 2), None),
    "recommended": (RECOMMENDED_ESP_IDF_FRAMEWORK_VERSION, None),
}
```

Selecting "latest" works, but selecting "dev" doesn't even start to compile, it immediately crashes. To me, this seems like an ESPHome bug, not an ESP-IDF bug, which is why I filed it. But I get that any non-recommended version is unsupported.

I have not yet figured out exactly which ESPHome change triggers the ESP-IDF bug, and hadn't planned to spend much time on it, because the bug was in ESP-IDF and is already fixed in the 5.1.3 release. If you are curious, this is the error I see:

```
/data/cache/platformio/packages/framework-espidf/components/esp_hw_support/sleep_modes.c: In function 'light_sleep_uart_prepare':
/data/cache/platformio/packages/framework-espidf/components/esp_hw_support/sleep_modes.c:152:61: error: 'CONFIG_ESP_CONSOLE_UART_BAUDRATE' undeclared (first use in this function); did you mean 'CONFIG_ESP_CONSOLE_UART_NUM'?
  152 | #define UART_FLUSH_US_PER_CHAR              (10*1000*1000 / CONFIG_ESP_CONSOLE_UART_BAUDRATE)
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/esp_hw_support/sleep_modes.c:484:113: note: in expansion of macro 'UART_FLUSH_US_PER_CHAR'
  484 |             (sleep_duration < (int64_t)((UART_LL_FIFO_DEF_LEN - uart_ll_get_txfifo_len(CONSOLE_UART_DEV) + 1) * UART_FLUSH_US_PER_CHAR) + SLEEP_UART_FLUSH_DONE_TO_SLEEP_US)) {
      |                                                                                                                 ^~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/esp_hw_support/sleep_modes.c:152:61: note: each undeclared identifier is reported only once for each function it appears in
  152 | #define UART_FLUSH_US_PER_CHAR              (10*1000*1000 / CONFIG_ESP_CONSOLE_UART_BAUDRATE)
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/esp_hw_support/sleep_modes.c:484:113: note: in expansion of macro 'UART_FLUSH_US_PER_CHAR'
  484 |             (sleep_duration < (int64_t)((UART_LL_FIFO_DEF_LEN - uart_ll_get_txfifo_len(CONSOLE_UART_DEV) + 1) * UART_FLUSH_US_PER_CHAR) + SLEEP_UART_FLUSH_DONE_TO_SLEEP_US)) {
      |                                                                                                                 ^~~~~~~~~~~~~~~~~~~~~~
*** [.pioenvs/soda-fountain/esp_hw_support/sleep_modes.o] Error 1
```

I believe it to be this bug: espressif/esp-idf#12314