esphome/issues #5570: "gethostbyname() argument 1 must be str, bytes or bytearray, not None" When trying upload from console

shammysha, 2024-03-06:
### The problem

Trying to upload project via ssh console, 

`esphome upload /opt/esphome/config/bathroom-pir.yaml`

But got error 

```
INFO ESPHome 2024.2.2
INFO Reading configuration /opt/esphome/config/bathroom-pir.yaml...
INFO Detected timezone 'Europe/Moscow'
INFO Resolving IP address of 
Traceback (most recent call last):
  File "/home/esphome/.local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/esphome/.local/lib/python3.11/site-packages/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/esphome/.local/lib/python3.11/site-packages/esphome/__main__.py", line 1028, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/esphome/.local/lib/python3.11/site-packages/esphome/__main__.py", line 429, in command_upload
    exit_code = upload_program(config, args, port)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/esphome/.local/lib/python3.11/site-packages/esphome/__main__.py", line 340, in upload_program
    return espota2.run_ota(host, remote_port, password, CORE.firmware_bin)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/esphome/.local/lib/python3.11/site-packages/esphome/espota2.py", line 353, in run_ota
    return run_ota_impl_(remote_host, remote_port, password, filename)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/esphome/.local/lib/python3.11/site-packages/esphome/espota2.py", line 317, in run_ota_impl_
    ip = resolve_ip_address(remote_host)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/esphome/.local/lib/python3.11/site-packages/esphome/helpers.py", line 145, in resolve_ip_address
    return socket.gethostbyname(urlparse(host_url).hostname)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: gethostbyname() argument 1 must be str, bytes or bytearray, not None
```

When I try to copy and repeat in console command, that used by interface...

`esphome --dashboard run /opt/esphome/config/bathroom-pir.yaml --device 192.168.1.206`

... Project succesfully compiled and uploaded 

But when I remove "device" parameter from command- upload fails

Seems, that upload script does not parse correctly "esphome"->"name" parameter anymore (substitution used, see code below)?

### Which version of ESPHome has the issue?

ESPHome 2024.2.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

No matter

### What platform are you using?

ESP32

### Board

any

### Example YAML snippet

```yaml
# ---------------------------
# bathroom-pir.yaml:
# ---------------------------
substitutions:
  devicename: "fish-dozer"
  pin_ccw: "GPIO12"
  pin_cw: "GPIO13"

packages:
  conf: !include
    file: config-template.yaml
      
esp8266:  
  board: esp12e
  restore_from_flash: True 

# ---------------------------
# config-template.yaml:
# ---------------------------

substitutions:
  wifi_use_address: ''
  log_level: INFO
  log_baud_rate: '115200'

esphome:
  name: ${devicename}
  
wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_pass
    - ssid: !secret wifi_garage_ssid
      password: !secret wifi_pass    
  
logger:
  level: ${log_level}
  baud_rate: ${log_baud_rate}
```

ssieb, 2024-03-06:
What is the output from `esphome config /opt/esphome/config/bathroom-pir.yaml`?

shammysha, 2024-03-06:
Got this..

https://github.com/esphome/esphome/blob/b3ff23ec76b2c4ce5f4072f1977da86741ec0a53/esphome/components/wifi/__init__.py#L247
https://github.com/esphome/esphome/blob/833affc1bf1adc3164a2db3dbd5933e45cd44d58/esphome/core/__init__.py#L553
https://github.com/esphome/esphome/blob/833affc1bf1adc3164a2db3dbd5933e45cd44d58/esphome/__main__.py#L96

Must set wifi_use_address in substitution to name+ doman

closed 2024-03-06 by shammysha