esphome/issues #5578: Updates alwasy break my displays

sabixx, 2024-03-08:
### The problem

Hi,

Here is some updates - I haven't figured out which. ESPHome, Home Assistant, HA Core or an update of Blueprint is breaking my panels. I'll have to install the firmware again to make them work again.

The displays look like this after the upgrade:
[image]

When I try to reinstall it it will fail, see the install log.

Many thanks in advance for any hints.

 

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
substitutions:

###### CHANGE ME START ######

  device_name: "nspanelmrt" 
  wifi_ssid: !secrets...
  wifi_password: !secrets...

  nextion_update_url: "http://192.168.2.25:8123/local/nspanel_eu.tft" # URL to local tft File
#  nextion_update_url: "https://raw.githubusercontent.com/Blackymas/NSPanel_HA_Blueprint/main/nspanel_eu.tft" # URL to Github

##### CHANGE ME END #####

##### DO NOT CHANGE ANYTHING! #####

packages:
  ##### download esphome code from Github
  remote_package:
    url: https://github.com/Blackymas/NSPanel_HA_Blueprint
    ref: main
    files: [nspanel_esphome.yaml]
    refresh: 300s

##### DO NOT CHANGE ANYTHING! #####
```

### Anything in the logs that might be useful for us?

```txt
here's the install log 

INFO ESPHome 2024.2.2
INFO Reading configuration /config/esphome/ns-panel-mrt.yaml...
INFO Updating https://github.com/Blackymas/NSPanel_HA_Blueprint@main
INFO Updating https://github.com/edwardtfn/esphome@nextion-v425
INFO Detected timezone 'Europe/Berlin'
[... 493 lines cut ...]
[17:30:30][I][script.watchdog:3306]:   Init:        True
[17:30:30][I][script.watchdog:3312]:   TFT:         4.2.2
[17:30:30][I][addon_upload_tft.script.watchdog:409]: Add-on Upload TFT:
[17:30:30][I][addon_upload_tft.script.watchdog:410]:   File model: Use nextion_update_url
[17:30:30][I][addon_upload_tft.script.watchdog:411]:   Valid TFT:  YES
[17:30:30][W][component:214]: Component time took a long time for an operation (0.09 s).
[17:30:30][W][component:215]: Components should block for at most 20-30ms.
[17:31:10][D][sensor:094]: 'RSSI': Sending state -77.00000 dBm with 0 decimals of accuracy
[17:31:18][D][text_sensor:064]: 'Version ESPHome': Sending state '4.2.6'
[17:31:18][D][text_sensor.version_esphome:1783]: ESPHome version: 4.2.6
[17:31:18][D][script:077]: Script 'check_versions' restarting (mode: restart)
[17:31:18][W][nextion:405]: Nextion reported variable name invalid!
[17:31:21][D][sensor:094]: 'ntc_source': Sending state 1.72700 V with 2 decimals of accuracy
[17:31:21][D][resistance:039]: 'resistance_sensor' - Resistance 12296.5Ω
[17:31:21][D][sensor:094]: 'resistance_sensor': Sending state 12296.50488 Ω with 1 decimals of accuracy
[17:31:21][D][ntc:026]: 'Temperature' - Temperature: 20.4°C
[17:31:21][D][sensor:094]: 'Temperature': Sending state 20.41909 °C with 1 decimals of accuracy
[17:31:22][I][ota:117]: Boot seems successful, resetting boot loop counter.
[17:31:22][D][esp32.preferences:114]: Saving 1 preferences to flash...
[17:31:22][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
```

### Additional information

It has happened a few times that it was broken after some update and typically, I could fix it by installing it or pressing the UpdateTFT button, but this time I can't figure out what is broken.

randybb, 2024-03-08:
It is logical that if you update one thing you need to update the other two as all three parts - ha blueprint, remote package and nextion fw are related each other. 
Anyway, this is an external component - so this is not the right place for such issues.

closed 2024-03-08 by randybb

sabixx, 2024-03-08:
thanks @randybb, didn't realize I posted it on the wrong repo..