esphome/issues #5579: RP2040 compile error with ESPHome 2024.2.2

on-kage, 2024-03-08:
### The problem

After upgrading to ESPHome 2024.2.2, I attempted to update my Raspberry Pico W device that is currently on ESPHome v2024.2.1. The PicoW is currently running as an irrigation controller and is still completely functional as is. I just wanted to keep the version up to date. I have also tried to "Clean Build Files" and retry with the same result

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.0

### What platform are you using?

RP2040

### Board

Raspberry Pi Pico W

### Example YAML snippet

```yaml
esphome:
  name: sprinklercontroller
  friendly_name: SprinklerController

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "<key>"

ota:
  password: "<pass>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.0.185
    gateway: 192.168.0.1
    subnet: 255.255.255.0
sprinkler:
  - id: sprinkler_ctrl
    main_switch: "Sprinklers"
    auto_advance_switch: "Sprinklers Auto Advance"
    multiplier_number: "Sprinkler Multiplier"
    repeat_number: "Sprinkler Repeat"
    valve_overlap: 3s
    valves:
      - valve_switch: "Zone 1"
        run_duration: 300s
        valve_switch_id: sprinkler_valve_sw1
      - valve_switch: "Zone 2"
        run_duration: 300s
        valve_switch_id: sprinkler_valve_sw2
[... 195 lines cut ...]
  439 |       vector<_Tp, _Alloc>::
      |       ^~~~~~~~~~~~~~~~~~~
0x184285e internal_error(char const*, ...)
	???:0
0xa8c7f3 gimple_seq_discard(gimple*)
	???:0
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
*** [.pioenvs/sprinklercontroller/src/esphome/components/api/api_pb2.cpp.o] Error 1
during GIMPLE pass: phiopt
src/esphome/components/api/api_server.cpp: In member function 'bool esphome::api::APIServer::check_password(const std::string&) const':
src/esphome/components/api/api_server.cpp:345:1: internal compiler error: Segmentation fault
  345 | }  // namespace esphome
      | ^
0x184285e internal_error(char const*, ...)
	???:0
0xa8c7f3 gimple_seq_discard(gimple*)
	???:0
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
*** [.pioenvs/sprinklercontroller/src/esphome/components/api/api_server.cpp.o] Error 1
========================= [FAILED] Took 21.75 seconds =========================
```

ssieb, 2024-03-08:
That is most likely a hardware problem.  What are you running HA on?

on-kage, 2024-03-08:
I am running on a BeeLink mini PC (https://a.co/d/itxagkK)
HA is running on a VM in VirtualBox (6.1.38_Ubuntu_r153438). Host OS is Linux Mint 21.1

ssieb, 2024-03-08:
I'm guessing it's a virtualbox issue then.  Try running esphome directly on the PC, not as an HA addon.

on-kage, 2024-03-08:
Running ESPHome in a docker directly on the PC did work to get compiled and loaded

closed 2024-03-08 by on-kage

blackie333, 2024-03-21:
Have experienced similar errors on my 4GB Arm-based host because esphome compilation has consumed all available ram memory on host.