esphome/issues #5580: Can't detect i2c device

AussieMakerGeek, 2024-03-09:
### The problem

I have tried multiple different json settings but I can't seem to get my AS3935 lightning sensor to be detected by ESPHome.

On Tasmota firmware it worked fine, Loading a generic 'i2c scan' arduino program onto the device (such as [this](https://learn.adafruit.com/scanning-i2c-addresses/arduino)), reliably finds the sensor on 0x03.
[image]
[image]

But ESPHome refuses to detect anything:
[image]

### Which version of ESPHome has the issue?

2024.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.0

### What platform are you using?

ESP8266

### Board

D1 Mini (clone)

### Component causing the issue

i2c

### Example YAML snippet

```yaml
esphome:
  name: lightning-sensor
  friendly_name: Lightning Sensor
esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "<redacted>"

ota:
  password: "<redacted>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lightning-Sensor"
    password: "<redacted>"

captive_portal:

i2c:
  sda: D2
  scl: D1
  #Also tried GPIO4/5
  scan: true
  frequency: 100kHz #Tried various speeds, and omitting..

as3935_i2c:
  irq_pin: GPIO13
  indoor: false

sensor:
  - platform: as3935
    lightning_energy:
      name: "Lightning Energy"
    distance:
      name: "Distance Storm"
binary_sensor:
  - platform: as3935
    name: "Storm Alert"
```

### Anything in the logs that might be useful for us?

```txt
Provided in screenshots
```

### Additional information

No external pullups, but I don't believe i should need them because it works with other firmware.

stefankleff, 2024-06-22:
Same issue here:
AS3935 (CJMCU Board) with Wemos D1 Mini via I2C 
does work with Tasmota but not with esphome

AussieMakerGeek, 2024-06-24:
I have not got back to this at all.  It has not been picked up by the devs. Perhaps in the newer versions of ESPHome it might work but I assume you're probably running a newer version.

illigtr, 2024-07-05:
I2C scan only starts at address 8, so it won't pickup address 0x03 of the as3935... but it will still communicate.