esphome/issues #5582: XPT2046 touch screen calibration is not working

smarthomecircle, 2024-03-09:
### The problem

I brought this board : https://s.click.aliexpress.com/e/_DnD2p7L

and I had configured the touch with calibration somewhere in september 2023.

Now since yesterday I tried updating to the latest version of esphome i.e 2024.2.2 and I see some options have changed. 
I updated the changes but since then my touch direction are always opposite or diagnol direction.

I am trying to calibrate the touch screen and I tried all the combination of transform option. (mirror_x, mirror_y, swap_x_y)
Finally I figured out the that mirror_x under transform does not change orientation of the touch. 

Also as per the documentation calibration has its own section now. https://esphome.io/components/touchscreen/#config-touchscreen
but when I use it it gives compilation error that calibration does not exists and the old calibration fields like `calibration_x_min` only works. 

I tried all possible combination of transform and calibration_x_min / calibration_y_min. 
but it does not give me correct touch value on the screen.

Finally I reached this yaml, where in the touch on the bottom left and top right is correct. 
But the touch on the top left and bottom right are wrong. 

```

captive_portal:

web_server:
  port: 80

##### Visual settings

font:      
  - file: 
      type: gfonts
      family: Mulish
      weight: medium
    id: mulish
    size: 30

########### CONNECTIONS
spi:
  - id: screen
    clk_pin: 14
    mosi_pin: 13
    miso_pin: 12
  - id: touch
    clk_pin: 25
    mosi_pin: 32
    miso_pin: 39   

touchscreen:
  id: my_touchscreen
  platform: xpt2046
  spi_id: touch
  cs_pin: 33
  interrupt_pin: 36
  update_interval: 50ms
  # report_interval: 1s
  threshold: 400
  calibration_x_min: 3860
  calibration_x_max: 280
  calibration_y_max: 340
  calibration_y_min: 3860
  transform:
    mirror_x: false
    mirror_y: false
    swap_xy: false
  on_touch:
    - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%d",
              touch.x,
              touch.y,
              touch.x_raw,
              touch.y_raw
              );

display:
  - platform: ili9xxx
    model: ILI9341
    
    spi_id: screen
    cs_pin: 15
    dc_pin: 2
    # reset_pin: en
    rotation: 90
 
    lambda: |-
      auto touch = id(my_touchscreen) -> get_touch();
      auto red = Color(255, 0, 0);

      auto orange = Color(255, 0, 0);
      auto green = Color(0, 255, 0);
      auto blue = Color(0, 0, 255);
      auto white = Color(255, 255, 255);
      it.fill(orange);
      it.print(10, 100, id(mulish), "Test Smart Home Circle");
      if (touch) {
        it.filled_circle(touch.value().x, touch.value().y, 30, white);
        ESP_LOGI("Test", "x=%d, y=%d",touch.value().x, touch.value().y);
        }
  

########## Display LED

# Define a PWM output on the ESP32
output:
  - platform: ledc
    pin: 21 #display LED
    id: gpio_21_backlight_pwm

############ Light Controls
# Define a monochromatic, dimmable light for the backlight
light:
  - platform: monochromatic
    output: gpio_21_backlight_pwm
    name: "Display Brightness"
    id: back_light

    restore_mode: ALWAYS_ON
```

This touch was working fine and well calibrated somewhere in sept 2023. But then I dint use the display today after updating the display, with the new changes, I started facing problems. 

Can someone help me out here?
I am sure there is some bug in the updated touch screen XPT2046 code, as this was all working fine in sept 2023.

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
captive_portal:

web_server:
  port: 80

##### Visual settings

font:      
  - file: 
      type: gfonts
      family: Mulish
      weight: medium
    id: mulish
    size: 30

########### CONNECTIONS
spi:
  - id: screen
    clk_pin: 14
    mosi_pin: 13
    miso_pin: 12
  - id: touch
    clk_pin: 25
    mosi_pin: 32
    miso_pin: 39   

touchscreen:
  id: my_touchscreen
  platform: xpt2046
  spi_id: touch
  cs_pin: 33
  interrupt_pin: 36
  update_interval: 50ms
  # report_interval: 1s
  threshold: 400
  calibration_x_min: 3860
  calibration_x_max: 280
  calibration_y_max: 340
  calibration_y_min: 3860
  transform:
    mirror_x: false
    mirror_y: false
    swap_xy: false
  on_touch:
    - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%d",
              touch.x,
              touch.y,
              touch.x_raw,
              touch.y_raw
              );

display:
  - platform: ili9xxx
    model: ILI9341
    
    spi_id: screen
    cs_pin: 15
    dc_pin: 2
    # reset_pin: en
    rotation: 90
 
    lambda: |-
      auto touch = id(my_touchscreen) -> get_touch();
      auto red = Color(255, 0, 0);

      auto orange = Color(255, 0, 0);
      auto green = Color(0, 255, 0);
      auto blue = Color(0, 0, 255);
      auto white = Color(255, 255, 255);
      it.fill(orange);
      it.print(10, 100, id(mulish), "Test Smart Home Circle");
      if (touch) {
        it.filled_circle(touch.value().x, touch.value().y, 30, white);
        ESP_LOGI("Test", "x=%d, y=%d",touch.value().x, touch.value().y);
        }
  

########## Display LED

# Define a PWM output on the ESP32
output:
  - platform: ledc
    pin: 21 #display LED
    id: gpio_21_backlight_pwm

############ Light Controls
# Define a monochromatic, dimmable light for the backlight
light:
  - platform: monochromatic
    output: gpio_21_backlight_pwm
    name: "Display Brightness"
    id: back_light

    restore_mode: ALWAYS_ON

```
```

### Anything in the logs that might be useful for us?

```txt
here is log for the touch calibration. 

top left
[I][cal:085]: x=29, y=17, x_raw=3527, y_raw=3602

[W][component:214]: Component display took a long time for an operation (0.09 s).
[W][component:215]: Components should block for at most 20-30ms.

[W][component:214]: Component display took a long time for an operation (0.09 s).
[W][component:215]: Components should block for at most 20-30ms.

bottom left
[I][cal:085]: x=288, y=24, x_raw=630, y_raw=3498

[W][component:214]: Component display took a long time for an operation (0.09 s).
[W][component:215]: Components should block for at most 20-30ms.

[W][component:214]: Component display took a long time for an operation (0.09 s).
[W][component:215]: Components should block for at most 20-30ms.

botom right
[I][cal:085]: x=283, y=202, x_raw=691, y_raw=888

[W][component:214]: Component display took a long time for an operation (0.09 s).
[W][component:215]: Components should block for at most 20-30ms.

[W][component:214]: Component display took a long time for an operation (0.09 s).
[W][component:215]: Components should block for at most 20-30ms.
top right
[I][cal:085]: x=33, y=227, x_raw=3487, y_raw=519

```
```

berndkrannich, 2024-07-08:
Related to #5696. I just retested with esphome 2024.6.6 and the issue persists. The workaround I mentioned in the other issue continues to apply. Unfortunately, I don't have enough confidence in my understanding of the codebase to suggest a proper fix via pull request myself.

smarthomecircle, 2024-07-20:
So I got it working after changing my concept of the x_min and x_max. 

I set the x_min as the minimum value i got from x_raw and same for y_min as the minimum value I got for y_raw by touching all the four corners. similarity I set x_max and y_max with maximum values I got x_raw and y_raw. 

Now to align the values of x_raw values with x values after touching two corners, we have to use the `mirror_x ` 
similarly to align y_raw with the y values, we have to mirror_y value. 

e.g for this log
```

top left
[I][cal:085]: x=29, y=17, x_raw=3527, y_raw=3602

[W][component:214]: Component display took a long time for an operation (0.09 s).
[W][component:215]: Components should block for at most 20-30ms.

[W][component:214]: Component display took a long time for an operation (0.09 s).
[W][component:215]: Components should block for at most 20-30ms.

bottom left
[I][cal:085]: x=288, y=24, x_raw=630, y_raw=3498
```
 x_raw is high when x value is low for top left corner , while x_raw is low while x is higher for the bottom left touch.

to correct this, use the mirror_x value. If it was true, then set it to false. 
Do the same to align y_raw values with y values. 

This is the config that finally worked for me. 

```

web_server:
  port: 80

##### Visual settings

font:      
  - file: 
      type: gfonts
      family: Mulish
      weight: medium
    id: mulish
    size: 30

########### CONNECTIONS
spi:
  - id: screen
    clk_pin: 14
    mosi_pin: 13
    miso_pin: 12
  - id: touch
    clk_pin: 25
    mosi_pin: 32
    miso_pin: 39   

touchscreen:
  id: my_touchscreen
  platform: xpt2046
  spi_id: touch
  cs_pin: 33
  interrupt_pin: 36
  update_interval: 50ms
  # report_interval: 1s
  threshold: 400
  calibration:
    x_max: 3860
    x_min: 280
    y_min: 340
    y_max: 3860
  transform:
    mirror_x: false
    mirror_y: false
    swap_xy: false
  on_touch:
    - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%d",
              touch.x,
              touch.y,
              touch.x_raw,
              touch.y_raw
              );

display:
  - platform: ili9xxx
    model: ILI9341
    
    spi_id: screen
    cs_pin: 15
    dc_pin: 2
    # reset_pin: en
    rotation: 90
 
    lambda: |-
      auto touch = id(my_touchscreen) -> get_touch();
      auto red = Color(255, 0, 0);
            //auto orange = Color(245, 102, 249);

      auto orange = Color(255, 100, 0);
      auto green = Color(0, 255, 0);
      auto blue = Color(0, 0, 255);
      auto white = Color(255, 255, 255);
      it.fill(orange);
      it.print(10, 100, id(mulish), "Smart Home Circle");
      if (touch) {
        auto x = touch.value().x;
        auto y = touch.value().y;

        if(x!=0 && y!=0){
          it.filled_circle(y, x, 30, white);
        }
        ESP_LOGI("Test", "x=%d, y=%d",touch.value().x, touch.value().y);
        }
  

########## Display LED

# Define a PWM output on the ESP32
output:
  - platform: ledc
    pin: 21 #display LED
    id: gpio_21_backlight_pwm

############ Light Controls
# Define a monochromatic, dimmable light for the backlight
light:
  - platform: monochromatic
    output: gpio_21_backlight_pwm
    name: "Display Brightness"
    id: back_light

    restore_mode: ALWAYS_ON

```

I might make a video about it on my channel : https://www.youtube.com/@SmartHomeCircle

rofo69, 2024-10-17:
I have the same issue. Using a cheap yellow display using this config:-

```
touchscreen:
  platform: xpt2046
  id: esp_touchscreen
  spi_id: spi_touch
  cs_pin: GPIO33
  interrupt_pin: GPIO36
  update_interval: 50ms
  threshold: 400
  calibration:
    x_min: 3860
    x_max: 280 #360
    y_min: 340 #280
    y_max: 3860

  transform:
   swap_xy: true
   mirror_x: true
   mirror_y: false
```
I am trying to use the device, rotated 90 degrees. Changing the display transform settings to this:-

    transform:
      swap_xy: true

Works fine.

However, doing the same on the config settings makes the Y axis incorrect, with Y being high at the top of the (rotated) screen.

Any value of mirror_x  to change this, does not affect the Y axis orientation at all (mirror_x affects Y because I have swap_xy = true). So there is nothing I can do to make Y low at the top, and high at the bottom.

clydebarrow, 2024-10-30:
This is likely the problem:

```
    x_min: 3860
    x_max: 280 #360
```

The *only* effect of reversing the coordinates is to implicitly set `mirror_x` (overriding the settings in `transform:`)so you should just use the natural coordinates and set the transform options as required.

closed 2024-10-31 by jesserockz

RyszardM12, 2025-02-10:
> So I got it working after changing my concept of the x_min and x_max.
>
> I set the x_min as the minimum value i got from x_raw and same for y_min as the minimum value I got for y_raw by touching all [...]