esphome/issues #5583: Hourly reboots on minimal configuration

l3zha, 2024-03-09:
### The problem

Uptime reaches almost an hour and a reboot occurs, sometimes it reboots earlier, but there was no uptime for more than an hour. I am newbie. I tried to reflash from scratch several times, but the result is the same

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.0

### What platform are you using?

ESP32

### Board

petkit solo feeder

### Example YAML snippet

```yaml
esphome:
  name: dogfeeder
  friendly_name: dogfeeder

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG
  baud_rate: 9600
# Enable Home Assistant API
api:
  encryption:
    key: !secret encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Dogfeeder Fallback Hotspot"
    password: "xrZYexCLi0RF"

captive_portal:
    
sensor:
  - platform: uptime
    name: Uptime Sensor

button:
  - platform: restart
    name: "Living Room Restart"

debug:
  update_interval: 5s

[... 262 lines cut ...]
[0;33m[W][component:214]: Component api took a long time for an operation (0.25 s).[0m
[0;33m[W][component:215]: Components should block for at most 20-30ms.[0m
[0;38m[VV][api.service:174]: send_text_sensor_state_response: TextSensorStateResponse {
  key: 2962208445
  state: 'Power On Reset'
  missing_state: NO
}[0m
[0;33m[W][component:214]: Component api took a long time for an operation (0.13 s).[0m
[0;33m[W][component:215]: Components should block for at most 20-30ms.[0m
[0;38m[VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=19804 (now=24804)[0m
[0;38m[VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=24804 (now=29806)[0m
[0;38m[VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=29804 (now=34804)[0m
[0;38m[VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=34804 (now=39804)[0m
[0;38m[VV][scheduler:226]: Running interval 'update' with interval=5000 last_execution=39804 (now=44804)[0m
[0;38m[VV][scheduler:226]: Running interval 'update' with interval=60000 last_execution=4294954404 (now=47108)[0m
[0;37m[V][sensor:043]: 'Uptime Sensor': Received new state 47.108002[0m
[0;36m[D][sensor:094]: 'Uptime Sensor': Sending state 47.10800 s with 0 decimals of accuracy[0m
[0;38m[VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 683514054
  state: 47.108
  missing_state: NO
}[0m
[0;33m[W][component:214]: Component uptime.sensor took a long time for an operation (0.37 s).[0m
[0;33m[W][component:215]: Components should block for at most 20-30ms.[0m
```

ssieb, 2024-03-09:
Those logs do not match your config.  Change the log level back to debug and why did you change the logger baud rate?  That's going to slow things down.  If you leave the logs running, see if you can see the reason for it restarting.

l3zha, 2024-03-10:
> Those logs do not match your config. Change the log level back to debug and why did you change the logger baud rate? That's going to slow things down. If you leave the logs running, see if you can see the reason for it restarting.

For some reason the log at a speed of 115200 looks like this:

€Тb R™€≈DшёДEIОdHR€т ¬ QMЎМАс(PN$А€ыхы-x±”•ЂЋЋ©DЄЛ~sLµнЈj/МьoSµќ•љOмуB÷Є/џ÷5…PkЁ/[≈UЏТакЁЌjuƒ÷cƒ€zЯ4kOрOнEEЏЁMЉ{QkЯЬЃXH≤ѓю%5лQі€дцЛmgХ•?©Ѓ€ЛђшBjяўЌ†?0ѓЯ®П$е€DьЪqbы&ѓѕґ[µ#—я‘Ё„С≥чЊІw@vп1“€_Е…нuИпµ]ы€>xov©ЯдµOµ€Њ™Я$№&ЏRMz{—¬Z;F√ЯВКЧKЎk"№ЂU÷†≥<Ўl{€єaJтђ¶пх>=mъ#Х€цЧU vэЭ|6J•€vм+XZэ

Today I installed Tasmota, the device still reboots, the reason for the reboot is "Vbat power on reset"

ssieb, 2024-03-10:
You need to tell `screen` the baud rate.
The serial logs should give a better reason.  I have no idea what that means.