esphome/issues #5586: D1 Mini board resets with Reset Info: Fatal exception:29

itrankolov, 2024-03-10:
### The problem

Device resets randomly with  Fatal exception:29 flag:2 (Exception) epc1:0x4023f40f epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000019 depc:0x00000000. 

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Wemos D2 Mini

### Example YAML snippet

```yaml
esphome:
  name: boiler-temps
  friendly_name: Boiler Temps

esp8266:
  board: d1_mini

# Enable logging
logger:
# DEBUG
  level: debug
# END DEBUG
# Enable Home Assistant API
api:
  encryption:
    key: "r8aFoaUlwKTYztVf264qbFY69Z1NWnIo1EkekIspi7U="

ota:
  password: "c2851b17fd1999a87d2b6c93d65d8cea"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Boiler-Temps Fallback Hotspot"
#    password: "5eZnRweAjXEI"
# DEBUG
debug:
  update_interval: 5s

text_sensor:
  - platform: debug
    device:
      name: "Device Info"
    reset_reason:
      name: "Reset Reason"
# ENG DEBUG
#captive_portal:

# DS18B20 Temperature Sensor
dallas:
  - pin: D5
    update_interval: 2s
sensor:
  - platform: dallas
    address: 0x520d7ad4461cf828
    name: "Heating Output"
    accuracy_decimals: 1
    resolution: 12
    filters:
    - offset: -0.04
  - platform: dallas
    address: 0x5f36ead445a4bc28
    name: "Heating Return"
    accuracy_decimals: 1
    resolution: 12
    filters:
    - offset: -0.6025
  - platform: dallas
    address: 0x7649f7d445e76528
    name: "DHW Out"
    accuracy_decimals: 1
    resolution: 12
    filters:
    - offset: -0.7275
  - platform: dallas
    address: 0x402600d445861d28
    name: "DHW IN"
    accuracy_decimals: 1
    resolution: 12
    filters:
    - offset: -0.915
#Uptime Sensor
  - platform: uptime
    name: ESP Boiler Uptime
    update_interval: 5s
  - platform: adc
    pin: VCC
    name: "VCC Voltage"
    update_interval: 1s
# DEBUG 
  - platform: debug
    free:
      name: "Heap Free"
    fragmentation:
      name: "Heap Fragmentation"
    block:
      name: "Heap Max Block"
    loop_time:
      name: "Loop Time"
```

### Anything in the logs that might be useful for us?

```txt
[02:19:45][D][debug:076]: ESPHome version 2024.2.2
[02:19:45][D][debug:080]: Free Heap Size: 27600 bytes
[02:19:45][D][debug:108]: Flash Chip: Size=4096kB Speed=40MHz Mode=DOUT
[02:19:45][D][debug:323]: Chip ID: 0x00D51620
[02:19:45][D][debug:324]: SDK Version: 2.2.2-dev(38a443e)
[02:19:45][D][debug:325]: Core Version: 3.0.2
[02:19:45][D][debug:326]: Boot Version=31 Mode=1
[02:19:45][D][debug:327]: CPU Frequency: 80
[02:19:45][D][debug:328]: Flash Chip ID=0x0016400B
[02:19:45][D][debug:329]: Reset Reason: Exception
[02:19:45][D][debug:330]: Reset Info: Fatal exception:29 flag:2 (Exception) epc1:0x4023f40f epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000019 depc:0x00000000
[02:19:45][D][text_sensor:064]: 'Device Info': Sending state '2024.2.2|Flash: 4096kB Speed:40MHz Mode:DOUT|Chip: 0x00d51620|SDK: 2.2.2-dev(38a443e)|Core: 3.0.2|Boot: 31|Mode: 1|CPU: 80|Flash: 0x0016400b|Reset: Exception|Fatal exception:29 flag:2 (Exception) epc1:0x4023f40f epc2:0x00000000 epc3:0x00000000 excvaddr:0x'
[02:19:45][D][text_sensor:064]: 'Reset Reason': Sending state 'Exception'
```

ssieb, 2024-03-10:
Can you get serial logs of the exception?  How often does it happen?

itrankolov, 2024-03-10:
Serial no for the moment. I can try but it will take some time. It happens randomly at most every 4-5 hours, sometimes 30 minutes.

itrankolov, 2024-03-10:
I've managed to gather some serial logs and I find it very confusing. At some random point in time it disconnects from the WiFi:
```
[W][wifi_esp8266:496]: Event: Disconnected ssid='Shishacite' bssid=48:8F:5A:C9:93:00 reason='Group Key Update Timeout'
[W][wifi:153]: WiFi Connection lost... Reconnecting...
[W][wifi:663]: Restarting WiFi adapter...
[W][wifi_esp8266:496]: Event: Disconnected ssid='Shishacite' bssid=48:8F:5A:C9:93:00 reason='Authentication Failed'
[I][wifi:304]: WiFi Connecting to 'Shishacite'...
[W][wifi_esp8266:214]: wifi_apply_hostname_(boiler-temps): lwIP error -16 on interface st (index 0)
[W][component:214]: Component wifi took a long time for an operation (0.12 s).
[W][component:215]: Components should block for at most 20-30ms.
```
After that disconnect it cannot reconnect until it reboots itself. Sometimes it reboots with:
```
[E][api:128]: No client connected to API. Rebooting...
[I][app:127]: Forcing a reboot...
[W][wifi_esp8266:496]: Event: Disconnected ssid='Shishacite' bssid=48:8F:5A:C9:93:00 reason='Association Leave'
```
Sometimes with:
```
User exception (panic/abort/assert)
                                   --------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

ctx: cont
sp: 3ffffe70 end: 3fffffd0 offset: 0000
3ffffe70:  3fff0bfc 00c41f23 3fff0afc 40219220  
3ffffe80:  000000fe 00000000 00000000 00000000  
3ffffe90:  00000000 00000000 00000000 401010f8  
3ffffea0:  3fffefd0 000000fe 3fff0250 00000001  
3ffffeb0:  00000000 00000000 3fffbdf4 00000001  
3ffffec0:  3fff2bf8 4021e91f 0000010c 4022551a  
3ffffed0:  3fff2bf8 4021e91f 00000020 40225539  
3ffffee0:  00000002 00410cb4 0000010c 40224d30  
3ffffef0:  00000002 3ffef780 3ffef7ac 4021e91f  
3fffff00:  3fff0000 00000000 00000000 00000000  
3fffff10:  3fff0000 00000000 3ffef7ac 4021f7c0  
3fffff20:  00000002 3ffef780 3ffef7ac 4021fb68  
3fffff30:  00000002 00410cb4 b7ced916 4b36b0f4  
3fffff40:  00000002 00410cb4 ba1cac08 4b36b0f7  
3fffff50:  3fff2bf4 00000000 3fffff80 3fff0d04  
3fffff60:  3fff2bf8 3ffef780 3ffef5c4 4021fddb  
3fffff70:  3fff2bf8 00000102 3ffef5c4 4021a43a  
3fffff80:  00c41f23 3fff0d04 00000000 3ffef998  
3fffff90:  3fff2c08 3fff13f0 3ffef5f8 3ffef998  
3fffffa0:  3fffdad0 00000000 3ffef984 3ffef998  
3fffffb0:  3fffdad0 00000000 3ffef984 40225048  
3fffffc0:  feefeffe feefeffe 3ffe8614 401007e1  
<<<stack<<<

last failed alloc call: 4021E91F(268)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```
But he main point is that it does not reconnect to WiFi until the reboot. Reasons for not reconnecting are random:
```
[W][wifi_esp8266:496]: Event: Disconnected ssid='Shishacite' bssid=48:8F:5A:C9:93:00 reason='Auth Expired'
[W][wifi_esp8266:496]: Event: Disconnected ssid='Shishacite' bssid=48:8F:5A:C9:93:00 reason='Association Failed'
[W][wifi_esp8266:496]: Event: Disconnected ssid='Shishacite' bssid=48:8F:5A:C9:93:00 reason='4-Way Handshake Timeout'
```
This is how it sees the WiFi:
```
[D][wifi:474]: Found networks:
[I][wifi:517]: - 'Shishacite' (48:8F:5A:C9:93:00) ▂▄▆█
[D][wifi:519]:     Channel: 11
[D][wifi:520]:     RSSI: -51 dB
```
So it does not seem like an issue with signal strenght.

ssieb, 2024-03-10:
`Group Key Update Timeout`  Check your wifi security settings.

itrankolov, 2024-03-10:
Wifi is working ok with all other D1 mini devices, including one after it reboots. It seems it is something related to its configuration. I will replace it with a new one and check if it still does the same.

itrankolov, 2024-03-12:
I've replaced the board with another one, but it seems there is someting wrong with the Dallas temperature sensor integration. When I remove that configuration the D1 does not restart.

ssieb, 2024-03-12:
Why is it restarting?  Is it still the out of memory exception?

itrankolov, 2024-03-12:
Yes. Now it is running for 34:41:28 without the configuration for dallas sensors.

itrankolov, 2024-03-12:
No, this is the full config!

ssieb, 2024-03-12:
Watch the "Heap Free" sensor value and see what it is before the exception.  And paste the exception you're seeing now.

itrankolov, 2024-03-12:
Last time before reboot "Heap Free" was 29936 - Fatal exception:29 flag:2 epc1:0x4000df64.

ssieb, 2024-03-12:
Yes.  Can you come to the esphome discord server?  It would be a lot easier to do this...

mogery, 2024-04-02:
Hey there, I am also getting this exception. Was it every resolved? My config:

```yaml
substitutions:
  name: esphome-web-d9611b
  friendly_name: ESPHome IRThing Geri

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp8266.yaml@main
  import_full_config: true

# To have a "next url" for improv serial
web_server:

remote_transmitter:
  pin: GPIO12
  carrier_duty_percent: 50%

output:
  - platform: template
    id: geri_ceiling_light_brightness
    type: float
    write_action: 
      then:
        - if:
            condition:
              light.is_on: geri_ceiling_light
            then:
              - remote_transmitter.transmit_samsung: 
                  data: !lambda |-
                    int brightness = (int)round(state * 7.0);
                    int temperature = (int)round((id(geri_ceiling_light).current_values.get_color_temperature() - 200.0) / 133.333344 * 7.0);
                    // ESP_LOGD("giron_ir", "Brightness: %i, Temperature: %i", brightness, temperature);
                    int base = 0b00111010110000000000000000000000;
                    int addr = ((0b0101 + brightness) << 4) + 0b1111 + temperature;
                    addr = ((addr & 0xF0) >> 4) | ((addr & 0x0F) << 4);
                    addr = ((addr & 0xCC) >> 2) | ((addr & 0x33) << 2);
                    addr = ((addr & 0xAA) >> 1) | ((addr & 0x55) << 1);
                    brightness = (brightness & 0b010) | ((brightness & 0b100) >> 2) | ((brightness & 0b001) << 2);
                    temperature = (temperature & 0b010) | ((temperature & 0b100) >> 2) | ((temperature & 0b001) << 2);
                    int cmd = base | (brightness << 17) | (temperature << 13) | addr;
                    // ESP_LOGD("giron_ir", "Command: %#010X", cmd);
                    return cmd;
            else:
              - remote_transmitter.transmit_samsung: 
                  data: 0x8ACE0023
        
  - platform: template
    id: geri_ceiling_light_temperature
    type: float
    write_action:
      then: # This is just a dummy output. It's handled by geri_ceiling_light_brightness     

light:
  - platform: color_temperature
    name: Ceiling Light
    id: geri_ceiling_light
    brightness: geri_ceiling_light_brightness
    color_temperature: geri_ceiling_light_temperature
    default_transition_length: 0s
    warm_white_color_temperature: 3000 K
    cold_white_color_temperature: 5000 K

sensor:
  - platform: homeassistant
    id: geri_temperature_sensor
    entity_id: sensor.temperature_humidity_sensor_9ca4_temperature

climate:
  - platform: coolix
    name: Geri AC
    id: geri_ac
    sensor: geri_temperature_sensor

debug:
  update_interval: 5s
```