esphome/issues #5587: ESP32-C3 won't update

danolson1, 2024-03-10:
### The problem

I am trying to upload a yaml file to my ESP32-C3 device and I get the following error message: "warning: "ARDUINO_USB_CDC_ON_BOOT" redefined"

Previously this worked fine. I'm not sure what changed, or where to find more information about this error. If anyone could point me in the right direction, I would appreciate it.

-----

Here is my yaml file (with minor edits to remove passwords, etc.):
```
# Notes: need to attach wifi antenna for good signal reception

substitutions:
  name: hp-5
  friendly_name: hp-5

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.flash_mode: dio
    #board_build.mcu: esp32c3 # not needed, not sure what this does
    board_build.extra_flags:
      # Override, defaults to '-DARDUINO_USB_CDC_ON_BOOT=1'     
      # This is necessary to allow mitsubishi heatpump to work correctly
      # More info here https://wiki.seeedstudio.com/XIAO_ESP32C3_Pin_Multiplexing/
      - "-DARDUINO_USB_CDC_ON_BOOT=0"  

esp32:
  board: seeed_xiao_esp32c3
  variant: esp32c3
  framework:
    type: arduino
    version: 2.0.14
    platform_version: 6.5.0 # 6.4.0 and 5.4.0 worked

# Enable logging
logger:
 #hardware_uart: UART0
 baud_rate: 0 # disable the logger

# Enable Home Assistant API
api:
  encryption:
    key: "###REMOVED###"

ota:
  password: "###REMOVED###"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hp-5 Fallback Hotspot"
    password: "###REMOVED###"

web_server:
  port: 80

  # Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ${name} ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: ${name} IP
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${name} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${name} WiFi Signal
    update_interval: 60s

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"

    # ESP32 only - change UART0 to UART1 or UART2 and remove the
    # logging:baud_rate above to allow the built-in UART0 to function for
    # logging.
    
    # ESP32C3, must be UART0
    hardware_uart: UART0

```
------
Here is the full output of my install script:
```
INFO ESPHome 2024.2.2
INFO Reading configuration /config/esphome/hp-5.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'America/New_York'
INFO Generating C++ source...
INFO Compiling app...
Processing hp-5 (board: seeed_xiao_esp32c3; framework: arduino; platform: platformio/espressif32@6.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- HeatPump @ 1.0.0+sha.5d1e146
Compiling .pioenvs/hp-5/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o
<command-line>: warning: "ARDUINO_USB_CDC_ON_BOOT" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pioenvs/hp-5/src/esphome/components/sensor/sensor.cpp.o
Compiling .pioenvs/hp-5/src/esphome/components/socket/bsd_sockets_impl.cpp.o
<command-line>: warning: "ARDUINO_USB_CDC_ON_BOOT" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pioenvs/hp-5/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
<command-line>: warning: "ARDUINO_USB_CDC_ON_BOOT" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: warning: "ARDUINO_USB_CDC_ON_BOOT" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pioenvs/hp-5/src/esphome/components/socket/lwip_sockets_impl.cpp.o
<command-line>: warning: "ARDUINO_USB_CDC_ON_BOOT" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pioenvs/hp-5/src/esphome/components/socket/socket.cpp.o
<command-line>: warning: "ARDUINO_USB_CDC_ON_BOOT" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pioenvs/hp-5/src/esphome/components/text_sensor/filter.cpp.o
<command-line>: warning: "ARDUINO_USB_CDC_ON_BOOT" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pioenvs/hp-5/src/esphome/components/text_sensor/text_sensor.cpp.o
<command-line>: warning: "ARDUINO_USB_CDC_ON_BOOT" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pioenvs/hp-5/src/esphome/components/time/automation.cpp.o
<command-line>: warning: "ARDUINO_USB_CDC_ON_BOOT" redefined
<command-line>: note: this is the location of the previous definition
In file included from src/esphome/components/api/proto.h:4,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'virtual void MitsubishiHeatPump::setup()':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:454:46: error: comparison between distinct pointer types 'HardwareSerial*' and 'HWCDC*' lacks a cast
             YESNO(this->get_hw_serial_() == &Serial)
                                              ^~~~~~
src/esphome/core/log.h:97:90: note: in definition of macro 'esph_log_config'
   esp_log_printf_(ESPHOME_LOG_LEVEL_CONFIG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                          ^~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:449:5: note: in expansion of macro 'ESP_LOGCONFIG'
     ESP_LOGCONFIG(
     ^~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:454:13: note: in expansion of macro 'YESNO'
             YESNO(this->get_hw_serial_() == &Serial)
             ^~~~~
*** [.pioenvs/hp-5/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o] Error 1
========================== [FAILED] Took 4.92 seconds ==========================
```

### Which version of ESPHome has the issue?

2024.2.2. I think the last version that worked was 2013.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.1.6

### What platform are you using?

ESP32

### Board

seeed xiao esp32c3

### Component causing the issue

esphome

ssieb, 2024-03-10:
You shouldn't be adding that define.  And why does the component require you to use UART0 for a C3?

ssieb, 2024-03-10:
Anyway, this is an issue with an external component, so it's not an esphome issue.  You can come ask on discord, but it's really a question for the provider of that component.

closed 2024-03-10 by ssieb

danolson1, 2024-03-10:
Ok, except that things were working fine with Esphome version 2023.12.5, and seem to have broken when version 2023.12.6 happened. The 2013.12.6 update also included a "ESP32-C3 USB_CDC fix" https://github.com/esphome/esphome/pull/6069

Somehow I think that fix broke things for me.

Also, even though the logs say the external component is generating the error message, I can't seem to find where it's coming from. What file is generating the error text?
```
<command-line>: warning: "ARDUINO_USB_CDC_ON_BOOT" redefined
<command-line>: note: this is the location of the previous definition
```

ssieb, 2024-03-10:
The error is happening because esphome already sets that define as necessary (as part of that change) and you're trying to set it again.  That particular issue is from you, not the component.  But you're apparently doing that because of the component.

danolson1, 2024-03-10:
Ok, so if esphome already sets that define, what is the correct way to override it?

Alternatively, is there a way I can set "ARDUINO_USB_MODE=0"?, since I think that the code change where this was set to 1 is what broke things for me.

ssieb, 2024-03-10:
You shouldn't need to set it.  If you do, it's a problem with the component which should be using the esphome framework.  But I see you're also modifying the platform versions as well, so you're really out on your own as the warnings already tell you.