esphome/issues #5589: [gradient] is an invalid option for [addressable_color_wipe]. Please check the indentation

532910, 2024-03-12:
### The problem

https://github.com/esphome/esphome-docs/blob/cde95b10cfff91aa6aab4d75c3c67fd391e792c9/components/light/index.rst#addressable-color-wipe-effect

1. In the example `gradient` is a part of `addressable_color_wipe` and is an invalid option for `addressable_color_wipe`
2. In the documentation `gradient` is a part of `color` that also is an invalid option for `color`

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

farajfarook, 2024-03-20:
Same issue
`[gradient] is an invalid option for [addressable_color_wipe]. Please check the indentation.`
```
........
            blue: 0%
            num_leds: 1
        add_led_interval: 100ms
        reverse: False
        
        [gradient] is an invalid option for [addressable_color_wipe]. Please check the indentation.
        gradient: False
```

ssieb, 2024-03-20:
Yes, the example is wrong.  The `gradient` option is in the color list.
```yaml
          colors:
            - red: 100%
              green: 100%
              blue: 100%
              num_leds: 1
              gradient: False
```

532910, 2024-07-19:
stale is not a reason to close issues