esphome/issues #5590: SONOFF S31 device logging constantly sending.

ClockwiseCounterclockwise, 2024-03-12:
### The problem

Hello, I'm not sure when this started but I see entries in the esphome log for the SONOFF S31 device where it keeps sending logs.  Before this, the sending of data would send all values, then there would be a pause before it would send again.  

Now I'm seeing just a scrolling wall of "sending state text" entries non-stop.  

n.b. I haven't changed any of the code.

e.g.
[01:39:16][D][sensor:093]: 'available2 Voltage': Sending state 121.14915 V with 1 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Power': Sending state 0.00000 W with 1 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Energy': Sending state 0.00001 kWh with 3 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Current': Sending state 0.00000 A with 1 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Voltage': Sending state 121.14915 V with 1 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Power': Sending state 0.00000 W with 1 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Energy': Sending state 0.00001 kWh with 3 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Current': Sending state 0.00000 A with 1 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Voltage': Sending state 121.29743 V with 1 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Power': Sending state 0.00000 W with 1 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Energy': Sending state 0.00001 kWh with 3 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Current': Sending state 0.00000 A with 1 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Voltage': Sending state 121.29743 V with 1 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Power': Sending state 0.00000 W with 1 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Energy': Sending state 0.00001 kWh with 3 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Current': Sending state 0.00000 A with 1 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Voltage': Sending state 121.29743 V with 1 decimals of accuracy
[01:39:16][D][sensor:093]: 'available2 Power': Sending state 0.00000 W with 1 decimals of accuracy

Thank you for your assistance.

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.0

### What platform are you using?

ESP8266

### Board

SONOFF S31 Plug

### Component causing the issue

n/a

### Example YAML snippet

```yaml
#Sonos
esphome:
  name: "available2"
  platform: ESP8266
  board: esp01_1m

# Enable logging
logger:
  baud_rate: 0 # (UART logging interferes with cse7766)

# Enable Home Assistant API
api:
  encryption:
    key: ----
ota:
  password: ----

web_server:
  port: 80

wifi:
  ssid: ----
  password: ----
  manual_ip:
    static_ip: ----
    gateway: ----
    subnet: ----
    dns1: ----
    dns2: ----

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ----
    password: ----

captive_portal:

uart:
  rx_pin: RX
  baud_rate: 4800

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "available2"
    on_press:
      - switch.toggle: relay
  - platform: status
    name: "available2"

sensor:
  - platform: wifi_signal
    name: "available2 wifi signal"
    update_interval: 60s
  - platform: cse7766
    current:
      name: "available2 Current"
      accuracy_decimals: 1
    voltage:
      name: "available2 Voltage"
      accuracy_decimals: 1
    power:
      name: "available2 Power"
      accuracy_decimals: 1
      id: power
  - platform: integration
    name: "available2 Energy"
    sensor: power
    time_unit: h
    unit_of_measurement: kWh
    filters:
      - multiply: 0.001

switch:
  - platform: gpio
    name: "available2 Relay"
    pin: GPIO12
    id: relay
    restore_mode: ALWAYS_ON

time:
  - platform: sntp
    id: the_time

status_led:
  pin:
    number: GPIO13
    inverted: True
```

ssieb, 2024-03-12:
You need to read the [changelog](https://esphome.io/changelog/) before updating.

closed 2024-03-12 by ssieb

ClockwiseCounterclockwise, 2024-03-13:
> You need to read the [changelog](https://esphome.io/changelog/) before updating.

My apologizes, Thank you ssieb.