esphome/issues #5593: ESP32 crashes with Grow R503 fingerprint sensor

mondalaci, 2024-03-12:
### The problem

I’m trying to make the Grow R503 work with my ESP32 WROOM-32, but it crashes.

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.0

### What platform are you using?

ESP32

### Board

ESP32 WROOM-32

### Component causing the issue

fingerprint_grow

### Example YAML snippet

```yaml
esphome:
  name: bearcave-entry

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

# Enable logging
logger:

ota:
  password: "xxx"

wifi:
  ssid: "mynet"
  password: "xxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bearcave-Entry"
    password: "xxx"

mdns:
  disabled: false

uart:
  rx_pin: GPIO21
  tx_pin: GPIO22
  baud_rate: 57600

fingerprint_grow:
  sensing_pin: GPIO23
  on_finger_scan_invalid:
    - homeassistant.event:
        event: esphome.test_node_finger_scan_invalid
  on_finger_scan_matched:
    - homeassistant.event:
        event: esphome.test_node_finger_scan_matched
[... 172 lines cut ...]
[18:04:01]
[18:04:01]
[18:04:01]ELF file SHA256: 0000000000000000
[18:04:01]
[18:04:01]Rebooting...
[18:04:01]ets Jul 29 2019 12:21:46
[18:04:01]
[18:04:01]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:04:01]configsip: 0, SPIWP:0xee
[18:04:01]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:04:01]mode:DIO, clock div:2
[18:04:01]load:0x3fff0030,len:1184
[18:04:01]load:0x40078000,len:13132
[18:04:01]load:0x40080400,len:3036
[18:04:01]entry 0x400805e4
[18:04:01][I][logger:359]: Log initialized
...
```
```

### Additional information

Both the power supply and touch induction power supply of the R503 are connected to 3.3V

As you can see, fingerprint detection works, but the board crashes until it settles in safe mode.

dekamilv, 2024-03-16:
Have same here. Tried on nodemcu v3 and esp32-wroom. Both boards in permanent rebooting.

dekamilv, 2024-03-17:
Found solution. Still did not manage to switch on fingerprint module but board stop crashing  #[https://github.com/esphome/issues/issues/5502](url)

alexborro, 2024-03-17:
Hi @mondalaci your issue is probably fixed in the next release. Could you please try the beta version `2024.3.0b2` ? This beta version already integrates a fix for this problem.

@dekamilv try this beta version and if it doesn't work, please open a new issue, post your yaml file and some logs (preferably from serial not WiFi, so I can see the boot process). Do not forget to **tag me** in the new issue, otherwise I don't get notified.

dekamilv, 2024-03-17:
Curiouser and curiouser.... Esphome beta does not start on Hassio. As per log it is started but hassio even don't know about such fact. Better I will leave fingerprint sensor at a side for the time being.

dekamilv, 2024-03-17:
> Hi @mondalaci your issue is probably fixed in the next release. Could you please try the beta version `2024.3.0b2` ? This beta version already integrates a fix for this problem.
>
> @dekamilv try this [...]

Hi Alexborro. After millions attempts today I finally managed to follow your suggest and its works! Thank you very much!!!

Installed ESP beta and re-flash ESP32.

Configuration:

> 
esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  baud_rate: 9600 #
  

wifi:
  ssid: "xxx"
  password: "xxx"
  power_save_mode: none

api:
  encryption:
    key: "xxx"
            
ota:
  password: "xxx"

uart:
  id: fingerprint
  rx_pin: GPIO9
  tx_pin: GPIO10
  baud_rate: 57600

fingerprint_grow:
  sensing_pin: GPIO13
  uart_id: fingerprint
  password: 0x00000000

  on_finger_scan_start:
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        color: GREEN
        speed: 10
        count: 1
  on_finger_scan_misplaced:
    - fingerprint_grow.aura_led_control:
        state: FLASHING
        speed: 25
        color: GREEN
        count: 2  

Final log:

> [0;36m[D][esp32.preferences:114]: Saving 1 preferences to flash...[0m
[0;36m[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed[0m
[0;32m[I][app:029]: Running through setup()...[0m
[0;35m[C][uart.arduino_esp32:077]: Setting up UART...[0m
[0;35m[C][fingerprint_grow:060]: Setting up Grow Fingerprint Reader...[0m
[0;36m[D][fingerprint_grow:232]: Checking password[0m
[0;36m[D][fingerprint_grow:237]: Password verified[0m
[0;36m[D][fingerprint_grow:260]: Getting parameters[0m
[0;36m[D][fingerprint_grow:263]: Got parameters[0m
[0;36m[D][fingerprint_grow:285]: Getting fingerprint count[0m
[0;36m[D][fingerprint_grow:288]: Got fingerprint count[0m

It's not working yet, but at least sensor start blinking and now I pretty sure it not came broken from Ali. 
Once again thank you very much!

mondalaci, 2024-03-21:
@alexborro I've tried to test `2024.3.0b2`, as suggested. Downloaded the tarball from https://github.com/esphome/esphome/releases/tag/2024.3.0b2 , extracted its content, ran `sudo setup.py install`, then my esphome installation got messed up:

```
laci@lian:~/projects/esphome$ esphome compile bearcave-entry.yaml 
INFO ESPHome 2024.2.2
INFO Reading configuration bearcave-entry.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing bearcave-entry (board: az-delivery-devkit-v4; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PermissionError: Traceback (most recent call last):
  File "/home/laci/.local/lib/python3.10/site-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/laci/.local/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/laci/.local/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/laci/.local/lib/python3.10/site-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
  File "/home/laci/.local/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/laci/.local/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/laci/.local/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/laci/.local/lib/python3.10/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/laci/.local/lib/python3.10/site-packages/platformio/run/cli.py", line 145, in cli
    process_env(
  File "/home/laci/.local/lib/python3.10/site-packages/platformio/run/cli.py", line 208, in process_env
    ).process()
  File "/home/laci/.local/lib/python3.10/site-packages/platformio/run/processor.py", line 81, in process
    install_project_env_dependencies(
  File "/home/laci/.local/lib/python3.10/site-packages/platformio/package/commands/install.py", line 132, in install_project_env_dependencies
    _install_project_env_platform(project_env, options),
  File "/home/laci/.local/lib/python3.10/site-packages/platformio/package/commands/install.py", line 147, in _install_project_env_platform
    if not pm.get_package(spec):
  File "/home/laci/.local/lib/python3.10/site-packages/platformio/package/manager/base.py", line 265, in get_package
    for pkg in self.get_installed():
  File "/home/laci/.local/lib/python3.10/site-packages/platformio/package/manager/base.py", line 241, in get_installed
    spec = self.build_legacy_spec(pkg.path)
  File "/home/laci/.local/lib/python3.10/site-packages/platformio/package/manager/_legacy.py", line 26, in build_legacy_spec
    for name in os.listdir(pkg_dir):
PermissionError: [Errno 13] Permission denied: '/home/laci/.platformio/platforms/espressif32'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

Would you please help me fix my esphome installation and install the suggested version?

mondalaci, 2024-04-10:
Works like a charm with esphome 2024.3.2

Thanks for the fix!