esphome/issues #5595: documentation or function for "accuracy_decimals" incorrect

christschnman, 2024-03-14:
### The problem

The documentation states that when using "accuracy_decimals" for a sensor, the value is sent to Home Assistant in full accuracy and only the display accuracy is preconfigured in the entity settings to the set decimal places.
After some testing, I cannot confirm the behavior stated here. Only the rounded value with the corresponding decimal places is passed on to HA.
Please change the documentation or the type of value transfer here.
I came across this behavior through the question about the difference between "accuracy_decimals" and a "round" filter and I still can't come to a clear conclusion about it.
Discussion here: https://community.home-assistant.io/t/difference-between-accuracy-decimals-and-round/703755/16

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.0

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

ADC/calculated temperature

### Example YAML snippet

```yaml
- platform: adc
    pin: A0
    name: "Spannung A0"
    id: spannung_a0
    update_interval: 0.01s
    unit_of_measurement: "V"
    accuracy_decimals: 3
    filters:
      - multiply: 3.3
      - throttle_average: 1s
    on_value:
      then:
        - lambda: |-
            id(temp).publish_state(id(spannung_a0).state);

  - platform: template
    name: "Temperature"
    id: temp
    unit_of_measurement: "°C"
    filters:
    accuracy_decimals: 1
      - calibrate_linear:
          - 1.443 -> 30.3
          - 2.053 -> 16.0
          - 2.190 -> 13.3
```

### Anything in the logs that might be useful for us?

```txt
[15:01:53][D][sensor:093]: 'Temperatur accuracy': Sending state 14.66433 °C with 2 decimals of accuracy
```

ssieb, 2024-03-14:
> Only the rounded value with the corresponding decimal places is passed on to HA.

How are you determining this?  Your pasted log message doesn't even correspond to the example config.

christschnman, 2024-03-14:
Thats right, i'm sorry. The config is from a different test than the log.
But in fact if I set "accuracy_decimals", there is no possibility in HA to increase the displayed decimals.
Please see this thread: https://community.home-assistant.io/t/difference-between-accuracy-decimals-and-round/703755/16

current config:
```
  - platform: adc
    pin: A0
    name: "Spannung A0"
    id: spannung_a0
    update_interval: 0.01s
    unit_of_measurement: "V"
    accuracy_decimals: 3
    filters:
      - multiply: 3.3
      - throttle_average: 1s
    on_value:
      then:
        - lambda: |-
            id(temp).publish_state(id(spannung_a0).state);

  - platform: template
    name: "Temperatur raw"
    id: temp
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    filters:
      - calibrate_linear:
          - 1.443 -> 30.3
          - 2.053 -> 16.0
          - 2.190 -> 13.3
```
log:
```
[20:23:17][D][sensor:093]: 'Spannung A0': Sending state 2.13672 V with 3 decimals of accuracy
[20:23:17][D][sensor:093]: 'Temperatur raw': Sending state 14.32299 °C with 1 decimals of accuracy
```
Up to this point, the full precision is available in ESPHome, but it seems (as stated in the logs) that from there only the set decimal places are transferred to HA and and not the full precision as the documentation suggests.

If I understood the documentation correctly, the value should be transferred to HA with full precision and just displayed with one decimal place, as stated in my case for the temperature.

In HA there is no way to increase the accuracy beyond this specified decimal point:

[image]

If that's the case, it's not a problem for me, I can work with it, but the documentation should be adjusted. Maybe someone can explain to me the benefit of accuracy_decimals over the round filter.

closed 2024-03-14 by ssieb

christschnman, 2024-03-14:
So the ESPHome integration on HA receives the value with full precision from the ESP-uController and ESPHome then passes the value rounded to HA, correct?

But please carefully read the description of “accuracy_decimals” in the documentation:

> - accuracy_decimals (Optional, int): Manually set the number of decimals to use when reporting values. This does not impact the actual value reported to Home Assistant, it just sets the number of decimals to use when displaying it.

This explanation clearly suggests that the value that is output is not changed (not rounded), but is only displayed rounded, exactly as can be specified in the entity settings. Or what is the meaning of the last sentence? Actually, the first and second sentences contradict each other somewhat.